merbanan/rtl_433 #1327: Fineoffset-WHx080 incorrect temperature when < 1°C

teabreakninja, 2020-03-02:
Have noticed some errors when my weather station is reading <1°C 
-----
time      : 2020-03-02 08:10:34
model     : Fineoffset-WHx080                      Msg type  : 0             Station ID: 201
Temperature: 307.4 C     Humidity  : 95 %          Wind degrees: 0           Wind avg speed: 15.91     Wind gust : 20.81
Total rainfall: 0.0      Battery   : LOW           Integrity : CRC
-----

The above temperature was reading 0.2°C in the station display

-----
time      : 2020-03-02 09:04:58
model     : Fineoffset-WHx080                      Msg type  : 0             Station ID: 201
Temperature: 1.6 C       Humidity  : 97 %          Wind degrees: 293         Wind avg speed: 18.36     Wind gust : 24.48
Total rainfall: 0.0      Battery   : LOW           Integrity : CRC
-----

When the temperature is above 1°C, then it is reporting the correct temperature as per the station display. The station is a Maplin NG96 (basically a Fine Offset WH1081)

zuckschwerdt, 2020-03-02:
That's somewhat strange as the temperature here is `raw*0.1 - 40.0`. That way e.g. 1.6°C is 416 in the data. So your 307.4 °C would have been 3474, that is in range for 2's complement but would be -622, so -62.2°C or -22.2°C if that -40 offset applies. Not matching the 0.2 °C you had.
Can you record a sample with the bad values?

teabreakninja, 2020-03-02:
Sure, next time it drops below 1°C I'll grab a sample or two.

teabreakninja, 2020-03-05:
Sample captured
[g007_433.92M_250k.cu8.zip](https://github.com/merbanan/rtl_433/files/4291274/g007_433.92M_250k.cu8.zip)

zuckschwerdt, 2020-03-05:
Do you know what temp value that should be?

teabreakninja, 2020-03-05:
it's currently showing 0.0°C in the weather display

zuckschwerdt, 2020-03-05:
If that's exactly 0.0°C then the temp isn't 12 bit but 10 bit, i.e. -40.0 to 62.4 -- which seems reasonable. You can test that theory by replacing `0x0f` with `0x03` in https://github.com/merbanan/rtl_433/blob/master/src/devices/fineoffset_wh1080.c#L191
```
int temp_raw      = ((br[2] & 0x03) << 8) | br[3];
```

teabreakninja, 2020-03-05:
Looking good!
time      : 2020-03-05 08:49:50
model     : Fine Offset Electronics WH1080/WH3080 Weather Station            Msg type  : 0             Station ID: 201
Temperature: 0.0 C       Humidity  : 95 %          Wind degrees: 270         Wind avg speed: 0.00
Wind gust : 0.00         Total rainfall: 0.0       Battery   : LOW           Integrity : CRC

zuckschwerdt, 2020-03-05:
Would be interesting what the top bits flag then? Negative temp or freezing would be kind of pointless. Maybe "lowest value so far" or such?
You can debug that by adding a line in data_make (at around line 237):
```
"temp_flags",               "Temp Flags",       DATA_INT,       (br[2] & 0x0C) >> 2,
```

teabreakninja, 2020-03-05:
time      : 2020-03-05 09:00:14
model     : Fine Offset Electronics WH1080/WH3080 Weather Station            Msg type  : 0             Station ID: 201
Temperature: 0.0 C       Humidity  : 95 %          Wind degrees: 248         Wind avg speed: 0.00
Wind gust : 0.00         Total rainfall: 0.0       Battery   : LOW           Integrity : CRC
Temp Flags: 3

zuckschwerdt, 2020-03-05:
Looking good. Can you watch that value over the next few days and report any findings?

teabreakninja, 2020-03-05:
Will do. Thanks for your help :+1:

partoneoftwo, 2020-03-10:
Hi

I'm new here. Not sure if this is the way to respond to this thread.

The Fineoffset devices do this, I have the same issue when collecting data
over USB to my weather station.
The temperature seems to roll over on 0 degrees celcius, back to -273. It
also rolls over at +43 degrees. It becomes impossible to trust it to power
any kind of home automation because of this.

All the best
Thomas

zuckschwerdt, 2020-03-10:
@partoneoftwo thanks, that's interesting: it rolls over at +43°C? It should be a little higher.
Importantly: what does it show at 43°C?

teabreakninja, 2020-03-10:
I never had any issues with values via USB, but the display's USB controller was unreliable and would require a reset now and again. It has stopped working totally so now I use the RTL for decoding/logging and the display to... well, display!

partoneoftwo, 2020-03-10:
My bad. I meant 25.6 degrees.
I found my own notes, conveniently comitted to Github.

Here they are:
https://github.com/partoneoftwo/hp3000-weather-station-data

The output of the temperature sensors are confusing. These are the thresholds

- At temperatures between 0.1 Degrees Celcius as 25.6 Degrees celcius The formula becomes Raw Value / 10
- at 0 Degrees Celcius, which becomes the raw value of 256 The formula becomes 256-Raw Value
- at 25.6 Degrees Celcius, which becomes the raw value of 256 The formula becomes 256-Raw Value / 10

zuckschwerdt, 2020-03-10:
@partoneoftwo If I understand correctly the `Raw` in your findings above is from some USB mechanism, not the value from our current implementation or the actual raw radio packet value?

Our current formula is : `(raw - 400) * 0.1` (where `raw` is *some* 12 bit from the radio data).
We assumed the field in the radio packet is 12 bit, but the leading two bits seem to do something different?
That leaves 10 bit, i.e. 0 to 1023, so -400 to 623, scaled -40.0 to 62.3 which looks reasonable and seems to work. @teabreakninja is currently looking what those two top bits might flag.

partoneoftwo, 2020-03-11:
@zuckschwerdt This is correct, the mechanism is USB. The values I am fetching with RTL_433 are rendered correctly.

I thought I might contribute with these insights since the problem description here seems similar if not identical to what I was experiencing. The root cause might be completely different, so maybe this is a dead end.

teabreakninja, 2020-03-12:
I've been monitoring the temperature readings and all seems good for the most part. I've attached some snippets of output, to show the two top bits result in either 3 or 0. i.e. they're all set or not.

On reviewing the temperatures, some of the readings seem a bit strange. If you look at the start of the file, you'll see that the temperature drops from 3.3°C to 0.7°C in just under a minute which I find a bit odd. Might be a red hearing and a result of the low battery (I'm waiting for the weather to improve so I can change them).

[weather-readings.json.zip](https://github.com/merbanan/rtl_433/files/4322794/weather-readings.json.zip)

zuckschwerdt, 2020-03-12:
Thanks. Changes are in merged to master now.

commit 9a20cb11 (2020-03-12): Fix Fineoffset-WHx080 temperature (closes #1327)
closed 2020-03-12 by zuckschwerdt