merbanan/rtl_433 #1328: Standardize fields across categories of device?

bjmc, 2020-03-02:
I've been tinkering with a bridge that will serialize rtl_433 messages to Apache Avro format and send them to a Kafka message broker. Several times, I've run into issues where [the Avro schema](http://avro.apache.org/docs/current/spec.html#schemas) I've defined turned out to be too strict and so my serialization failed with type errors. (To take one example, it appears some TPMS devices represent `flags` as a string, others as an integer, and some omit it entirely).

I don't know how practical this would be, but I was thinking it would be great if fields could be standardized across categories of device. I was hoping that maybe all weather devices could use the same field names and data types, similarly for all TPMS devices, etc.

Would there be any appetite for defining and enforcing [JSONSchema documents](https://json-schema.org/) for similar categories of device?

zuckschwerdt, 2020-03-02:
There is some work towards this documented in https://triq.org/rtl_433/DATA_FORMAT.html
Maybe you can summarize your findings (like with `flags`) and we can start a discussion.
Not sure if it's possible to fit the all the various quirks from data packets in a fully normalized format. Perhaps best to start small ;)

merbanan, 2020-03-03:
Most devices are unique in their own way. Sometimes we just haven't figured out the format encoding. It would be nice if things could be standardized but I don't see how to do it. AFAIK there is no other container level standard we could piggy back on. Then it would just be an arbitrary classification. For example TPMS is a well contained concept that could be a class. But a what about a weather station? Is just a thermo/hygro-meter a weather station? In that case a classification will just be arbitrary. What is the difference between a cooking thermometer and a pool thermometer? Both will have a temperature sensor but different use. So I think just having a key named temperature_C is quite informative on its own.

For me the only reliable way is to handle each sensor model on it's own. If you filter on the model then you know how to use the data properly. I am open to suggestions.

bjmc, 2020-03-10:
I agree that any classification we agree on will be ultimately arbitrary, but I'm not sure I agree that makes it useless. Maybe there's no existing standard for us to lean on, but there's no reason this piece of software couldn't define its own standard to be consistent within the devices it supports.

You ask "Is just a thermo/hygro-meter a weather station?" and I'd answer, "Sure, yes. it measures things about the weather." That's an arbitrary judgement, but feels intuitively correct to me.

Are there specific edge cases you're worried about being hard to classify? We could undertake this effort gradually by adding an "uncategorized" category that would initially contain everything, and gradually migrate/standardize to agreed-upon classes of device. At the end, perhaps a few oddball devices could remain uncategorized.

merbanan, 2021-04-12:
I understand the logic with this issue but grouping/classifying devices is hard. The way it works now seems to work for most of the users.

closed 2021-04-12 by merbanan