merbanan/rtl_433 #1331: using -r argument returns different results than the original capture

dtiller, 2020-03-04:
Gentlemen,

I'm writing another decoder (#150 when I last cloned), and I've run into an interesting issue. I executed rtl_433 like this:

rtl_433 -R 150 -f 904.1M -S all

I get output like this (my decoder searches for a preamble, dumps the bitbuffer, then inverts and dumps the buffer again):
[g004_904.1M_1000k.cu8.zip](https://github.com/merbanan/rtl_433/files/4290005/g004_904.1M_1000k.cu8.zip)

```rtl_433 version unknown inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/Users/dtiller/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 1 out of 150 device decoding protocols [ 150 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 904.100MHz.

PREAMBLE: found at: 48
bitbuffer:: Number of rows: 1 
[00] {572} aa aa aa aa aa aa aa 86 2f fc ff e4 7c 6a 6f 29 0e 08 a2 eb 52 66 b1 64 f7 3a 8b 47 59 d3 56 d9 9f 23 f2 90 78 f4 e1 78 06 b5 82 a8 c9 14 cd 02 19 11 ab 5d a5 cc 61 c9 ee 8a 12 0f 2c a4 af b1 36 cf c5 f6 78 0a e1 00 
Inverted
bitbuffer:: Number of rows: 1 
[00] {572} 55 55 55 55 55 55 55 79 d0 03 00 1b 83 95 90 d6 f1 f7 5d 14 ad 99 4e 9b 08 c5 74 b8 a6 2c a9 26 60 dc 0d 6f 87 0b 1e 87 f9 4a 7d 57 36 eb 32 fd e6 ee 54 a2 5a 33 9e 36 11 75 ed f0 d3 5b 50 4e c9 30 3a 09 87 f5 1e f0 
*** Saving signal to file g004_904.1M_1000k.cu8 (46061 samples, 131072 bytes)
^CSignal caught, exiting!```

You can see the aa aa or 55 55 preamble clearly. 

When I then execute rtl_433 with the -r option without the -s option, I get:

```rtl_433 -R 150 -r g004_904.1M_1000k.cu8 
rtl_433 version unknown inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/Users/dtiller/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 1 out of 150 device decoding protocols [ 150 ]
Test mode active. Reading samples from file: g004_904.1M_1000k.cu8
I LIVE!!!
PREAMBLE: found at: 496
bitbuffer:: Number of rows: 1 
[00] {496} f0 03 ff ff c3 ff ff fc 03 ff cc f3 30 f3 fc 33 0c 03 f0 03 03 30 3f f0 3c cc 30 f0 f0 3f 0f c3 c3 fc 0f c0 c0 3f 00 fc 0f 0f 30 cc 33 c0 f0 cf 03 0f 03 0f 00 03 c0 fc 0c 3f cc 0c c0 03 
Inverted
bitbuffer:: Number of rows: 1 
[00] {496} 0f fc 00 00 3c 00 00 03 fc 00 33 0c cf 0c 03 cc f3 fc 0f fc fc cf c0 0f c3 33 cf 0f 0f c0 f0 3c 3c 03 f0 3f 3f c0 ff 03 f0 f0 cf 33 cc 3f 0f 30 fc f0 fc f0 ff fc 3f 03 f3 c0 33 f3 3f fc 
I LIVE!!!
PREAMBLE: found at: 1024
bitbuffer:: Number of rows: 1 
[00] {1024} f0 33 ff ff c3 ff ff f0 c0 ff c0 f0 30 f3 fc 33 0c 03 f0 00 03 f0 3f ff 3c c0 30 f0 f3 3c 0c f0 c3 fc fc 3f 33 03 3c 00 fc cf 0f fc 3c cc f3 cf 0f 0f fc 30 3f fc 30 0c 00 ff 03 ff c3 f0 0c ff 00 00 ff fc 0f 00 3f ff 03 c0 0c 0f c3 c3 fc 00 30 0f 0c 0c 0f 30 3c cf cf 30 cf c3 c0 f0 0f c3 0f f3 f0 03 30 0c 30 03 fc 3f c3 00 c0 33 ff 3c 0c 3c f3 c3 ff c0 ff fc f0 ff 00 00 33 f0 0c 00 
Inverted
bitbuffer:: Number of rows: 1 
[00] {1024} 0f cc 00 00 3c 00 00 0f 3f 00 3f 0f cf 0c 03 cc f3 fc 0f ff fc 0f c0 00 c3 3f cf 0f 0c c3 f3 0f 3c 03 03 c0 cc fc c3 ff 03 30 f0 03 c3 33 0c 30 f0 f0 03 cf c0 03 cf f3 ff 00 fc 00 3c 0f f3 00 ff ff 00 03 f0 ff c0 00 fc 3f f3 f0 3c 3c 03 ff cf f0 f3 f3 f0 cf c3 30 30 cf 30 3c 3f 0f f0 3c f0 0c 0f fc cf f3 cf fc 03 c0 3c ff 3f cc 00 c3 f3 c3 0c 3c 00 3f 00 03 0f 00 ff ff cc 0f f3 ff 
```

Not even close.

If I add the -s 1000000 option, I get different values again:

```rtl_433 -R 150 -s 1000000 -r g004_904.1M_1000k.cu8 
rtl_433 version unknown inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/Users/dtiller/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 1 out of 150 device decoding protocols [ 150 ]
Test mode active. Reading samples from file: g004_904.1M_1000k.cu8
I LIVE!!!
PREAMBLE: found at: 172
bitbuffer:: Number of rows: 1 
[00] {172} 87 f7 f8 f6 d7 68 c2 67 96 a4 cc 9c c8 c6 4a d9 32 8a 08 cb a6 10 
Inverted
bitbuffer:: Number of rows: 1 
[00] {172} 78 08 07 09 28 97 3d 98 69 5b 33 63 37 39 b5 26 cd 75 f7 34 59 e0 
I LIVE!!!
PREAMBLE: found at: 302
bitbuffer:: Number of rows: 1 
[00] {302} 9f df 97 25 6d 30 67 a9 56 6b 6f 63 dd 75 5d 7a 46 7b 38 3c 8f 22 d6 12 aa bd 74 8d 6c 65 19 92 7a 55 7b d6 07 20 
Inverted
bitbuffer:: Number of rows: 1 
[00] {302} 60 20 68 da 92 cf 98 56 a9 94 90 9c 22 8a a2 85 b9 84 c7 c3 70 dd 29 ed 55 42 8b 72 93 9a e6 6d 85 aa 84 29 f8 dc 
```

Shouldn't I get the same results in all 3 cases? The docs say the sample rate is divined from the filename - that doesn't seem to be happening.

I attached the above sample file.

Any ideas? Thanks!

zuckschwerdt, 2020-03-05:
> Shouldn't I get the same results in all 3 cases? The docs say the sample rate is divined from the filename - that doesn't seem to be happening.

Yeah, sorry, that's not currently done. There is only a default.

dtiller, 2020-03-05:
Christian,

Thank yo for your reply. The main thrust of this issue is that, even with the explicit sample rate parameter, I still do not get the same results as when I originally ran the capture.

If you compare the runs in the first and third 'code' boxes in the original post, you'll see that the results decoded from the RTLSDR and the results decoded from the sample file that was written by the RTLSDR run are wildly different.

Run from the RTLSDR (rtl_433 -R 150 -f 904.1M -S all) yielded:
[00] {572} aa aa aa aa aa aa aa 86 2f fc ff e4 7c 6a 6f 29 0e 08 a2 eb 52 66 b1 64 f7 3a 8b 47 59 d3 56 d9 9f 23 f2 90 78 f4 e1 78 06 b5 82 a8 c9 14 cd 02 19 11 ab 5d a5 cc 61 c9 ee 8a 12 0f 2c a4 af b1 36 cf c5 f6 78 0a e1 00 

Using the file written from that run with (rtl_433 -R 150 -s 1000000 -r g004_904.1M_1000k.cu8) I get:

[00] {172} 87 f7 f8 f6 d7 68 c2 67 96 a4 cc 9c c8 c6 4a d9 32 8a 08 cb a6 10

[00] {302} 9f df 97 25 6d 30 67 a9 56 6b 6f 63 dd 75 5d 7a 46 7b 38 3c 8f 22 d6 12 aa bd 74 8d 6c 65 19 92 7a 55 7b d6 07 20 

I'd really like to be able to use known good data to tweak the parameters of my decoder. 

Thanks again!

zuckschwerdt, 2020-03-05:
Does your decoder use FSK? Then you likely want the `-Y minmax` demod to match the default for higher frequencies.

dtiller, 2020-03-05:
Christian,

It does indeed use FSK, and the -Y minmax did help - thanks! I get some additional buffers that weren't there in the original capture, but I do get the correct bitbuffer later on.

While you're adding the sample rate code, could you please make the behavior between using a real device and a data file the same? I know that the RTL code probably uses callbacks, but that can be emulated for files, too. (I did it as a hack to rtl_fm a while ago. It's wasn't pretty, but it could be  🙄)

Thanks again!

zuckschwerdt, 2023-10-04:
This was solved with eae5971