merbanan/rtl_433 #1335: HackRF: No supported devices found

OZ1SEJ, 2020-03-14:
I'm trying rtl_433 with my hackrf, which results in "No supported devices found." I've installed hackrf, libhackrf-dev and soapysdr. Do I need to do something special?

zuckschwerdt, 2020-03-14:
Check that
- you compiled rtl_433 with support for SoapySDR (`rtl_433 -V` on first line).
- you have the [SoapyHackRF](https://github.com/pothosware/SoapyHackRF) module installed (`SoapySDRUtil --info`)
- SoapySDR finds the receiver (`SoapySDRUtil --find`)

closed 2020-03-14 by zuckschwerdt

makuser, 2020-03-17:
@OZ1SEJ I have the same problem as you. Which distro and version are you using and how did you install soapysdr? I am running Ubuntu 19.10 and 18.04 and tried the official pothosware ppa on both machines, but also compiled the sources myself. No success.

What is the output of the commands that @zuckschwerdt asked for on your machine?

Posting my output below after manually compiling and installing SoapySDR and its SoapyHackRF module. Using the official ppa yielded similar results, although different versions:
```
14:39:11,21 marc@MAKlaptop: ~ > rtl_433 -V
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/marc/.config/rtl_433/rtl_433.conf"...
Reading conf from "/home/marc/.config/rtl_433/rtl_433.conf".
14:39:16,22 marc@MAKlaptop: ~ > SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-gf722f9ce
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libHackRFSupport.so (0.3.3-3c514ce)
Available factories... hackrf
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
14:39:21,23 marc@MAKlaptop: ~ > SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  device = HackRF One
  driver = hackrf
  label = HackRF One #0 57b068dc23632263
  part_id = a000cb3c00674763
  serial = 000000000000000057b068dc23632263
  version = 2018.01.1

14:39:26,24 marc@MAKlaptop: ~ > 
```

I do however think that this has everything to do with soapysdr and nothing with rtl_433, as for example gqrx supports the HackRF both via SoapySDR/SoapyHackRF and also via gr-osmosdr. The HackRF works just fine via gr-osmosdr, just not via SoapySDR when using gqrx, so I think we need to find and fix the problem at the origin (library in this case), not the client application that uses it.

I have not had much time to figure out why that is exactly, but maybe you mind sharing some more information about your setup so that we can figure out if either of us did something wrong.

zuckschwerdt, 2020-03-17:
You mentioned a ppa. Make sure you don't have two SoapySDR versions installed, e.g. check that
`ls -al /usr/lib/libSoapySDR.* /usr/lib/SoapySDR/modules*` is empty.

EvilAcid, 2020-05-15:
i Have the Same Problem

rtl_433 dosnd find my HackRF One.

Raspberry Pi4

```
pi@raspberrypi4-iot:~/cubicsdr/SoapySDR/build $ rtl_433 -G 4
rtl_433 version 20.02-50-g9700fee branch master at 202005142008 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

Use -G for testing only. Enable protocols with -R if you really need them.

Registered 146 out of 152 device decoding protocols [ 1-4 6-8 10-17 19-26 29-64 67-152 ]
rtl_433: warning: 106 "Wireless M-Bus, Mode R, 4.8kbps (-f 868330000)" does not support CSV output
rtl_433: warning: 107 "Wireless M-Bus, Mode F, 2.4kbps" does not support CSV output
No supported devices found.
```

```
pi@raspberrypi4-iot:~/cubicsdr/SoapySDR/build $ sudo SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-gf722f9ce
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8 (missing)
No modules found!
Available factories... No factories found!
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]
```

the HackRF One works an my Local machin with the same usbcabel.

Thanks for the Software and help =)

zuckschwerdt, 2020-05-15:
@EvilAcid you don't have any support modules installed? (esp. HackRF)
```
Search path:  /usr/local/lib/SoapySDR/modules0.8 (missing)
```

EvilAcid, 2020-05-15:
@zuckschwerdt 

Many thanks for the very fast replay =)
And sorry for my EN =)

I have installet only rtl_433 --> dont working
I have installed hackRF
```
pi@raspberrypi4-iot:~ $ sudo hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Board ID Number: 2 (HackRF One)
Firmware Version: 2014.08.1 (API:1.00)
Part ID Number: 0xa000cb3c 0x0068475a
```
but rtl_433 dosnt work

i try to install SoapySDR but nothing working =)

zuckschwerdt, 2020-05-15:
The installation order matters: first SoapySDR, then rtl_433.
And use `cmake -DENABLE_SOAPYSDR=ON ..` to force SoapySDR detection. I.e.
```
cd rtl_433 ; mkdir build ; cd build ; cmake -DENABLE_SOAPYSDR=ON .. && make
```
But you also need the https://github.com/pothosware/SoapyHackRF module.

EvilAcid, 2020-05-15:
Hey @zuckschwerdt
Many thanks for your ideas
 
I doo what you say, but the answare are the same.
```
pi@raspberrypi4-iot:~/rtl_433/build $ cmake -DENABLE_SOAPYSDR=ON ..
-- The C compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found Git version: refs/heads/master commit 9700feeb9d5b494df5f45d39eb0a4759f29b8ee2 from 2020-05-14T20:08:42+0200
-- Using Git version tag: 20.02-50-g9700fee on master at 202005142008
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29")
-- Found LibRTLSDR: /usr/local/lib/librtlsdr.so
-- Found LibUSB: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so
-- RTL-SDR device input will be compiled.
-- SoapySDR device input will be compiled.
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/rtl_433/build
pi@raspberrypi4-iot:~/rtl_433/build $
pi@raspberrypi4-iot:~/rtl_433/build $ make
Scanning dependencies of target r_433
[  1%] Building C object src/CMakeFiles/r_433.dir/abuf.c.o
[  1%] Building C object src/CMakeFiles/r_433.dir/am_analyze.c.o
[  2%] Building C object src/CMakeFiles/r_433.dir/baseband.c.o
[  2%] Building C object src/CMakeFiles/r_433.dir/bitbuffer.c.o
[  3%] Building C object src/CMakeFiles/r_433.dir/compat_paths.c.o
[  4%] Building C object src/CMakeFiles/r_433.dir/compat_time.c.o
[  4%] Building C object src/CMakeFiles/r_433.dir/confparse.c.o
[  5%] Building C object src/CMakeFiles/r_433.dir/data.c.o
[  5%] Building C object src/CMakeFiles/r_433.dir/decoder_util.c.o
[  6%] Building C object src/CMakeFiles/r_433.dir/fileformat.c.o
[  6%] Building C object src/CMakeFiles/r_433.dir/list.c.o
[  7%] Building C object src/CMakeFiles/r_433.dir/mongoose.c.o
[  7%] Building C object src/CMakeFiles/r_433.dir/optparse.c.o
[  8%] Building C object src/CMakeFiles/r_433.dir/output_influx.c.o
[  9%] Building C object src/CMakeFiles/r_433.dir/output_mqtt.c.o
[  9%] Building C object src/CMakeFiles/r_433.dir/pulse_demod.c.o
[ 10%] Building C object src/CMakeFiles/r_433.dir/pulse_detect.c.o
[ 10%] Building C object src/CMakeFiles/r_433.dir/pulse_detect_fsk.c.o
[ 11%] Building C object src/CMakeFiles/r_433.dir/r_api.c.o
[ 11%] Building C object src/CMakeFiles/r_433.dir/r_util.c.o
[ 12%] Building C object src/CMakeFiles/r_433.dir/samp_grab.c.o
[ 13%] Building C object src/CMakeFiles/r_433.dir/sdr.c.o
[ 13%] Building C object src/CMakeFiles/r_433.dir/term_ctl.c.o
[ 14%] Building C object src/CMakeFiles/r_433.dir/util.c.o
[ 14%] Building C object src/CMakeFiles/r_433.dir/write_sigrok.c.o
[ 15%] Building C object src/CMakeFiles/r_433.dir/devices/acurite.c.o
[ 15%] Building C object src/CMakeFiles/r_433.dir/devices/akhan_100F14.c.o
[ 16%] Building C object src/CMakeFiles/r_433.dir/devices/alecto.c.o
[ 17%] Building C object src/CMakeFiles/r_433.dir/devices/ambient_weather.c.o
[ 17%] Building C object src/CMakeFiles/r_433.dir/devices/ambientweather_tx8300.c.o
[ 18%] Building C object src/CMakeFiles/r_433.dir/devices/ambientweather_wh31e.c.o
[ 18%] Building C object src/CMakeFiles/r_433.dir/devices/archos_tbh.c.o
[ 19%] Building C object src/CMakeFiles/r_433.dir/devices/auriol_afw2a1.c.o
[ 19%] Building C object src/CMakeFiles/r_433.dir/devices/auriol_hg02832.c.o
[ 20%] Building C object src/CMakeFiles/r_433.dir/devices/blyss.c.o
[ 20%] Building C object src/CMakeFiles/r_433.dir/devices/brennenstuhl_rcs_2044.c.o
[ 21%] Building C object src/CMakeFiles/r_433.dir/devices/bresser_3ch.c.o
[ 22%] Building C object src/CMakeFiles/r_433.dir/devices/bresser_5in1.c.o
[ 22%] Building C object src/CMakeFiles/r_433.dir/devices/bt_rain.c.o
[ 23%] Building C object src/CMakeFiles/r_433.dir/devices/calibeur.c.o
[ 23%] Building C object src/CMakeFiles/r_433.dir/devices/cardin.c.o
[ 24%] Building C object src/CMakeFiles/r_433.dir/devices/chuango.c.o
[... 185 lines cut ...]
-- Up-to-date: /usr/local/etc/rtl_433/sonoff_rm433.conf
-- Up-to-date: /usr/local/etc/rtl_433/steffen_switch.conf
-- Up-to-date: /usr/local/etc/rtl_433/valeo_car_key.conf
-- Up-to-date: /usr/local/etc/rtl_433/verisure_alarm.conf

```

But the answer is the same 
```
pi@raspberrypi4-iot:~/rtl_433/build $ rtl_433 -G 4
rtl_433 version 20.02-50-g9700fee branch master at 202005142008 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

        Use -G for testing only. Enable protocols with -R if you really need them.

Registered 146 out of 152 device decoding protocols [ 1-4 6-8 10-17 19-26 29-64 67-152 ]
rtl_433: warning: 106 "Wireless M-Bus, Mode R, 4.8kbps (-f 868330000)" does not support CSV output
rtl_433: warning: 107 "Wireless M-Bus, Mode F, 2.4kbps" does not support CSV output
No supported devices found.

```

EvilAcid, 2020-05-15:
now i'm confused
I found the HackRF in SoapySDRUtil. On rtl_433 it cannot be opened.
```
pi@raspberrypi4-iot:~ $ rtl_433 -d ""
rtl_433 version 20.02-50-g9700fee branch master at 202005142008 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 124 out of 152 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-152 ]
[INFO] Opening HackRF One #0 406464c82383314b...
[INFO] Could not Open HackRF Device
pi@raspberrypi4-iot:~ $ SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  device = HackRF One
  driver = hackrf
  label = HackRF One #0 406464c82383314b
  part_id = a000cb3c0068475a
  serial = 0000000000000000406464c82383314b
  version = 2014.08.1

pi@raspberrypi4-iot:~ $

```

zuckschwerdt, 2020-05-15:
Looks good! Not sure about the "Could not Open HackRF Device", maybe permissions? Try `sudo` perhaps.

EvilAcid, 2020-05-15:
nope, the same

hmm :-( 
```
pi@raspberrypi4-iot:~ $ sudo rtl_433 -d ""
rtl_433 version 20.02-50-g9700fee branch master at 202005142008 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 124 out of 152 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-152 ]
[INFO] Opening HackRF One #0 406464c82383314b...
[INFO] Could not Open HackRF Device
```

zuckschwerdt, 2020-05-15:
Is the device perhaps busy? The `[INFO]` messages are from the SoapyHackRF driver. The device is found but can't be used.
Hopefully someone with a HackRF can help out here -- I never had access to any HackRF.

EvilAcid, 2020-05-15:
I think i have a solution

i have upgrade the firmare from 2014.08.1 to 2018.01.1

```
cd /tmp
wget https://github.com/mossmann/hackrf/releases/download/v2018.01.1/hackrf-2018.01.1.tar.xz
tar -xvf hackrf-2018.01.1.tar.xz
cd hackrf-2018.01.1/firmware-bin/
hackrf_spiflash -w hackrf_one_usb.bin
```
Then Restart (press Reset) or reconnect the Power(usb)

[image]

Big Thank to @zuckschwerdt

zuckschwerdt, 2020-05-15:
Great! Thanks for documenting your findings and the solution.

soyoeml, 2020-10-16:
My case is a little bit different. It cannot find modules and factories.
run SoapySDRUtil --info
Please check your libHackRFSupport.so is in /opt/local/lib/SoapySDR/modules0.7/ folder or not.

Sometimes when you run make install, it installed to /usr/local/lib/SoapySDR/modules0.7/.

You need cp /usr/local/lib/SoapySDR/modules0.7/libHackRFSupport.so /opt/local/lib/SoapySDR/modules0.7/

Then run SoapySDRUtil --info to see if there is such a module.