merbanan/rtl_433 #1337: New unknown device DC118W

Fredy72, 2020-03-21:
Hi, I have this product DC118W, probably a Dooya brand.
[https://www.aliexpress.com/item/4000452301739.html?spm=2114.12010615.8148356.25.3bbc331a9InsSk](url)

[g001_433.92M_250k.cu8.txt](https://github.com/merbanan/rtl_433/files/4362809/g001_433.92M_250k.cu8.txt)
[g002_433.92M_250k.cu8.txt](https://github.com/merbanan/rtl_433/files/4362810/g002_433.92M_250k.cu8.txt)
[g003_433.92M_250k.cu8.txt](https://github.com/merbanan/rtl_433/files/4362811/g003_433.92M_250k.cu8.txt)
[g005_433.92M_250k.cu8.txt](https://github.com/merbanan/rtl_433/files/4362812/g005_433.92M_250k.cu8.txt)
[g006_433.92M_250k.cu8.txt](https://github.com/merbanan/rtl_433/files/4362813/g006_433.92M_250k.cu8.txt)

It should communicate as a DC90 accesories.
[https://www.aliexpress.com/item/32811888324.html?spm=a2g0o.productlist.0.0.3d8b7622hPWInf&algo_pvid=e499c3b6-fc3f-4787-b17e-7bcfe67de21a&algo_expid=e499c3b6-fc3f-4787-b17e-7bcfe67de21a-0&btsid=0be3764515847919567978101e228d&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_](url)

[https://www.moreland.dk/g/pdf/80016387/0/manual.pdf](url)

I found a little about the protocol.
[https://github.com/sui77/rc-switch/issues/54](url)

[https://github.com/arendst/Tasmota/pull/5658/commits/6441c7cd9325f2191ee98d341b8c885c5f00bc4d](url)

[https://github.com/codedealer/pimatic-dooya](url)

[https://github.com/Portisch/RF-Bridge-EFM8BB1/commit/c53fa5e1c47fc8ecda0a43c72c517b458666c6ce](url)

[https://github.com/Portisch/RF-Bridge-EFM8BB1/issues/29](url)

[https://github.com/arendst/Tasmota/issues/3673](url)

[https://github.com/liebrandapps/emuDC90](url)

I would like to be able to read the status so I can add the product to the Home Assistant.
Thank you very much.

zuckschwerdt, 2020-03-21:
The samples don't have enough silence at the start and won't work as-is. Here is a manually extracted OOK file. Drop this on https://triq.org/pdv/ to get an idea of the protocol.

[g001_DC118W_433.92M_250k.ook](https://github.com/merbanan/rtl_433/files/4362897/g001_DC118W_433.92M_250k.ook.txt)

Fredy72, 2020-03-21:
Clean continuous record, I sent 5x the data from the product.
[test.cu8.txt](https://github.com/merbanan/rtl_433/files/4363145/test.cu8.txt)
I don't know how to convert them to OOK to Pulseview.

Fredy72, 2020-03-21:
rtl_433 -X 'n=DC118W,m=OOK_PWM,s=360,l=720,r=2048,g=0,t=50,y=4760' -A

this is pulse for (FF-EE) - 17 wakeup:
Detected OOK package	2020-03-21 18:42:40
Analyzing pulses...
Total count:  205,  width: 253.99 ms		(63497 S)
Pulse width distribution:
 [ 0] count:    5,  width: 4764 us [4636;4804]	(1191 S)
 [ 1] count:  130,  width:  360 us [356;364]	(  90 S)
 [ 2] count:   70,  width:  716 us [712;724]	( 179 S)
Gap width distribution:
 [ 0] count:    5,  width: 1560 us [1560;1560]	( 390 S)
 [ 1] count:  130,  width:  716 us [716;724]	( 179 S)
 [ 2] count:   65,  width:  360 us [356;368]	(  90 S)
 [ 3] count:    4,  width: 2040 us [2040;2044]	( 510 S)
Pulse period distribution:
 [ 0] count:    5,  width: 6324 us [6196;6364]	(1581 S)
 [ 1] count:  195,  width: 1076 us [1076;1088]	( 269 S)
 [ 2] count:    4,  width: 2760 us [2760;2764]	( 690 S)
Level estimates [high, low]:   8251,      9
RSSI: -3.0 dB SNR: 29.6 dB Noise: -32.6 dB
Frequency offsets [F1, F2]:    1915,      0	(+7.3 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 360, long_width: 716, reset_limit: 2048, sync_width: 4764
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=360,l=716,r=2048,g=0,t=0,y=4764'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 5 
[00] {40} ff f2 0e 8e ee : 11111111 11110010 00001110 10001110 11101110 
[01] {40} ff f2 0e 8e ee : 11111111 11110010 00001110 10001110 11101110 
[02] {40} ff f2 0e 8e ee : 11111111 11110010 00001110 10001110 11101110 
[03] {40} ff f2 0e 8e ee : 11111111 11110010 00001110 10001110 11101110 
[04] {40} ff f2 0e 8e ee : 11111111 11110010 00001110 10001110 11101110

this is pulse for (FF-33) -  204 learning ?
Detected OOK package	2020-03-21 18:42:49
Analyzing pulses...
Total count:  410,  width: 540.99 ms		(135247 S)
Pulse width distribution:
 [ 0] count:   10,  width: 4760 us [4600;4804]	(1190 S)
 [ 1] count:  240,  width:  360 us [352;364]	(  90 S)
 [ 2] count:  160,  width:  716 us [716;724]	( 179 S)
Gap width distribution:
 [ 0] count:   10,  width: 1560 us [1560;1564]	( 390 S)
 [ 1] count:  230,  width:  716 us [716;724]	( 179 S)
 [ 2] count:  160,  width:  360 us [356;368]	(  90 S)
 [ 3] count:    8,  width: 2396 us [2396;2400]	( 599 S)
 [ 4] count:    1,  width: 33884 us [33884;33884]	(8471 S)
Pulse period distribution:
 [ 0] count:   10,  width: 6320 us [6164;6364]	(1580 S)
 [ 1] count:  390,  width: 1076 us [1076;1088]	( 269 S)
 [ 2] count:    8,  width: 2756 us [2756;2764]	( 689 S)
 [ 3] count:    1,  width: 34248 us [34248;34248]	(8562 S)
Level estimates [high, low]:   7855,      4
RSSI: -3.2 dB SNR: 32.9 dB Noise: -36.1 dB
Frequency offsets [F1, F2]:    1954,      0	(+7.5 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 360, long_width: 716, reset_limit: 33888, sync_width: 4760
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=360,l=716,r=33888,g=0,t=0,y=4760'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 10 
[00] {40} ff f2 0e 8e 33 : 11111111 11110010 00001110 10001110 00110011 
[01] {40} ff f2 0e 8e 33 : 11111111 11110010 00001110 10001110 00110011 
[02] {40} ff f2 0e 8e 33 : 11111111 11110010 00001110 10001110 00110011 
[03] {40} ff f2 0e 8e 33 : 11111111 11110010 00001110 10001110 00110011 
[04] {40} ff f2 0e 8e 33 : 11111111 11110010 00001110 10001110 00110011

and "ff f2 0e 8e cc" is pulse for (FF-CC) -  51 wakedown

Fredy72, 2020-03-21:
10x one codes a 10x second codes
[test10.ook.txt](https://github.com/merbanan/rtl_433/files/4363672/test10.ook.txt)

klohner, 2020-03-25:
I took a look through the samples.  They look like ASK, 360μs/symbol, PWM over 3 symbols length.  I'm guessing that symbols `110` is a logical `1` bit, and `100` is a logical `0` bit.

Messages are 40 logical bits (120 symbols) and seem to repeat 5 times each signal burst.  Between each message repetition is a delimiter about 8000μs -- a gap around 1640μs, a long pulse around 4830μs, and another gap around 1530μs.

This seems to do a good job decoding messages in the `.cu8` files:

```rtl_433 -s 250k -l 100 -R 0 -X n=DC118W,m=OOK_PWM,s=360,l=720,r=1090,bits=40,invert -r g005_433.92M_250k.cu8```

The only message I see in the `.cu8` files is `{40}000df171cc`  but I'll assume there's 3 different known messages if I invert the messages decoded in the comment @Fredy72 posted above.

```
000df17111 [DC118W - 17 wakeup]
000df171cc [DC118W - 204 learning]
000df17133 [DC118W - 51 wakedown]
```

Using this assumed symbol decoding, the "17 wakeup", "204 learning" and "51 wakedown" descriptions in @Fredy72's comment make sense - the numbers happen to be the decimal representation of the last two hex digits.

Without more info on the device ID and the wind/sun/rain measurements that this sensor reported in the message, I can't guess at the protocol any further.  FWIW, here's the [bitbench](http://triq.net/bitbench?c=000df17111%20%5BDC118W%2017%20wakeup%5D&c=000df171cc%20%5BDC118W%20204%20learning%5D&c=000df17133%20%5BDC118W%2051%20wakedown%5D&f=DATA%3Ahh%20hh%20hh%20hh%20MESSAGE%20TYPE%3Add) I put together.

Fredy72, 2020-03-25:
According to the manual above and my short experience, this device transmits only three states. Learning state of the device for the original gate and then the state open and close for the blind, respectively for the gate that controls it.
Learning starts with the push of a button, just as another button sends the current status. The close and open states are governed by the internal logic of the wind, sun and rain sensors, as indicated in the manual above. The time intervals after which the status is sent are mentioned in the manual.
Same codes (17 and 51) I found in the link [https://github.com/codedealer/pimatic-dooya/blob/master/dooya.coffee](url). On the same page it is described that it is TETRABIT encoding.
So to use the device it is enough to load these three states. Maybe the device can send low battery status, but I haven't tested it.

Fredy72, 2020-03-25:
I tried in real time: rtl_433 -l 100 -X n=DC118W,m=OOK_PWM,s=360,l=720,r=1090,bits=40,invert
But: pulse_FSK_detect(): Maximum number of pulses reached!
I tried: rtl_433 -l 1000 -X n=DC118W,m=OOK_PWM,s=360,l=720,r=1090,bits=40,invert
Shows readable results (every 5 times in a row):
model     : DC118W       count     : 1             num_rows  : 1             rows      : 
len       : 40           data      : 000df17111
codes     : {40}000df17111
Or:
model     : DC118W       count     : 1             num_rows  : 1             rows      : 
len       : 40           data      : 000df171cc
codes     : {40}000df171cc

zuckschwerdt, 2020-03-25:
There might be a misunderstanding (at least I just realized). The "DC118W Wind Sun Rain Sensor" senses the environment but transmits only like a doorbell or garage door opener, right?

A flex decoder .conf file would be best then, you can hardcode the id (house code) as `match=`. Either employ two decoders (one for up, one for down) or map the last bits to "up"/"down" (see the other confs as example).

Fredy72, 2020-03-25:
Yes, according to the manual and my experience with the device, I think it only works as a two state doorbell.
A suitable conf file would really be enough for me.

Fredy72, 2020-03-25:
I just don't fully understand the meaning of -l 1000. Because without that, he doesn't know any signal.

zuckschwerdt, 2020-03-25:
I just recently changed the `-l 1000` to `-Y level=-12` which is the signal level in dB at which a `1` is seen, below that it's a `0` (very broadly speaking).

Using `-M level` will show the detected levels. In your test.c8 I see RSSI -0.1 dB (the 1 is practically maxed out), Noise -34.4 dB (the 0 is nearly silent). This is the ideal case, any level discrimination between, say, -1 and -30 should work here.

The -12 dB is a good default level to pick, the automatic detection should arrive at somewhat the same level. Not sure why it doesn't but I'll investigate.

Fredy72, 2020-03-25:
Thanks.
Command rtl_433 -M level -X n=DC118W,m=OOK_PWM,s=360,l=720,r=1090,bits=40,invert -Y level=-9
work well, but -12 not detect all signal.
Log:
time      : 2020-03-25 20:30:27
model     : DC118W       count     : 1             num_rows  : 1             rows      : 
len       : 40           data      : 000df171cc
codes     : {40}000df171cc
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -2.0 dB      SNR       : 13.3 dB       Noise     : -15.3 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-03-25 20:30:30
model     : DC118W       count     : 1             num_rows  : 1             rows      : 
len       : 40           data      : 000df17111
codes     : {40}000df17111
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -2.1 dB      SNR       : 13.3 dB       Noise     : -15.5 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-03-25 20:30:30
model     : DC118W       count     : 1             num_rows  : 1             rows      : 
len       : 40           data      : 000df17111
codes     : {40}000df17111
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -2.1 dB      SNR       : 13.3 dB       Noise     : -15.5 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-03-25 20:30:30
model     : DC118W       count     : 1             num_rows  : 1             rows      : 
len       : 40           data      : 000df17111
codes     : {40}000df17111
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -2.1 dB      SNR       : 13.3 dB       Noise     : -15.5 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-03-25 20:30:30
model     : DC118W       count     : 1             num_rows  : 1             rows      : 
len       : 40           data      : 000df17111
codes     : {40}000df17111
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -2.1 dB      SNR       : 13.3 dB       Noise     : -15.5 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-03-25 20:30:30
model     : DC118W       count     : 1             num_rows  : 1             rows      : 
len       : 40           data      : 000df17111
codes     : {40}000df17111
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -2.1 dB      SNR       : 13.3 dB       Noise     : -15.5 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-03-25 20:30:32
model     : DC118W       count     : 1             num_rows  : 1             rows      : 
len       : 40           data      : 000df17111
codes     : {40}000df17111
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -2.0 dB      SNR       : 13.6 dB       Noise     : -15.6 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-03-25 20:30:32
model     : DC118W       count     : 1             num_rows  : 1             rows      : 
len       : 40           data      : 000df17111
codes     : {40}000df17111
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -2.0 dB      SNR       : 13.6 dB       Noise     : -15.6 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-03-25 20:30:32
model     : DC118W       count     : 1             num_rows  : 1             rows      : 
len       : 40           data      : 000df17111
codes     : {40}000df17111
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -2.0 dB      SNR       : 13.6 dB       Noise     : -15.6 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-03-25 20:30:32
model     : DC118W       count     : 1             num_rows  : 1             rows      : 
len       : 40           data      : 000df17111
[... 20 lines cut ...]
len       : 40           data      : 000df171cc
codes     : {40}000df171cc
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -2.1 dB      SNR       : 13.1 dB       Noise     : -15.2 dB
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
[... cut ...]
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-03-25 20:30:33
model     : DC118W       count     : 1             num_rows  : 1             rows      : 
len       : 40           data      : 000df171cc
codes     : {40}000df171cc
Modulation: ASK          Freq      : 433.9 MHz
RSSI      : -2.1 dB      SNR       : 13.1 dB       Noise     : -15.2 dB

Fredy72, 2020-03-25:
This work well:
rtl_433 -M level -X n=DC118W,m=OOK_PWM,s=360,l=720,r=1090,bits=40,invert,match={32}0x000df171,get=@32:{8}:event:[204:learning] -Y level=-9
But, this SIGSEGV:
rtl_433 -M level -X n=DC118W,m=OOK_PWM,s=360,l=720,r=1090,bits=40,invert,match={32}0x000df171,get=@32:{8}:event:[204:learning 17:wakeup 51:wakedown] -Y level=-9
Please help me with the parameters.

zuckschwerdt, 2020-03-25:
I don't spot any problems. Can you track where the segv happens, maybe try a debug build?
`cmake -DCMAKE_BUILD_TYPE=Debug .. && make`

Fredy72, 2020-03-25:
Debian 10 (x86_64):
[1.log](https://github.com/merbanan/rtl_433/files/4383283/1.log)
Debian 10 (armv7l):
[2.log](https://github.com/merbanan/rtl_433/files/4383294/2.log)

Fredy72, 2020-03-25:
Finally, I need to remove 5x repetitions. I failed with these parameters: rows, repeats, unique.

klohner, 2020-03-25:
maybe something like this?

```
-X n=DC118W,m=OOK_PWM,s=360,l=720,g=1090,r=11000,bits=40,unique,invert
```

Fredy72, 2020-03-25:
Super! Thank you respectfully.

Fredy72, 2020-03-25:
The device in Home Assistant is running, the states are evaluated according to the code number in it.

CONF:
pulse_detect level=-9
decoder {
    name=DC118W,
    modulation=OOK_PWM,
    short=360,
    long=720,
    reset=11000,
    gap=1090,
    bits=40,
    unique,
    invert,
    get=@0:{32}:id,
    get=@32:{8}:state
}

zuckschwerdt, 2020-03-25:
Thanks for the debug logs! The bug is fixed with 35091bb now.
You need quotes `"` around the `-X` argument :)
`rtl_433 -M level -X "n=DC118W,m=OOK_PWM,s=360,l=720,r=11000,bits=40,unique,invert,match={32}0x000df171,get=@32:{8}:event:[204:learning 17:wakeup 51:wakedown]" -Y level=-9`

Fredy72, 2020-03-25:
Okay. Thank you all.

ErikAndren, 2023-07-27:
Is this the same protocol as #1545 and is to be treated as a duplicate?

zuckschwerdt, 2023-07-27:
> Is this the same protocol as #1545 and is to be treated as a duplicate?

Good find. Thanks.

gdt, 2023-09-29:
I'm going to call this done.  If that's wrong, please test against up-to-date git master and open a new issue or ask for help in a discussion.

closed 2023-09-29 by gdt

nibbles-, 2024-11-20:
Sorry for gravedigging.

I believe I have a device of this kind. Should this now work out of the box or do I have do anything special?
I am currently using the RTL_433 addon for hass and the Mosquitto broker addon.
Default config

I see a bunch of devices but anything that looks like a DC118W

nibbles-, 2024-11-20:
Thanks. Since I am running the addon for Home Assistant I am not quite sure how to load more than one config-file.
So I just copied it to the end of the default file.
It looks like its now recieving the information.
However it doesnt make much sense when I look at it with MQTT Explorer

[image]

Nothing in the message gives any hint that it has wind readings for example. 
Could there be anything else missing?

zuckschwerdt, 2024-11-20:
Pasting to the main conf file is correct.

As commented above
> Yes, according to the manual and my experience with the device, I think it only works as a two state doorbell.

There are no readings, just a virtual "too much wind"-button.

That button=9 is unknown, we so far have only seen: 1:open 3:close 5:stop on the DC1602 and 204:learning 17:wakeup 51:wakedown on the DC118W, as far as I can follow.

Maybe try to force all the possible outputs and write a table of them.

nibbles-, 2024-11-20:
Ah. Missed that part about only being two state.

I think there is more to it then. 
By just restarting RTL_433 i get messages with button 0,2,4,9,"open",14,"stop" & 10

zuckschwerdt, 2024-11-20:
You have a Dooya DC118W, right? It should send 40-bit messages. It seems you are receiving something different, 161 bits is too long. And the messages look like random values.

Try to add `bits<=41,` to the conf.

Also if restarting rtl_433 changes what you receive then something is broken.

nibbles-, 2024-11-20:
I mean I have some form of windreader that is called DC118W. If its Dooya or something else I do not know. It came with the house.
[image]

Nothing espicially different
[image]

nibbles-, 2024-11-20:
The message spam has dropped. Is it the button value that should change depending on state? 
[image]

zuckschwerdt, 2024-11-20:
I guess so. Keep an eye on the ID, it should remain constant for your device. `id` (here 6465 decimal) is the first 24 bit of `data` so `001941` -- you can add a `match=001941,` to the decoder conf to only see that one device.