merbanan/rtl_433 #1338: Hackrf - segmentationfault

loewal, 2020-03-22:
Can someone help....?

rtl_433 -d driver=hackrf -g LNA=24,AMP=14,VGA=32 -M newmodel -l 0 -A -c /usr/local/etc/rtl_433/sonoff_rm433.conf

[INFO] Opening HackRF One #0 223c69dc299f894f...
zsh: segmentation fault  rtl_433 -d driver=hackrf -g LNA=24,AMP=14,VGA=32 -M newmodel -Y level=-12.1 -

zuckschwerdt, 2020-03-22:
Is that a dash on the end, are you reading from stdin? Can you try a debug build:
```
cmake -DCMAKE_BUILD_TYPE=Debug .. && make
```

loewal, 2020-03-22:
This is the output now:

rtl_433 version 20.02-16-g06e8746 branch master at 202003161612 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Registered 123 out of 150 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 ]
[INFO] Opening HackRF One #0 223c69dc299f894f...
AddressSanitizer:DEADLYSIGNAL
=================================================================
==32759==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x7fff6dae2e52 bp 0x7ffee4900da0 sp 0x7ffee4900da0 T0)
==32759==The signal is caused by a READ memory access.
==32759==Hint: address points to the zero page.
    #0 0x7fff6dae2e51 in _platform_strlen (libsystem_platform.dylib:x86_64+0xe51)
    #1 0x10b96a309 in wrap_strlen (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x18309)
    #2 0x10b9156f3 in SoapySDRDevice_setupStream (libSoapySDR.0.8.0.dylib:x86_64+0x1b6f3)
    #3 0x10b4d4f46 in sdr_open_soapy sdr.c:702
    #4 0x10b4d16be in sdr_open sdr.c:808
    #5 0x10b302294 in main rtl_433.c:1497
    #6 0x7fff6d8eccc8 in start (libdyld.dylib:x86_64+0x1acc8)

==32759==Register values:
rax = 0x000000010c60368c  rbx = 0x0000615000005f00  rcx = 0x0000000000000001  rdx = 0x0000000000000001
rdi = 0x0000000000000000  rsi = 0x0000607000000810  rbp = 0x00007ffee4900da0  rsp = 0x00007ffee4900da0
 r8 = 0x0000000000000000   r9 = 0x0000000000000000  r10 = 0x00007fff95642be8  r11 = 0x000000010b9156a0
r12 = 0x000000010c603688  r13 = 0x0000000000000001  r14 = 0x0000000000000001  r15 = 0x0000000000000000
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (libsystem_platform.dylib:x86_64+0xe51) in _platform_strlen
==32759==ABORTING
/usr/local/bin/rtldiag: line 24: 32759 Abort trap: 6           rtl_433 -d driver=hackrf -g LNA=24,AMP=14,VGA=32 -M newmodel -Y level=-12.1 -A -c /usr/local/etc/rtl_433/sonoff_rm433.conf

loewal, 2020-03-22:
> rtl_433 version 20.02-16-g06e8746 branch master at 202003161612 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Registered 123 out of 150 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 ]
[INFO] Opening HackRF One #0 223c69dc299f894f...
AddressSanitizer:DEADLYSIGNAL
=================================================================
==32759==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x7fff6dae2e52 bp 0x7ffee4900da0 sp 0x7ffee4900da0 T0)
==32759==The signal is caused by a READ memory access.
==32759==Hint: address points to the zero page.
    #0 0x7fff6dae2e51 in _platform_strlen (libsystem_platform.dylib:x86_64+0xe51)
    #1 0x10b96a309 in wrap_strlen (libclang_rt.asan_osx_dynamic.dylib:x86_64h+0x18309)
    #2 0x10b9156f3 in SoapySDRDevice_setupStream (libSoapySDR.0.8.0.dylib:x86_64+0x1b6f3)
    #3 0x10b4d4f46 in sdr_open_soapy sdr.c:702
    #4 0x10b4d16be in sdr_open sdr.c:808
    #5 0x10b302294 in main rtl_433.c:1497
    #6 0x7fff6d8eccc8 in start (libdyld.dylib:x86_64+0x1acc8)

==32759==Register values:
rax = 0x000000010c60368c  rbx = 0x0000615000005f00  rcx = 0x0000000000000001  rdx = 0x0000000000000001
rdi = 0x0000000000000000  rsi = 0x0000607000000810  rbp = 0x00007ffee4900da0  rsp = 0x00007ffee4900da0
 r8 = 0x0000000000000000   r9 = 0x0000000000000000  r10 = 0x00007fff95642be8  r11 = 0x000000010b9156a0
r12 = 0x000000010c603688  r13 = 0x0000000000000001  r14 = 0x0000000000000001  r15 = 0x0000000000000000
AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (libsystem_platform.dylib:x86_64+0xe51) in _platform_strlen
==32759==ABORTING
/usr/local/bin/rtldiag: line 24: 32759 Abort trap: 6           rtl_433 -d driver=hackrf -g LNA=24,AMP=14,VGA=32 -M newmodel -Y level=-12.1 -A -c /usr/local/etc/rtl_433/sonoff_rm433.conf

loewal, 2020-03-22:
I also did a sudo make install

zuckschwerdt, 2020-03-22:
Looks like you have SoapySDR 0.7 and also 0.8 installed. Remove one of them and rebuild.

loewal, 2020-03-22:
You're my hero!
Thnx @zuckschwerdt

loewal, 2020-03-22:
However...there is no detection at all.

zuckschwerdt, 2020-03-22:
Did you play with the gain and level settings? Perhaps check with CubicSDR or Gqrx that there is a sufficient signal at that gain and frequency setting. Also likely `-s 2M` will work best.

loewal, 2020-03-22:
Yes, i changed my settings like this...
rtl_433 -d driver=hackrf -g LNA=36,AMP=14,VGA=42 -M newmodel -s 2M -A -c /usr/local/etc/rtl_433/sonoff_rm433.conf
Only 00000000000000000000000000000000.....and so on
Overload?

loewal, 2020-03-22:
and then no more output.
And yes...I changed all gain-settings...
No change...pity

loewal, 2020-03-22:
[image]

loewal, 2020-03-22:
[image]

loewal, 2020-03-22:
sorry! use these values in GQRX, SDRangel and CubicSDR

zuckschwerdt, 2020-03-22:
I never had a HackRF, can't really advice on anything specific. You could try to dump the raw rf with `-w raw_stream.cs16 -T 10s` (for 10 seconds here, just depends on the file size you can handle safely). Then drop on https://triq.org/iqs/ to visualize the "waterfall".

loewal, 2020-03-23:
> and working.....

~/ rtl_433 -d driver=hackrf -g LNA=36,AMP=14,VGA=42 -f 433920000 -Y auto -M newmodel -s 2M -A
rtl_433 version 20.02-16-g06e8746 branch master at 202003161612 inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Reading conf from "rtl_433.conf".
Registered 122 out of 150 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 ]
[INFO] Opening HackRF One #0 223c69dc299f894f...
Sample rate set to 2000000 S/s.
Setting gain element LNA: 36.000000 dB
Setting gain element AMP: 14.000000 dB
Setting gain element VGA: 42.000000 dB
Tuned to 433.920MHz.
Detected OOK package	2020-03-23 07:54:30
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-03-23 07:54:30
model     : GT-WT02      ID Code   : 65
Channel   : 1            Battery   : 0             Temperature: 3.9 C        Humidity  : 55 %          Button    : 0             Integrity : CHECKSUM
Analyzing pulses...
Total count:  240,  width: 883.87 ms		(1767743 S)
Pulse width distribution:
 [ 0] count:  240,  width:  488 us [483;492]	( 977 S)
Gap width distribution:
 [ 0] count:   12,  width: 9036 us [9016;9084]	(18073 S)
 [ 1] count:  137,  width: 2092 us [2074;2190]	(4184 S)
 [ 2] count:   90,  width: 4128 us [4110;4196]	(8255 S)
Pulse period distribution:
 [ 0] count:   12,  width: 9525 us [9507;9574]	(19050 S)
 [ 1] count:  137,  width: 2580 us [2559;2680]	(5161 S)
 [ 2] count:   90,  width: 4616 us [4597;4684]	(9233 S)
Level estimates [high, low]:   2163,    302
RSSI: -17.6 dB SNR: 17.1 dB Noise: -34.7 dB
Frequency offsets [F1, F2]:   -1393,      0	(-42.5 kHz, +0.0 kHz)
Guessing modulation: Pulse Position Modulation with fixed pulse width
Attempting demodulation... short_width: 2092, long_width: 4128, reset_limit: 9084, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PPM,s=2092,l=4128,g=4196,r=9084'
pulse_demod_ppm(): Analyzer Device
bitbuffer:: Number of rows: 13
[00] { 0}                :
[01] {37} 41 80 27 6f 50 : 01000001 10000000 00100111 01101111 01010
[02] { 1} 00             : 0
[03] {37} 41 80 27 6f 50 : 01000001 10000000 00100111 01101111 01010
[04] { 1} 00             : 0
[05] {37} 41 80 27 6f 50 : 01000001 10000000 00100111 01101111 01010
[06] { 1} 00             : 0
[07] {37} 41 80 27 6f 50 : 01000001 10000000 00100111 01101111 01010
[08] { 1} 00             : 0
[09] {37} 41 80 27 6f 50 : 01000001 10000000 00100111 01101111 01010
[10] { 1} 00             : 0
[11] {37} 41 80 27 6f 50 : 01000001 10000000 00100111 01101111 01010
[12] { 0}                :

closed 2020-03-23 by zuckschwerdt