merbanan/rtl_433 #1343: r_dev->verbose does not match cfg->verbosity

x0urc3, 2020-04-09:
- According to the following, verbosity specified at the command line is not the same as the one available to the decoder 
https://github.com/merbanan/rtl_433/blob/0e355dc8973c1bb0b9b4dff68a737ef5945741bd/src/r_api.c#L167-L170

- This is inconsistent with the man page and src/devices/new_template.c
- I could send a PR to fix this but which one is correct?

x0urc3, 2020-04-09:
Grepping src/devices/* shows that the following two usage variation exist
```if (decoder->verbose)```
and
```if (decoder->verbose > 1)```

zuckschwerdt, 2020-04-09:
That's intentional. The first level of verbose increases the app verbosity, the next level the decoder verbosity. Also every decoder has it's own verbosity setting (for use with the future api).

PR #1344 (2020-04-09): Update template guideline for verbosity
Update verbosity guideline for new device templates 

Fixes #1343
closed 2020-04-09 by zuckschwerdt