merbanan/rtl_433 #1351: Messages from Fine Offset WH65B are also decoded (erroneously) as ESIC EMT7110

courtarro, 2020-04-13:
I will be digging deeper into this, but I'm logging all traffic at 915MHz in order to capture my WH65B weather station. These are identified and decoded correctly, but some are also decoded as messages from an ESIC EMT7110 with nonsense payloads. My guess is the protocols overlap enough, along with similar checksum approaches, as to cause one to be confused for the other.

I will gather some burst IQ recordings and post them soon.

Is there a mechanism in place by which one decoder can supersede another from a different file, and thus suppress recognition? I know I can manually disable a decoder, but I assume it's better for things to work without that being necessary.

zuckschwerdt, 2020-04-13:
Decoders are independent. Each decoder gets a fresh bitbuffer built from the pulse data, there is no feedback possible.

Mindavi, 2020-04-14:
Maybe some sanity checks could be added to the decoder, if the values are out of realistic bounds. Otherwise, just disabling the protocol is your best bet.

Either you'll implement it yourself or you can send some samples and I can look into adding some logic to ignore nonsense values. Depends on how nonsense they are exactly, it might not be feasible to do proper sanity checks.

merbanan, 2020-04-14:
@courtarro should be fixed.

closed 2020-04-14 by merbanan