merbanan/rtl_433 #1354: New species with Rubicon protocol: TFA 30.3197

Alf-deb, 2020-04-16:
Since 2003 I am successfully using a remote temperature sensor bought together with the matching receiver TFA 30.3051. Sensor was well supported by rtl_433 with protocol "02" = "Rubiscon Tem,perature Sensor". Now I bought a new sensor from Amazon.de, same Type (TFA 30.3197), manufacturing date 2006. This was immediately recognized by the old receiver and shows correct values.

However I was unable to decode any data with latest rtl_433 build, the sensor was not recognized at all.
So I tried to decode both sensors, old and new one using the "-X" protocol with rubiscon timing and got:

old TFA 30.3197
---------------
```
$ rtl_433 -f 434M -R 0 -X 'n=Rubiscon,m=OOK_PPM,s=1000,l=2000,r=4800,g=3000 repeat=10'

time      : 2020-04-13 15:55:38
model     : Rubiscon     count     : 12            num_rows  : 12            rows      : 
len       : 36           data      : a180e4f3a, 
len       : 36           data      : a180e4f3a, 
len       : 36           data      : a180e4f3a, 
len       : 36           data      : a180e4f3a, 
len       : 36           data      : a180e4f3a, 
len       : 36           data      : a180e4f3a, 
len       : 36           data      : a180e4f3a, 
len       : 36           data      : a180e4f3a, 
len       : 36           data      : a180e4f3a, 
len       : 36           data      : a180e4f3a, 
len       : 36           data      : a180e4f3a, 
len       : 37           data      : a180e4f3a0
codes     : {36}a180e4f3a, {36}a180e4f3a, {36}a180e4f3a, {36}a180e4f3a, {36}a180e4f3a, {36}a180e4f3a, {36}a180e4f3a, {36}a180e4f3a, {36}a180e4f3a, {36}a180e4f3a, {36}a180e4f3a, {37}a180e4f3a0
```
Reading is correct:
nibble 3+4+5 => 0e4 => hex2dec = 228 = 22.8°C

new TFA 30.3197
---------------
```
rtl_433 -f 434M -R 0 -X 'n=TFA 3197,m=OOK_PPM,s=1000,l=2000,r=4800,g=3000 repeat=8'

time      : 2020-04-13 17:11:26
model     : TFA 3197     count     : 13            num_rows  : 13            rows      : 
len       : 0            data      : , 
len       : 37           data      : 5680caf400, 
len       : 37           data      : 5680caf400, 
len       : 37           data      : 5680caf400, 
len       : 37           data      : 5680caf400, 
len       : 37           data      : 5680caf400, 
len       : 37           data      : 5680caf400, 
len       : 37           data      : 5680caf400, 
len       : 37           data      : 5680caf400, 
len       : 37           data      : 5680caf400, 
len       : 37           data      : 5680caf400, 
len       : 37           data      : 5680caf400, 
len       : 36           data      : 5680caf40
codes     : {0}, {37}5680caf400, {37}5680caf400, {37}5680caf400, {37}5680caf400, {37}5680caf400, {37}5680caf400, {37}5680caf400, {37}5680caf400, {37}5680caf400, {37}5680caf400, {37}5680caf400, {36}5680caf40
```

Reading is correct:
nibble 3+4+5 => 0ca => hex2dec = 202 = 20.2°C

Now I had a look at the source code "rubiscon.c" and "nexus.c" and found a usefull comment about the similarity of those 2 protocols. As I do not have any skills in C-programming I just tried to reverse the crc-check condition in "nexus.c" by modifying line 58 and removing the exclamation mark at the beginning.
from:
`!rubicson_crc_check(bb)) {
`
into:
`rubicson_crc_check(bb)) {`

And voila, now my new TFA 30.3197 was successfully decoded by the protocol "19" = "Nexus-TH"!
The only minor difference to the old sensor: temperature is displayd with 2 decimal digits (20.20 instead of 20.2) and Humidity values are trash, can be any from 1% to 254%, so there seems to be no sensor attached to an unterminated input line of the hardware.

My guess: Rubiscon-protocol is still used, but as the  original hardware dated 2003 is no longer available they now use the same hardware as Nexus-sensors do and flash the Rubiscon-firmware on it. So protocol is a sort of "Rubiscon Generation 2".
The only drawback was that now with the modified "nexus.c" I got 3-4 devices - which luckily have different ID's (House Codes) so I can distinguish them - mine has "70". To just grab the temperature value for further processing I use following bash command:

```
rtl_433 -R 19 -T 130 -F csv 2>/dev/null \
| awk -F , '/Nexus-TH,70/ {print substr($8, 1, length($8)-2); exit}'
```

It would be great if for this class of sensors a separate protocol were made available to avoid my dirty hack.
And, as a general new feature for rtl_433 in situations of more than 1 sensor with same protocol a selection by "ID" (the Option "-E 1" is useless in that case).

Kind regards and thanks for this uesefull tool,
Ingo

zuckschwerdt, 2020-04-17:
Interesting. Can you capture a few samples and add a zip with those here? The hack with the `!` just turns the check off ;) We need to find something else. I'll have a look.

Generally rtl_433 is not a one-shot receiver but should run as a server. If you only need single (last good) value perhaps modify one of the example scripts to catch that.

Alf-deb, 2020-04-17:
I did capure already to see the difference between old and new TFA 30.3197.
I do attach them, they had been recorded with gqrx using a 4 kHz bandwidth filter (+/-2kHz around center frequency). Samples show AM demodulated and i/Q signal. Signal/noise ratio was excellent and better than 30db. Filenames are self-explaining.

If you need more help/info, tests of code - let me know.

Sorry - the forum software here does not accept any of my archives. I tried zip, tar.gz, tried to avoid special characters, tried the contained wav-files, .. without success. Always tells me:
"Something went really wrong, and we can't process that file"
Could you supply an e-mail address for me?

Alf-deb, 2020-04-17:
> Interesting. Can you capture a few samples and add a zip with those
> here? The hack with the |!| just turns the check off ;) We need to find
> something else. I'll have a look.

Not just something else, without the hack my sensor is not recognized at
all!

I am now trying again wit the archive as attachment.

Alf-deb, 2020-04-17:
[samples.zip](https://github.com/merbanan/rtl_433/files/4494899/samples.zip)

Alf-deb, 2020-04-17:
Finally I got it accept my zip.

ALEEF02, 2020-04-24:
@Alf-deb If everything is working for you, may I suggest closing this issue?

zuckschwerdt, 2020-04-24:
Sorry, I didn't have time to dig into this. Still needs work ;)

zuckschwerdt, 2020-04-24:
@Alf-deb can you collect more valid codes like `{37}5680caf400`. Perhaps a list of 20 or more good codes?

Alf-deb, 2020-04-25:
Am 24.04.20 um 22:25 schrieb Christian W. Zuckschwerdt:
> @Alf-deb <https://github.com/Alf-deb> can you collect more valid codes
> like |{37}5680caf400|. Perhaps a list of 20 or more good codes?

I suppose you only need 1 single 37-bit out of the complete datagramm
consistiong of 1 0-bit row, 11 37-bit rows  and 1 36-bit row for the new
TFA-30.3197 sensor, right?

zuckschwerdt, 2020-04-25:
Yes. Just pick one good code per transmission.

Alf-deb, 2020-04-25:
So, here the samples come.
Remark: the first nibble has changed from formerly 0x5 to 0x4 for some reason, maybe because of a battery change? The new TFA-30.3197 has been mounted already outside instead of the old one because it delivers a significantly stronger signal (hardware design/antenna?) So I needed to verify to capture the right one and placed an ice-bag on it for some time to cool it artficially down. All temperature values in °C are also verified against the paired commercial receiver.

```
captured code    °C
{37}468065f1e0 10.1
{37}468063f440  9.0
{37}468064fea0 10.0
{37}468068f5e0 10.4
{37}468070f070 11.2
{37}468079fc40 12.1
{37}468082fc60 13.0
{37}468088f280 13.6
{37}46809bf6b0 15.5
{37}46809ef1c0 15.8
{37}4680a0fc30 16.0
{37}4680a2f1a0 16.2
{37}4680a4f400 16.4
{37}4680a5fb40 16.5
{37}4680a7f6d0 16.7
{37}4680a8ff40 16.8
{37}4680aaf2d0 17.0
{37}4680adf830 17.3
{37}4680aff5a0 17.5
{37}4680b0fad0 17.6
{37}4680b1f590 17.7
{37}4680b3f800 16.9
{37}4680b4f2e0 17.0
{37}4680b5fda0 18.1
{37}4680b7f030 18.3
{37}4680bbfb70 18.7
{37}4680baf430 18.6  
```
Command used for capturing:
`rtl_433 -f 434M -R 0 -X 'n=TFA 3197,m=OOK_PPM,s=1000,l=2000,r=4800,g=3000 repeat=8' | grep  'codes     : {0}, {37}46'`

As you know, the humidity readings are trash, that sensor does not support humidity.
For completeness I do attach also 2 screenshots from a ncurses x-term showing the output of just
`rtl_433`
for the old _Rubiscon_ sensor and the new _TFA-30.3197_ with my patched _nexus.c_.

Additionally a live plot as I record it by sampling every 15 minutes and plotting 3 days using gnuplot-nox as a vector graphic

[image]

[image]
[image]

Alf-deb, 2020-04-25:
Additional information I just discovered:
With my patched nexus.c this protocol also catches and decodes my old Rubiscon sensor (ID = House Code 230) - see screenshot:
[image]

zuckschwerdt, 2020-05-15:
Some news: The TFA 30.3051 sensor checksum looks like a reverse Galois LFSR (byte reflect, gen 31, if you prepend a 0-nibble: key 43 final XOR 82 using xor). I'll add a decoder soon.

zuckschwerdt, 2020-05-15:
If I try the code you supplied in the original posting above `5680caf40` it works as Rubicson:
`rtl_433 -y '5680caf40/5680caf40/5680caf40'`.
Turns out it was the first incomplete row of `len : 0` that broke things. It's now fixed.

Capture a sample of your new TFA 30.3051 sensor and upload here if you like to make sure we check for regressions. Use `-S all` to capture.

closed 2020-05-15 by zuckschwerdt

Alf-deb, 2020-05-16:
I immediately tried your fixes/patches and can confirm: It works!

My "old" TFA 30.3197 continues to work as before without any problems
using "Rubiscon" (protocol 2).

My "new" TFA 30.3197 is now recognized as "Solight-TE44" (protocol 85).

Within my neighborhood I did not find "Nexus-TH" sensors any more which
appears to be correct, because before those did report invalid/trash
humidity values.

I have one remaining cosmetic thing for "Solight-TE44" now:
The reported temperature value is displayed with 2 decimal digits what
is one magnitude beyond any reasonable accuracy and what the sensor
actually reports. Please change (I suppose line 75 in solight_te44.c)

Battery status is no longer reported - don't know whether it is actually
monitored.

Naming of "House Code" has changed into "Id" - seems the new policy?

That's all, many thanks for your good work,
Ingo

or unsubscribe

zuckschwerdt, 2020-05-16:
I noted the support for "TFA 30.3197" in both decoders.

The temperature reading has two digits to get more accuracy when using the Fahrenheit conversion, I guess.
The label "House code" is only for the default display, with JSON or CSV it should be "id" always.
I'll add battery perhaps later, after testing.

Thanks for the feedback.