merbanan/rtl_433 #1356: Acurite 6045M Lightning detector with rtl_433

MksRasp, 2020-04-19:
with the latest update of rtl_433 the Acurite 6045M Lightning detector is not parsed but the 5in1 station is parsed correctly, I had to go back to the old rtl_433 and everything back to normal and weewx is working again.
I followed this guide for installation
https://github.com/matthewwall/weewx-sdr
all went well except rtl_433 which broke the weewx operation but going back to old version of rtl_433 fixed the issue.

zuckschwerdt, 2020-04-19:
The PR is likely #1348 
@rct can you look into this?

rct, 2020-04-19:
Yes, sorry the last change #1348 to the 6045 decoder requires an update to weewx-sdr as the fields have changed, they were being decoded incorrectly.

Last week I opened [an issue and posted the updated output](https://github.com/matthewwall/weewx-sdr/issues/81) against weewx-sdr but it doesn't look like anyone has had a chance to work on that.  It should be a simple change, but I'm not currently running weewx-sdr so I didn't attempt it.

Last week I also posted a heads up to the [rtl_433 google group](https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/rtl_433/gvTUFWJlFxE/S3rQdM6zDwAJ)

@zuckschwerdt - I don't think this is an rtl_433 issue, so I think you can probably

rct, 2020-04-19:
A quick look at sdr.py, it might be as simple as [removing references to `ussb1` around line 787](https://github.com/matthewwall/weewx-sdr/blob/master/bin/user/sdr.py#L787) - that's the field that went away.

closed 2020-04-19 by zuckschwerdt

MksRasp, 2020-04-20:
Thank you for your support
I edited the sdr.py and weewx.conf made these changes
1-changed the identifier from "Acurite Lightning 6045M" to "Acurite-6045M"
2-Added "V2" to the end of sensor map in weewx.conf
I am not sure about what I did but the weewx is working now with the recent update to rtl_433.