merbanan/rtl_433 #1360: maverick et732 not working

gregvis, 2020-04-21:
Hello,

I'm trying to read the temperatures from a maverick et732 (should be protocol 71), but i'm not getting any results. 

I've taken some captures of the data and have tried analyzing using the settings from the maverick.c file
` rtl_433 -R 0 -X 'name=et732,m=OOK_MC_ZEROBIT,s=230,r=4000,bits=104'`
but I don't think the decoding is correct because i'm getting weird values like this:
`time      : @0.323164s
model     : test         count     : 1             num_rows  : 1             rows      :
len       : 104          data      : 00222002202222022000802020
codes     : {104}00222002202222022000802020
`

I have attached a few captures where my 2 temps should be in the range of 70-76 degrees
[maverick_et732.zip](https://github.com/merbanan/rtl_433/files/4512199/maverick_et732.zip)

Mindavi, 2020-04-24:
Haven't looked to deep into this, but seems like we don't have any et-732 samples in the rtl_433_tests library, so maybe the decoder is lying about it being able to decode et-732 samples. The preamble is nothing like we expect in the decoder, as far as I can see.

I'm afraid the conclusion will be that it's not supported, but it would be at least a good idea to compare the samples (rtl_433_tests et-733 samples vs these samples) in e.g. pulseview or https://triq.org/iqs/ to see if there's any resemblance between them.

klohner, 2020-04-30:
It looks like the actual signals are a little outside  433.92 MHz and `rtl_433` is picking up a harmonic instead of the center frequency of the signal and its automatic level detection isn't able to lock in. By manually specifying a detection level (`-Y level=-12` in this case), all these sample files decoded successfully.

@gregvis I'd recommend you play around with `-f` frequency values a little above/below 433.92M to find the one your device is actually using.  By doing this, `rtl_433` should be able to decode your signals using its automatic level detection.

```
$ rtl_433 -f 250k -r g010_433.92M_250k.cu8 -X n=et732,m=OOK_MC_ZEROBIT,s=244,r=4000,bits=104 -Y level=-12

time      : @0.325508s
model     : et732        count     : 1             num_rows  : 1             rows      :
len       : 104          data      : 55666aa66a6966a66a59a6a556
codes     : {104}55666aa66a6966a66a59a6a556

time      : @0.325508s
model     : et732        count     : 1             num_rows  : 1             rows      :
len       : 104          data      : 55666aa66a6966a66a59a6a556
codes     : {104}55666aa66a6966a66a59a6a556

time      : @0.325508s
model     : Maverick-ET73x                         Session_ID: 20484
Status    : default      TemperatureSensor1: 18.00 C                         TemperatureSensor2: 20.00 C

time      : @0.325508s
model     : Maverick-ET73x                         Session_ID: 20484
Status    : default      TemperatureSensor1: 18.00 C                         TemperatureSensor2: 20.00 C

$ rtl_433 -f 250k -r g009_433.92M_250k.cu8 -Y level=-12

time      : @0.477804s
model     : Maverick-ET73x                         Session_ID: 20484
Status    : init         TemperatureSensor1: 18.00 C                         TemperatureSensor2: 20.00 C

$ rtl_433 -f 250k -r g010_433.92M_250k.cu8 -Y level=-12

time      : @0.325508s
model     : Maverick-ET73x                         Session_ID: 20484
Status    : default      TemperatureSensor1: 18.00 C                         TemperatureSensor2: 20.00 C

time      : @0.325508s
model     : Maverick-ET73x                         Session_ID: 20484
Status    : default      TemperatureSensor1: 18.00 C                         TemperatureSensor2: 20.00 C

$ rtl_433 -f 250k -r g011_433.92M_250k.cu8 -Y level=-12

time      : @0.517556s
model     : Maverick-ET73x                         Session_ID: 20484
Status    : default      TemperatureSensor1: 18.00 C                         TemperatureSensor2: 20.00 C

$ rtl_433 -f 250k -r g015_433.92M_250k.cu8 -Y level=-12

time      : @0.573392s
model     : Maverick-ET73x                         Session_ID: 20484
Status    : default      TemperatureSensor1: 18.00 C                         TemperatureSensor2: 20.00 C

$ rtl_433 -f 250k -r g016_433.92M_250k.cu8 -Y level=-12

time      : @0.507356s
model     : Maverick-ET73x                         Session_ID: 20484
Status    : default      TemperatureSensor1: 18.00 C                         TemperatureSensor2: 20.00 C

$ rtl_433 -f 250k -r g017_433.92M_250k.cu8 -Y level=-12

time      : @0.329228s
model     : Maverick-ET73x                         Session_ID: 20484
Status    : default      TemperatureSensor1: 18.00 C                         TemperatureSensor2: 20.00 C

time      : @0.329228s
model     : Maverick-ET73x                         Session_ID: 20484
Status    : default      TemperatureSensor1: 18.00 C                         TemperatureSensor2: 20.00 C

$ rtl_433 -f 250k -r g018_433.92M_250k.cu8 -Y level=-12

time      : @0.546964s
model     : Maverick-ET73x                         Session_ID: 20484
Status    : default      TemperatureSensor1: 19.00 C                         TemperatureSensor2: 20.00 C

time      : @0.546964s
model     : Maverick-ET73x                         Session_ID: 20484
Status    : default      TemperatureSensor1: 19.00 C                         TemperatureSensor2: 20.00 C

$ rtl_433 -f 250k -r g020_433.92M_250k.cu8 -Y level=-12

time      : @0.559400s
model     : Maverick-ET73x                         Session_ID: 20484
Status    : default      TemperatureSensor1: 21.00 C                         TemperatureSensor2: 20.00 C

$ rtl_433 -f 250k -r g022_433.92M_250k.cu8 -Y level=-12

time      : @0.397852s
model     : Maverick-ET73x                         Session_ID: 20484
Status    : default      TemperatureSensor1: 21.00 C                         TemperatureSensor2: 20.00 C

$ rtl_433 -f 250k -r g023_433.92M_250k.cu8 -Y level=-12

time      : @0.517312s
model     : Maverick-ET73x                         Session_ID: 20484
Status    : default      TemperatureSensor1: 22.00 C                         TemperatureSensor2: 20.00 C

time      : @0.517312s
model     : Maverick-ET73x                         Session_ID: 20484
Status    : default      TemperatureSensor1: 22.00 C                         TemperatureSensor2: 20.00 C
```

gregvis, 2020-04-30:
@klohner  Thanks for the suggestion to change the frequency, that ultimately solved the problem. 

I fired up Gqrx to see at what frequency the signal was the strongest. Turns out it was at 433.776 mhz. Plugging this into rtl_433 I was able to finally get temperature readings.

`~$ rtl_433 -f 433.776m
rtl_433 version 20.02-28-gf8d9a01 branch master at 202004202153 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/greg/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 122 out of 151 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.776MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-04-30 12:04:39
model     : Maverick-ET73x                         Session_ID: 8947
Status    : default      TemperatureSensor1: 19.00 C                         TemperatureSensor2: 20.00 C`

closed 2020-04-30 by gregvis

dennyreiter, 2020-04-30:
Thank you!  I just received my Maverick Tuesday and was struggling with it.

zuckschwerdt, 2020-05-01:
In most cases it's best not to hit the sender frequency dead-on. E.g if the sender is at 433.776M then use `rtl_433 -f 433.75M` or `rtl_433 -f 433.7M`.