merbanan/rtl_433 #1363: RSSI, SNR, Noise, and Freq all 0

ALEEF02, 2020-04-24:
Hey guys,
I'm trying to use rtl_433 to intercept the packets from my weather devices. I am trying to best position my antenna, so I activated the `-M level` to see the SNR. But, all the SNRs are 0, along with RSSI, Noise, and Freq. Is this an issue on my end? Thanks 👍 
[image]

zuckschwerdt, 2020-04-24:
There should be no code path where all those values are `0`. What version are you running?

ALEEF02, 2020-04-24:
I am on commit https://github.com/merbanan/rtl_433/commit/c47bef0815e5240253236c023f21deeff5979058

ALEEF02, 2020-04-24:
Further testing shows that this seems to only be happening with 915MHz, not 433.92MHz @zuckschwerdt 
[image]

merbanan, 2020-04-24:
Can you try adding -Y classic -s 1000k ?

ALEEF02, 2020-04-24:
@merbanan That's working!
[image]

merbanan, 2020-04-24:
The new fsk demod code is missing some calculations.

ALEEF02, 2020-04-28:
Is there a fix on the way, or should I continue to use `-Y classic -s 1000k` and close this issue?

zuckschwerdt, 2020-04-28:
We do want to have this fixed ;)

zuckschwerdt, 2020-04-29:
It's fixed. Thanks for the detailed report!

commit 6135a23d (2020-04-29): Fix missing levels with minmax demod (fixes #1363)
closed 2020-04-29 by zuckschwerdt