merbanan/rtl_433 #1365: Doesn't decode on 64bit Raspian Kernel

czechmark, 2020-04-25:
Tried to upgrade my raspberry pi to the 64 bit kernel.  Everything except rtl_433 seemed to work.  Then I rebooted with the 32 bit kernel (didn't change anything else), and the system worked again.  I am happy running the 32 bit, so not a bit issue just thought you might like to know.

zuckschwerdt, 2020-04-25:
That's just the kernel? And your userland ist still the same, 32 bit? Strange, maybe something with libusb?

czechmark, 2020-04-25:
Yes it was working, I did some package upgrade (apt update apt upgrade...), put 2 lines in /boot/config.txt to boot the 64 bit kernel... Went and ate dinner, then noticed that 433 wasn't working.. played around a bit including downloading the latest rt-433, rebooted everything, it still wasn't working.  When I woke up in the morning, I thought maybe its the 32 vs 64, commented out the 2 lines I put in /boot/config/txt and rebooted and everything was working... (I automatically start rt-433 via crontab), so I am pretty sure it is just the kernel that is different.  all the libs are the same.

rct, 2020-04-25:
If you haven't already, you might want to check with the 64 bit kernel any chance the rtl dvb drivers got loaded? Also, probably worth checking if some of the basic rtl_*  stuff works.

zuckschwerdt, 2020-05-25:
This is likely specific to that kernel or the environment. Please update with more info (e.g. are other sdr tools working?), reopen if something can be fixed in rtl_433.

closed 2020-05-25 by zuckschwerdt