merbanan/rtl_433 #1366: Manual gain settings unable to find signals that work fine in "auto"

courtarro, 2020-04-26:
I have several strong signals nearby from a weather station at 915M. With auto gain, I have no trouble receiving them:

    rtl_433 -M level -Y classic -f 915000000 -s 250000

I use classic mode because I want the SNR levels. This works fine and yields packets with an SNR of about 30dB. However, when I manually set the gain to any value:

    rtl_433 -M level -Y classic -f 915000000 -s 250000 -g 15

I receive **no packets**. I tried gains `[1.4, 8.7, 15.7, 20.7, 25.4, 32.8, 36.4, 42.1, 49.6]`. Is there something about auto gain mode that changes when the gain is manually set? I notice that the RTL-SDR architecture often includes multiple gain levels and AGCs. Are these possibly the reason I'm having this problem?

My main reason for wanting to set the gain manually is avoiding variation in when and how the signals are received, since the transmitter and receiver have fixed position. I don't want other nearby signals to affect the gain levels and potentially disrupt my signal of interest.

zuckschwerdt, 2020-04-29:
AFAIK, it should work. Maybe you can record short samples with auto and some other gains and compare. Use e.g. `rtl_433 -M level -Y classic -f 915M -s 250k -g 15 -w test-gain15.cu8 -T 10s` and https://triq.org/iqs/

theasp, 2020-05-10:
Neat site!  I also can't capture anything when setting the gain. I've captured 0db, 15db and 40db and 49.6db here:
https://industrial.gt0.ca/~asp/rtl_443/

0db is quite a bit stronger than 49.6db, so perhaps it turns the AGC on?

merbanan, 2020-05-10:
0=auto

theasp, 2020-05-10:
Yes, but then how do you control the gain independently of having AGC turned on?

merbanan, 2020-05-10:
By using any values beside 0.

merbanan, 2020-05-10:
The gain used is written in the command line output.

theasp, 2020-05-10:
Ok, if I want a gain of 20, and AGC turned on, how can I do that?

merbanan, 2020-05-10:
You can't. Either fixed gain or AGC.

theasp, 2020-05-10:
With AGC on, what gain will be used?

theasp, 2020-05-10:
I'm asking, because they can be set independently in other software.

merbanan, 2020-05-10:
AGC = automatic gain control. So the actual gain used is unknown.

merbanan, 2020-05-10:
Well the AGC is a property of the Realtek chip. The gain is a property of the tuner chip. AFAIK the AGC will override the manual gain but that might not correlate well with reality.

Exactly how all the gain stages work and how the AGC interacts with them is unknown to me. We have had this logic since the start of the project I think.

theasp, 2020-05-10:
With sdrangel, this is what the waterfall looks like with the gain set to 25db:
[image]

25db with AGC:
[image]

40db with AGC:
[image]

There may be a way to control them independently using using a SoapySDR device, which should do what @courtarro was looking for.

theasp, 2020-05-10:
This should do it if rtl_443 is compiled with SoapySDR support:
```
rtl_433 -d driver=rtlsdr -t digital_agc=true -g TUNER=30
```

gdt, 2023-09-25:
What's the status of this issue?  Is there a bug?

ksaye, 2023-09-30:
I see the same issue.  When using autogain, it sees packets just fine.  When setting the gain "-g 1" through "-g 50", I do not see packets.

I think I am seeing autogain change, as I guess it should, in normal operation.  When autogain is set, over time it stops picking up wanted packets (reading the water meter, protocol 149) and I have to restart rtl_433 to see them again.  My hopes are that I can find a desired setting and force it and not have to restart.

gdt, 2023-09-30:
@ksaye Are you using recent git master?  What kind of tuner?   Have you tried -20 through 0 (not really sure; there is apparently negative gain, and running with lots of -v and all -M options is advisable).

ksaye, 2023-10-02:
> @ksaye Are you using recent git master? What kind of tuner? Have you tried -20 through 0 (not really sure; there is apparently negative gain, and running with lots of -v and all -M options is advisable).

I am running version 20.11-1, from the package store on a Pi3:

      apt list --installed | grep rtl
      
      WARNING: apt does not have a stable CLI interface. Use with caution in scripts.
      
      librtlsdr0/stable,now 0.6.0-3 armhf [installed,automatic]
      rtl-433/stable,now 20.11-1 armhf [installed]
      soapysdr0.7-module-rtlsdr/stable,now 0.3.0-2 armhf [installed,automatic]

The tuner is just an inexpensive USB RTL2838:

      trying device  0:  Realtek, RTL2838UHIDIR, SN: 915

Trying the command: "rtl_433 -g -20 -F json -M level -R 149 -f 914M" I get no messages.

      rtl_433 -g -20 -F json -M level -R 149 -f 914M
      rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
      Use -h for usage help and see https://triq.org/ for documentation.
      Trying conf file at "rtl_433.conf"...
      Trying conf file at "/home/ksaye/.config/rtl_433/rtl_433.conf"...
      Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
      Trying conf file at "/etc/rtl_433/rtl_433.conf"...
      
      New defaults active, use "-Y classic -s 250k" for the old defaults!
      
      Registered 1 out of 175 device decoding protocols [ 149 ]
      Found Rafael Micro R820T tuner
      Exact sample rate is: 1000000.026491 Hz
      [R82XX] PLL not locked!
      Sample rate set to 1000000 S/s.
      Tuner gain set to 0.000000 dB.
      Tuned to 914.000MHz.
      Allocating 15 zero-copy buffers

Based on your comments, I have tried (in Docker) to build it from GitHub and running the command:

      rtl_433 version nightly-1-g72c0f21b branch master at 202309292256 inputs file rtl_tcp RTL-SDR with TLS
      rtl_433 -F json -g 20 -M level -R 149 -f 914M

I am able to see some messages, so perhaps the issue is only with the build in the package store?

      2023-10-02 11:36:26.157750 Running command rtl_433 -F json -g 20 -M level -R 149 -f 914M
      2023-10-02 11:37:34.043721 {"time" : "2023-10-02 11:37:33", "model" : "ERT-SCM", "id" : 26732440, "physical_tamper" : 0, "ert_type" : 13, "encoder_tamper" : 2, "consumption_data" : 126575, "mic" : "CRC", "mod" : "ASK", "freq" : 914.017, "rssi" : -11.392, "snr" : 21.209, "noise" : -32.602}

I will continue to monitor it with the "-g 20" setting, and assuming all goes well, I will withdrawal my question.

gdt, 2023-10-03:
Thanks.  20.11 is too old and how it behaves is out of scope for the tracker, so building and testing with recent is indeed the right approach.

ksaye, 2023-10-03:
@gdt, when using autogain, is there anyway to read the current gain setting from the R820T?

gdt, 2023-10-04:
I don't know.  I'd suggest reading the pgogramming documention for the R820T.

zuckschwerdt, 2023-10-04:
> when using autogain, is there anyway to read the current gain setting from the R820T?

Solid no.

If you get to try a newer version of rtl_433 note that there is now also `-Y autolevel` which will help to read signals at what would usually be too low gain settings.

We can't really help with RTL-SDR gain settings. But note that automatic gain (gain=0) is not the same as AGC (automatic gain control), which might additionally be needed (`-t digital_agc=1`).

If you try different gains then record samples (`-S all)` or better the channel (e.g. `-w myfile_433.92M_250k.cu8 -T 20`) and compare the waterfall on https://triq.org/pdv/

closed 2023-10-04 by zuckschwerdt

ksaye, 2023-10-04:
> Solid no.
>
> If you get to try a newer version of rtl_433 note that there is now also `-Y autolevel` which will help to read signals at what would usually be too low gain settings.
>
> We can't really [...]

Thanks for the clarity and I do agree that this issue should be closed, since it has been proven that when using the '-g 20' switch I can see messages when using a build from this repo.