merbanan/rtl_433 #1367: Oregon Scientific temp sensor with two sensor_id's?

Wilm0r, 2020-04-26:
I've got a PCR338 that isn't supported by rtl_433 yet. While trying to add support for it I noticed unrecognised messages looking like this:

```
f0 24 26 61 00 10 00 8d 2f a0
```

Checksum is consistently (across slightly different messages) correct for 15 nibbles. Then I realised it contains the exact same temperature transmitted by the thermal sensor that came with the same set (RGR126NX). (Which BTW seems to do temperature only, humidity field is present but always 0.)

Have you seen this before, one sensor ~randomly transmitting the same data with different sensor IDs? (rtl_433 is picking up the ones with f824 only so far, the above is f024)

I could just send in a PR adding this as an alias but I wonder first whether folks more familiar with this code know what this may mean.

Wilm0r, 2020-04-26:
Update: Looks to me like it's 50/50 between the two sensor_id's. Doesn't look like it in the dump below but note the timestamps: Looks like rtl_433 is double-parsing blocks quite regularly. 
```json
{"time" : "2020-04-26 12:54:12", "brand" : "OS", "model" : "Oregon-THN388", "id" : 102, "channel" : 2, "battery_ok" : 1, "temperature_C" : 14.700}
{"time" : "2020-04-26 12:55:11", "brand" : "OS", "model" : "Oregon-THGR810", "id" : 102, "channel" : 2, "battery_ok" : 1, "temperature_C" : 14.700, "humidity" : 0}
{"time" : "2020-04-26 12:55:11", "brand" : "OS", "model" : "Oregon-THGR810", "id" : 102, "channel" : 2, "battery_ok" : 1, "temperature_C" : 14.700, "humidity" : 0}
{"time" : "2020-04-26 12:56:10", "brand" : "OS", "model" : "Oregon-THN388", "id" : 102, "channel" : 2, "battery_ok" : 1, "temperature_C" : 14.700}
{"time" : "2020-04-26 12:57:09", "brand" : "OS", "model" : "Oregon-THGR810", "id" : 102, "channel" : 2, "battery_ok" : 1, "temperature_C" : 14.700, "humidity" : 0}
{"time" : "2020-04-26 12:57:09", "brand" : "OS", "model" : "Oregon-THGR810", "id" : 102, "channel" : 2, "battery_ok" : 1, "temperature_C" : 14.700, "humidity" : 0}
{"time" : "2020-04-26 12:58:08", "brand" : "OS", "model" : "Oregon-THN388", "id" : 102, "channel" : 2, "battery_ok" : 1, "temperature_C" : 14.700}
{"time" : "2020-04-26 12:59:07", "brand" : "OS", "model" : "Oregon-THGR810", "id" : 102, "channel" : 2, "battery_ok" : 1, "temperature_C" : 14.800, "humidity" : 0}
{"time" : "2020-04-26 13:00:06", "brand" : "OS", "model" : "Oregon-THN388", "id" : 102, "channel" : 2, "battery_ok" : 1, "temperature_C" : 14.800}
{"time" : "2020-04-26 13:00:06", "brand" : "OS", "model" : "Oregon-THN388", "id" : 102, "channel" : 2, "battery_ok" : 1, "temperature_C" : 14.800}
{"time" : "2020-04-26 13:03:03", "brand" : "OS", "model" : "Oregon-THGR810", "id" : 102, "channel" : 2, "battery_ok" : 1, "temperature_C" : 14.900, "humidity" : 0}
{"time" : "2020-04-26 13:04:02", "brand" : "OS", "model" : "Oregon-THN388", "id" : 102, "channel" : 2, "battery_ok" : 1, "temperature_C" : 15.000}
{"time" : "2020-04-26 13:04:02", "brand" : "OS", "model" : "Oregon-THN388", "id" : 102, "channel" : 2, "battery_ok" : 1, "temperature_C" : 15.000}
{"time" : "2020-04-26 13:05:01", "brand" : "OS", "model" : "Oregon-THGR810", "id" : 102, "channel" : 2, "battery_ok" : 1, "temperature_C" : 15.000, "humidity" : 0}
{"time" : "2020-04-26 13:06:00", "brand" : "OS", "model" : "Oregon-THN388", "id" : 102, "channel" : 2, "battery_ok" : 1, "temperature_C" : 15.000}
{"time" : "2020-04-26 13:06:00", "brand" : "OS", "model" : "Oregon-THN388", "id" : 102, "channel" : 2, "battery_ok" : 1, "temperature_C" : 15.000}
{"time" : "2020-04-26 13:06:59", "brand" : "OS", "model" : "Oregon-THGR810", "id" : 102, "channel" : 2, "battery_ok" : 1, "temperature_C" : 15.000, "humidity" : 0}
{"time" : "2020-04-26 13:07:58", "brand" : "OS", "model" : "Oregon-THN388", "id" : 102, "channel" : 2, "battery_ok" : 1, "temperature_C" : 15.100}
{"time" : "2020-04-26 13:07:58", "brand" : "OS", "model" : "Oregon-THN388", "id" : 102, "channel" : 2, "battery_ok" : 1, "temperature_C" : 15.100}
```
Not sure how exactly you want this integrated (with a new model number or not? Seems confusing to do so even though more correct) but if I hear nothing I may just send a PR later. :-)

zuckschwerdt, 2020-04-26:
There are some security-type sensors that send "old protocol" interleaved with "new protocol". In one case even betraying the added encrpytion of the new protocol ;)
But for simple weather sensors I've never seen any kind of two-protocol-compat.

zuckschwerdt, 2020-04-26:
I just realize that this is a OS quirk, some model types do not use the whole 16 bit as type id. So `id & 0xf0ff == f024` might actually be needed check. Not sure what the extra 4 bit encode.

Wilm0r, 2020-04-27:
That's intriguing. :-/ So to me it seems quite reliably interleaved at least, for a very much unknown purpose. No other bits in that nibble getting set randomly.

zuckschwerdt, 2020-04-27:
Just a suggestion, I didn't think it through ;) Really just [duck testing](https://en.wikipedia.org/wiki/Duck_test) based on payload/output.

gdt, 2023-09-25:
@Wilm0r Can you test with latest git master and freshen the debugging info?  Or should we be closing?