merbanan/rtl_433 #1368: Bugs in documentation

OgreTransporter, 2020-04-29:
Hi,

I try to create the API documentation and get the following errors:
```
1>CUSTOMBUILD : warning : the dot tool could not be found at /usr/bin/dot
1>CUSTOMBUILD : error : source images/ is not a readable file or directory... skipping.
1>D:/workspace/src/include/pulse_detect.h:109: error: The following parameter of pulse_detect_set_levels(pulse_detect_t *pulse_detect, int use_mag_est, float fixed_high_level, float min_high_level, float high_low_ratio) is not documented:
1>  parameter 'use_mag_est' (warning treated as error, aborting now)
1>CUSTOMBUILD : error : Included by graph for 'decoder.h' not generated, too many nodes (129), threshold is 50. Consider increasing DOT_GRAPH_MAX_NODES.
```

zuckschwerdt, 2020-04-29:
Interesting about the `dot` tool. I'm not aware that we use that. Maybe just a bad default? Do you know where `dot` is used in our documentation?

OgreTransporter, 2020-04-29:
dot is used for caller and called-by graphs. Also for the graphs of inherited classes.

The default value for HAVE_DOT is NO, but there are still problems under Windows if DOT_PATH is a Linux path. Therefore I let HAVE_DOT and DOT_PATH set by CMake. If dot exists, HAVE_DOT is set to YES and DOT_PATH is set to the appropriate path. If dot is not present, HAVE_DOT is set to NO and DOT_PATH is set to empty.

closed 2020-04-30 by zuckschwerdt