merbanan/rtl_433 #1370: Compilation error

f5gkw, 2020-04-30:
Hello.
I try to install on raspberry Strecth.

I got an error :

pi@Raspi-POCSAG:~/rtl_433/build $ sudo make install

[  1%] Building C object src/CMakeFiles/r_433.dir/abuf.c.o
/home/pi/rtl_433/src/abuf.c: In function ‘abuf_cat’:
/home/pi/rtl_433/src/abuf.c:46:12: internal compiler error: Erreur de segmentation
     size_t len = strlen(str);
            ^~~
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-6/README.Bugs> for instructions.
src/CMakeFiles/r_433.dir/build.make:62 : la recette pour la cible « src/CMakeFiles/r_433.dir/abuf.c.o » a échouée
make[2]: *** [src/CMakeFiles/r_433.dir/abuf.c.o] Erreur 1
CMakeFiles/Makefile2:1035 : la recette pour la cible « src/CMakeFiles/r_433.dir/all » a échouée
make[1]: *** [src/CMakeFiles/r_433.dir/all] Erreur 2
Makefile:138 : la recette pour la cible « all » a échouée
make: *** [all] Erreur 2

what can I do ?

zuckschwerdt, 2020-04-30:
> /home/pi/rtl_433/src/abuf.c:46:12: internal compiler error: Erreur de segmentation

That looks like your compiler crashed. I'd say the SD card maybe be bad? Maybe check all installed packages with e.g.:
`dpkg-query -f '${binary:Package}\n' -W |while read p ; do echo PKG $p ; sudo dpkg --verify $p ; done |grep -v '??5?????? c ' |grep -B 1 \?`

f5gkw, 2020-05-01:
Hello Christian,
thank you.
Probably, because I also tried to install Kalibrate and the compiler
refuses to make executables...
I tried to reinstall  gcc c++ and some other files as I saw on forums,
without success..
I think I will gain time by reinstall my distro from scratch. (I use it
mainly with "pagermon" for pocsag monitoring )
73 !
Jean-Philippe "f5gkw"

closed 2020-05-01 by zuckschwerdt