merbanan/rtl_433 #1372: gcc10 -fanalyzer warnings

Mindavi, 2020-05-03:
Hi, I've been playing around with the new gcc10 -fanalyzer mode.

See [blog post](https://developers.redhat.com/blog/2020/03/26/static-analysis-in-gcc-10/) and [gcc manual](https://gcc.gnu.org/onlinedocs/gcc/Static-Analyzer-Options.html).

This has brought up some potential leaks or issues. I think one is already mentioned in #1316, but other may be valid as well. Of course, some could be false positives, but I'm not sure.

<details><summary>rtltcp_read_loop</summary>

```
In function 'rtltcp_read_loop':
/home/rick/rtl_433/src/sdr.c:226:61: warning: leak of '<unknown>' [CWE-401] [-Wanalyzer-malloc-leak]
  226 |             r = recv(dev->rtl_tcp, &buffer[n_read], buf_len - n_read, MSG_WAITALL);
      |                                                     ~~~~~~~~^~~~~~~~
  'sdr_start': events 1-4
    |
    | 1169 | int sdr_start(sdr_dev_t *dev, sdr_read_cb_t cb, void *ctx, uint32_t buf_num, uint32_t buf_len)
    |      |     ^~~~~~~~~
    |      |     |
    |      |     (1) entry to 'sdr_start'
    | 1170 | {
    | 1171 |     if (dev->rtl_tcp)
    |      |        ~
    |      |        |
    |      |        (2) following 'true' branch...
    | 1172 |         return rtltcp_read_loop(dev, cb, ctx, buf_num, buf_len);
    |      |                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    |      |                |
    |      |                (3) ...to here
    |      |                (4) calling 'rtltcp_read_loop' from 'sdr_start'
    |
    +--> 'rtltcp_read_loop': events 5-18
           |
           |  207 | static int rtltcp_read_loop(sdr_dev_t *dev, sdr_read_cb_t cb, void *ctx, uint32_t buf_num, uint32_t buf_len)
           |      |            ^~~~~~~~~~~~~~~~
           |      |            |
           |      |            (5) entry to 'rtltcp_read_loop'
           |  208 | {
           |  209 |     if (dev->buffer_size != buf_len) {
           |      |        ~    
           |      |        |
           |      |        (6) following 'true' branch...
           |  210 |         free(dev->buffer);
           |      |         ~~~~~~~~~~~~~~~~~
           |      |         |
           |      |         (7) ...to here
           |  211 |         dev->buffer = malloc(buf_len);
           |      |                       ~~~~~~~~~~~~~~~
           |      |                       |
           |      |                       (8) allocated here
           |  212 |         if (!dev->buffer) {
           |      |            ~
           |      |            |
           |      |            (9) assuming '<unknown>' is non-NULL
           |      |            (10) following 'false' branch...
           |......
           |  216 |         dev->buffer_size = buf_len;
           |      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~
           |      |                          |
           |      |                          (11) ...to here
           |......
           |  227 |             if (r <= 0)
           |      |                ~
           |      |                |
           |      |                (12) following 'true' branch (when 'r <= 0')...
           |......
           |  234 |         if (r < 0) {
           |      |            ~
           |      |            |
           |      |            (13) ...to here
           |......
[... 1036 lines cut ...]
           |
         'mbuf_init': event 23
           |
           | 1550 |   mbuf_resize(mbuf, initial_size);
           |      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
           |      |   |
           |      |   (23) returning to 'mbuf_init' from 'mbuf_resize'
           |
    <------+
    |
  'mg_send_dns_query': events 24-25
    |
    |11712 |   mbuf_init(&pkt, 64 /* Start small, it'll grow as needed. */);
    |      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    |      |   |
    |      |   (24) returning to 'mg_send_dns_query' from 'mbuf_init'
    |11713 | 
    |11714 |   msg->transaction_id = ++mg_dns_tid;
    |      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    |      |                       |
    |      |                       (25) 'msg' could be NULL: unchecked value from (2)
    |
```

</details>

zuckschwerdt, 2020-05-03:
Thanks! We can perhaps classify them as:
- false positive: if possible change the syntax to help the checker. (like done here: https://github.com/merbanan/rtl_433/blob/master/src/data.c#L212)
- static leak: cleaned up on exit anyway, fix as elegant as possible (e.g. #1316)
- dynamic leak: needs a fix asap, no matter if the code is ugly.

I guess the warnings will only be of the first two. I just installed GCC 10 and will look into that ome more.

gvanem, 2020-05-04:
I can confirm that there are tons of memleaks in RTL_433. Confirmed via `_DEBUG`-mode and [Visual Leak Detector](https://github.com/KindDragon/vld.git) using MSVC-2019 at run-time. No static analyser needed to establish that fact.

zuckschwerdt, 2020-05-04:
No dynamic leaks as far as I know? Static one-time allocs are well-defined to clean up at exit but should be addressed someday if possible.
But if you find dynamic leaks please point them out asap.

merbanan, 2020-05-04:
I am running several instances 24/7 so I am quite sure that there are no leaks in the sdr processing path. There might be in the message handling path but I've survived for quite a while now. But yeah we should clean up static allocations also. I remember CSA complaining about something also, but that was mainly because of code complexity.

zuckschwerdt, 2020-05-04:
I didn't check the last two but all the others are false positive. Perhaps someone can find out how to make that easier for the checker?

gvanem, 2020-05-04:
Here is one:
```
WARNING: Visual Leak Detector detected memory leaks!
---------- Block 7 at 0x0D3EEC40: 18266 bytes ----------
  Leak Hash: 0xA5F04199, Count: 1, Total 18266 bytes
  Call Stack (TID 13060):
    ucrtbase.dll!malloc()
    F:\gv\dx-radio\RTL_433\src\confparse.c (68): rtl_433.exe!readconf() + 0xB bytes
  Data:
    23 20 63 6F    6E 66 69 67    20 66 6F 72    20 72 74 6C     #.config .for.rtl
    5F 34 33 33    0A 0A 23 20    41 20 76 61    6C 69 64 20     _433..#. A.valid.
    63 6F 6E 66    69 67 20 6C    69 6E 65 20    69 73 20 61     config.l ine.is.a
    20 6B 65 79    77 6F 72 64    20 66 6F 6C    6C 6F 77 65     .keyword .followe
    64 20 62 79    20 61 6E 20    61 72 67 75    6D 65 6E 74     d.by.an. argument
    20 74 6F 20    74 68 65 20    65 6E 64 20    6F 66 20 6C     .to.the. end.of.l
    69 6E 65 2E    0A 23 20 57    68 69 74 65    73 70 61 63     ine..#.W hitespac
    65 20 61 72    6F 75 6E 64    20 74 68 65    20 6B 65 79     e.around .the.key
    77 6F 72 64    20 69 73 20    69 67 6E 6F    72 65 64 2C     word.is. ignored,
    20 77 68 69    74 65 73 70    61 63 65 20    69 73 20 73     .whitesp ace.is.s
    70 61 63 65    20 61 6E 64    20 74 61 62    0A 23 20 43     pace.and .tab.#.C
    6F 6D 6D 65    6E 74 73 20    73 74 61 72    74 20 77 69     omments. start.wi
    74 68 20 61    20 68 61 73    68 20 73 69    67 6E 2C 20     th.a.has h.sign,.
    6E 6F 20 69    6E 6C 69 6E    65 20 63 6F    6D 6D 65 6E     no.inlin e.commen
    74 73 2C 20    65 6D 70 74    79 20 6C 69    6E 65 73 20     ts,.empt y.lines.
    61 72 65 20    6F 6B 2E 0A    23 0A 23 20    42 6F 6F 6C     are.ok.. #.#.Bool

Visual Leak Detector detected 1 memory leak (18266 bytes).
Largest number used: 41755695 bytes.
Total allocations: 42014247 bytes.
```

18266 bytes == the size of my config-file + 1 byte.

gvanem, 2020-05-04:
Even a  `rtl_433 -?` gives me 133 leaks.

zuckschwerdt, 2020-05-04:
That's https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L677 and just a static leak. Not pretty but harmless. I'll see if I can add a simple workaround, that way other leaks can be more prominent.

gvanem, 2020-05-04:
Yes, but why not do `cfg->conf_file = readconf(path);` and add `r_free_cfg()` in several places?

gvanem, 2020-05-04:
PS. I once suggested a layout like this (pseudo-code):
```
{
  struct dm_state *demod = NULL;
  int    rc = parse_cmd_line();

  if (rc == 0)
  {
    rc = init_demodulator (in_filename, out_filename, &demod);
    if (rc == 0)
    {
      if (demod->in_file)
         rc = run_infile_loop (demod);
      else if (sync_mode)
         rc = run_live_sync_loop (demod);
      else
         rc = run_live_async_loop (demod);
    }
  }
  main_cleanup (demod); // free 'demod' and all output-handlers
  return rc;
}
```

Currently there is massive allocations just to show a help-page.

gvanem, 2020-05-04:
And experimenting with `-F mqtt` shows a leak in [data_output_mqtt_create()](https://github.com/merbanan/rtl_433/blob/master/src/output_mqtt.c#L98).

zuckschwerdt, 2020-05-04:
> Yes, but why not do `cfg->conf_file = readconf(path);` and add `r_free_cfg()` in several places?

The confs are recursive, but something similar is now in feat-mfree branch, s.a. d041d73

Mindavi, 2020-08-10:
Honestly, I don't think any of these are very important to fix. The only thing that we don't really control all too well is the mongoose code, so that might possibly be a problem. However, as far as I can tell these are mostly false positives or possibly some abuse of the C language. Anyhow, nothing to worry about.

I think the only things to check are the following: `decoder_output_bitrow` -> because the static analyzer thinks this might lead to a double free. Maybe some massaging of the code resolves this, or we might need to actually fix a bug there. Let's look at `flex_create_device` as well in this issue, because the information shown here might be useful in finding the bug #1316.

gdt, 2023-09-25:
Is there any reason for this issue to stay open?  It would be most helpful instead to find and fix bugs and file PRs, or to file a PR with static analysis config so anyone can run it easily, or something like that, which I would have described if I understood better.

zuckschwerdt, 2023-10-21:
I've added the Static analyzer as default with GCC 10+.
Most of the false positives could be syntax'ed away but the alloc in vdata_make() just can't be proven.

commit 96226b5a (2023-10-21): minor: Add GCC Static analysis (closes #1372)
closed 2023-10-21 by zuckschwerdt