merbanan/rtl_433 #1373: Ecowitt WH32E not decoding, Except it does on one machine only

allenw864, 2020-05-05:
I'm running an older version of rtl_433 on a Asus RT-N16 router running openWRT for weather data.
The command line is:
rtl_433 -M time -R 78  -p 50 -f 915050000 -C customary -d 0 -F "mqtt://ipaddress:1883, user=username, pass=password, retain=0, usechannel=beforeid, events=stat/sensor/weather_data" 

The Weather station shows up without issue as FineOffset WH24 and a Fine Offset WH25 for the indoor temp/hum/press sensor.
I purchased 4 Ecowitt WH32E 915Mhz temp/hum/press sensors, http://www.amazon.com/gp/product/B07MZ4RWY1
They show up as Fine Offset WH32B with correct values (been using for 6 months without issue).

Recently, I have moved the same cheap chinese SDR to an Intel Nuc Running Unbuntu 18.04 and now only the WH24 and WH25 show up.
I'm including samples from each of the Ecowitt WH32E as well as the samples being decoded on the RT-N16 box.
[rtl_433-samples.zip](https://github.com/merbanan/rtl_433/files/4578337/rtl_433-samples.zip)
I hope this is the correct way to report this, if not my apologies.

zuckschwerdt, 2020-05-05:
Does rtl_433 see the packets and just not decode? Try `-S unknown` to catch all undecoded transmissions.
Otherwise check that the receiver is at least 10-20cm from the computer to reduce interference.

allenw864, 2020-05-05:
Yes, the sdr does receive the packets, The sdr is plugged into an powered USB hub and is at least 20cm from the iNUC. 
I've already used the -S unknown command and included those 4 files (one for each of the ecowitts sensors) in the .zip file in the OP.  I verified the samples are good with https://triq.org/iqs/
The 'ecowitt-WH32E-decoded.txt' file that is in the zip was what was decoded when I ran those .cu8 samples on the old Asus router setup using the -r command.
I'm not a programmer, but I did look thru the rtl_433 code and could find no mention of the Ecowitt WH32E sensor. But my guess is that it is very close to the FineOffset WH32B.
On a hunch, I downloaded a earlier version (rtl_433-18.12-220-g79847de-win-x64) on my windows machine and was able to decode the sample .cu8 files as the FineOffset WH32B, but it will not decode on the newer releases.

zuckschwerdt, 2020-05-05:
Ah ok. The samples don't work well with the new FSK demod. Use `-Y classic`. Also the family code isn't what's expected for a WH32B(WH32E). Try `-vv` to see the error message.
Remove this line and recompile: https://github.com/merbanan/rtl_433/blob/master/src/devices/fineoffset.c#L468
Perhaps the family-code 0xd (instead of 0xe) signals the missing pressure sensor?

allenw864, 2020-05-05:
FYI, In the OP I said the readings were correct for the WH32E, but they are not. the pressure in incorrect.
fortunately I can use the pressure of the WH25

zuckschwerdt, 2020-05-05:
Yes, it looks like pressure is marked as not-available (value 0xffff = -1) on those sensors.

samrushing, 2020-06-18:
I also own a WH32E (FCC ID: WA5WH32E) and was unable to decode until allowing 0xd for msg_type.  Mine came as a set from AmbientWeather, meant to be used with an ObserverIP.  The indoor sensor is a WH32B and includes the pressure data, the outdoor sensor is a WH32E and sends 0xffff for the pressure value.

zuckschwerdt, 2020-06-18:
Thanks for the feedback, so indeed the family-code 0xd (instead of 0xe) signals the missing pressure sensor?
I guess I'll just allow both codes and decide the pressure sensor on the observed `-1` to be safe?

zuckschwerdt, 2020-07-20:
With #1431 this should now be reported as "Fineoffset-WH32", can you verify?

closed 2020-07-27 by zuckschwerdt