merbanan/rtl_433 #1374: Cotech 36-7959 (uctech ft020/ft-0205)

burken-, 2020-05-06:
[cotech_36-7959.zip](https://github.com/merbanan/rtl_433/files/4589068/cotech_36-7959.zip)

I have a Cotech 36-7959 weather station which transmits over 433Mhz. It doesn't get recognized as any of the existing devices in rtl_433 and I have been trying to figure out how to decode and read the values but without any luck. I have attached recorded signals and measured values for each signal, this was done using the following command _rtl_433 -f 433.92M -s 1000k -S all_.

From what I can tell the weather station transmits two different kind of signals. The weather station displays updated values no matter which of the signals that gets transmitted and it seems to update all the values every time.

**I tried to run rtl_433 with the "-A" flag and every second time I get a result similar to this**:
`Detected OOK package    2020-04-18 11:31:39
Analyzing pulses...
Total count:   98,  width: 123.54 ms            (30886 S)
Pulse width distribution:
 [ 0] count:   69,  width:  484 us [456;516]    ( 121 S)
 [ 1] count:   29,  width:  976 us [964;988]    ( 244 S)
Gap width distribution:
 [ 0] count:   68,  width:  488 us [460;520]    ( 122 S)
 [ 1] count:   29,  width:  976 us [964;988]    ( 244 S)
Pulse period distribution:
 [ 0] count:   57,  width:  976 us [964;992]    ( 244 S)
 [ 1] count:   23,  width: 1460 us [1432;1484]  ( 365 S)
 [ 2] count:   17,  width: 1952 us [1936;1972]  ( 488 S)
Level estimates [high, low]:   8392,    208
RSSI: -2.9 dB SNR: 16.1 dB Noise: -19.0 dB
Frequency offsets [F1, F2]:   -6866,      0     (-26.2 kHz, +0.0 kHz)
Guessing modulation: Manchester coding
Attempting demodulation... short_width: 484, long_width: 0, reset_limit: 992, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_MC_ZEROBIT,s=484,l=0,r=992'
pulse_demod_manchester_zerobit(): Analyzer Device
bitbuffer:: Number of rows: 1
[00] {127} 00 29 8a 88 38 56 b6 04 39 06 8a 53 ff f7 f6 ea
`
**And every second time it looks like this**:
`Detected OOK package    2020-04-18 11:32:27
Analyzing pulses...
Total count:  199,  width: 253.92 ms            (63481 S)
Pulse width distribution:
 [ 0] count:  143,  width:  488 us [444;520]    ( 122 S)
 [ 1] count:   56,  width:  972 us [964;984]    ( 243 S)
Gap width distribution:
 [ 0] count:  141,  width:  484 us [460;520]    ( 121 S)
 [ 1] count:   56,  width:  976 us [964;1040]   ( 244 S)
 [ 2] count:    1,  width: 5872 us [5872;5872]  (1468 S)
Pulse period distribution:
 [ 0] count:  127,  width:  976 us [916;992]    ( 244 S)
 [ 1] count:   30,  width: 1464 us [1440;1512]  ( 366 S)
 [ 2] count:   40,  width: 1952 us [1936;1968]  ( 488 S)
 [ 3] count:    1,  width: 6852 us [6852;6852]  (1713 S)
Level estimates [high, low]:  10696,    337
RSSI: -1.9 dB SNR: 15.0 dB Noise: -16.9 dB
Frequency offsets [F1, F2]:   -6386,      0     (-24.4 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
Attempting demodulation... short_width: 488, long_width: 972, reset_limit: 5876, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=488,l=972,r=5876,g=1044,t=192,y=0'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2
[00] {99} ff 97 1b 6d 57 7d dd eb 25 ff f7 ef 40
[01] {100} ff cb 8d b6 ab be ee f5 92 ff fb f7 a0`

Any pointers, help, ideas would be appriciated.

klohner, 2020-05-07:
A flex decoder like this might help align and decode these messages:

```
-X n=cotech-36-7959,m=OOK_MC_ZEROBIT,s=488,r=1200,preamble={8}05
```

Here's a [bitbench of the cotech_36-7959.zip messages](https://triq.net/bitbench?c=%7B114%7D315000002f00c6211ac8bffefeee4%20%5Bg001_433.92M_1000k.cu8%5D&c=%7B114%7D315000002f40c6211ac8bffefec30%20%5Bg002_433.92M_1000k.cu8%5D&c=%7B114%7D315000002f00c6211ac8bffefeee4%20%5Bg003_433.92M_1000k.cu8%5D&c=%7B114%7D315000002f00c6211ac8bffefeee4%20%5Bg004_433.92M_1000k.cu8%5D&c=%7B114%7D3151000018c0c6211a88bffefef5c%20%5Bg005_433.92M_1000k.cu8%5D&c=%7B114%7D3151000018c0c6211a88bffefef5c%20%5Bg006_433.92M_1000k.cu8%5D&c=%7B114%7D315100000340c6211a88bffefeea0%20%5Bg007_433.92M_1000k.cu8%5D&c=%7B114%7D315100000380c6211a88bffefed18%20%5Bg008_433.92M_1000k.cu8%5D&c=%7B114%7D3150000016c0c6211ac8bffefec2c%20%5Bg009_433.92M_1000k.cu8%5D&c=%7B114%7D3150000016c0c6211ac8bffefec2c%20%5Bg010_433.92M_1000k.cu8%5D&c=%7B114%7D315100000200c6211ac8bffefede0%20%5Bg011_433.92M_1000k.cu8%5D&c=%7B114%7D315100000240c6211ac8bffefef34%20%5Bg012_433.92M_1000k.cu8%5D&c=%7B114%7D315001852f80c6211a88bffefede8%20%5Bg013_433.92M_1000k.cu8%5D&c=%7B114%7D315002446f40c6211a88bffefeec0%20%5Bg014_433.92M_1000k.cu8%5D&c=%7B114%7D315002c46f00c6211a88bffefefe0%20%5Bg015_433.92M_1000k.cu8%5D&c=%7B114%7D315003452f00c6211a88bffefee74%20%5Bg016_433.92M_1000k.cu8%5D&c=%7B114%7D315000002e00c6211b48bffefee1c%20%5Bg017_433.92M_1000k.cu8%5D&c=%7B114%7D315000002e00c6211b48bffefee1c%20%5Bg018_433.92M_1000k.cu8%5D&c=%7B114%7D315000003b80c6211b48bffefed68%20%5Bg019_433.92M_1000k.cu8%5D&c=%7B114%7D315000003c40c6211b48bffefef94%20%5Bg020_433.92M_1000k.cu8%5D&c=%7B114%7D315100000ac0c6211b48bffefed6c%20%5Bg021_433.92M_1000k.cu8%5D&c=%7B114%7D315100000ac0c6211b48bffefed6c%20%5Bg022_433.92M_1000k.cu8%5D&c=%7B114%7D315000000e00c6211b48bffefee88%20%5Bg023_433.92M_1000k.cu8%5D&c=%7B114%7D315000000e40c6211b48bffefec5c%20%5Bg024_433.92M_1000k.cu8%5D&c=%7B114%7D315000001f00c6211b48bffefee78%20%5Bg025_433.92M_1000k.cu8%5D&c=%7B114%7D315000001f80c6211b48bffefef14%20%5Bg026_433.92M_1000k.cu8%5D&c=%7B114%7D315000002fc0c6211b48bffefef7c%20%5Bg027_433.92M_1000k.cu8%5D&c=%7B114%7D3150000030c0c6211b48bffefed04%20%5Bg028_433.92M_1000k.cu8%5D&c=%7B114%7D31500041efc0d2211b48bffefecbc%20%5Bg029_433.92M_1000k.cu8%5D&c=%7B114%7D315000002f80dd611b48fffefed04%20%5Bg030_433.92M_1000k.cu8%5D&f=2b%2016b%2016b%2016b%2016b%2016b%2016b%208b%20CHK%3A8h).

Next step might be to try to figure out which bits go to which reported weather values.  The final byte of each message looks like some kind of CRC or checksum.

klohner, 2020-05-07:
Better might be to use this to get rid of those first couple of bits:

```
 -X n=cotech-36-7959,m=OOK_MC_ZEROBIT,s=488,r=1200,preamble={12}014
```

Feeding these messages to [reveng](http://reveng.sourceforge.net/), we can figure out the CRC.

```
$ reveng -w 8 -s c5400000bc0318846b22fffbfbb9 c5400000bd0318846b22fffbfb0c c5400000bc0318846b22fffbfbb9 c5400000bc0318846b22fffbfbb9 c5440000630318846a22fffbfbd7
width=8  poly=0x31  init=0xc0  refin=false  refout=false  xorout=0x00  check=0x0d  residue=0x00  name=(none)
```

burken-, 2020-05-07:
@klohner Thanks a bunch! I can't claim to understand how to figure out the flex decoder settings from the cu8-files so it was very helpful to get a jump start.

burken-, 2020-05-12:
Ok, got most of the [values (mapped on bitbench)](https://triq.net/bitbench?c=%7B112%7Dc5400000bc0318846b22fffbfbb9%20%5Bg001_433.92M_1000k.cu8%5D&c=%7B112%7Dc5400000bd0318846b22fffbfb0c%20%5Bg002_433.92M_1000k.cu8%5D&c=%7B112%7Dc5400000bc0318846b22fffbfbb9%20%5Bg003_433.92M_1000k.cu8%5D&c=%7B112%7Dc5400000bc0318846b22fffbfbb9%20%5Bg004_433.92M_1000k.cu8%5D&c=%7B112%7Dc5440000630318846a22fffbfbd7%20%5Bg005_433.92M_1000k.cu8%5D&c=%7B112%7Dc5440000630318846a22fffbfbd7%20%5Bg006_433.92M_1000k.cu8%5D&c=%7B112%7Dc54400000d0318846a22fffbfba8%20%5Bg007_433.92M_1000k.cu8%5D&c=%7B112%7Dc54400000e0318846a22fffbfb46%20%5Bg008_433.92M_1000k.cu8%5D&c=%7B112%7Dc54000005b0318846b22fffbfb0b%20%5Bg009_433.92M_1000k.cu8%5D&c=%7B112%7Dc54000005b0318846b22fffbfb0b%20%5Bg010_433.92M_1000k.cu8%5D&c=%7B112%7Dc5440000080318846b22fffbfb78%20%5Bg011_433.92M_1000k.cu8%5D&c=%7B112%7Dc5440000090318846b22fffbfbcd%20%5Bg012_433.92M_1000k.cu8%5D&c=%7B112%7Dc5400614be0318846a22fffbfb7a%20%5Bg013_433.92M_1000k.cu8%5D&c=%7B112%7Dc5400911bd0318846a22fffbfbb0%20%5Bg014_433.92M_1000k.cu8%5D&c=%7B112%7Dc5400b11bc0318846a22fffbfbf8%20%5Bg015_433.92M_1000k.cu8%5D&c=%7B112%7Dc5400d14bc0318846a22fffbfb9d%20%5Bg016_433.92M_1000k.cu8%5D&c=%7B112%7Dc5400000b80318846d22fffbfb87%20%5Bg017_433.92M_1000k.cu8%5D&c=%7B112%7Dc5400000b80318846d22fffbfb87%20%5Bg018_433.92M_1000k.cu8%5D&c=%7B112%7Dc5400000ee0318846d22fffbfb5a%20%5Bg019_433.92M_1000k.cu8%5D&c=%7B112%7Dc5400000f10318846d22fffbfbe5%20%5Bg020_433.92M_1000k.cu8%5D&c=%7B112%7Dc54400002b0318846d22fffbfb5b%20%5Bg021_433.92M_1000k.cu8%5D&c=%7B112%7Dc54400002b0318846d22fffbfb5b%20%5Bg022_433.92M_1000k.cu8%5D&c=%7B112%7Dc5400000380318846d22fffbfba2%20%5Bg023_433.92M_1000k.cu8%5D&c=%7B112%7Dc5400000390318846d22fffbfb17%20%5Bg024_433.92M_1000k.cu8%5D&c=%7B112%7Dc54000007c0318846d22fffbfb9e%20%5Bg025_433.92M_1000k.cu8%5D&c=%7B112%7Dc54000007e0318846d22fffbfbc5%20%5Bg026_433.92M_1000k.cu8%5D&c=%7B112%7Dc5400000bf0318846d22fffbfbdf%20%5Bg027_433.92M_1000k.cu8%5D&c=%7B112%7Dc5400000c30318846d22fffbfb41%20%5Bg028_433.92M_1000k.cu8%5D&c=%7B112%7Dc5400107bf0348846d22fffbfb2f%20%5Bg029_433.92M_1000k.cu8%5D&c=%7B112%7Dc5400000be0375846d23fffbfb41%20%5Bg030_433.92M_1000k.cu8%5D&c=%7B112%7Dc9a80000e60000846e1ffffbfb03&c=%7B112%7Dc9a80000e70000846e1ffffbfbb6&c=%7B112%7Dc9a80000e70000846e20fffbfb70&c=%7B112%7Dcd300000da0000846e20fffbfbca&c=%7B112%7Dcd300000da0000846d20fffbfb8e&c=%7B112%7Dcd3400005a0000846d20fffbfb06&c=%7B112%7Dcd340000590000846d20fffbfbe8&c=%7B112%7Dcd340000610000846d20fffbfb4a&c=%7B112%7Dcd300000000000846d20fffbfb9d&c=%7B112%7Dcd300000dd0000846d20fffbfbd6&c=%7B112%7Dc6880000e90000846d20fffbfb8c&c=%7B112%7Dc6880000e80000846d20fffbfb39&c=%7B112%7Dc6880914e90000846d20fffbfbb0&c=%7B112%7Dc6880d14e90000846d20fffbfb7b&c=%7B112%7Dc6880a14e80000846d20fffbfb1e&c=%7B112%7Dc6880714ea0000846d20fffbfbcf&c=%7B112%7Dc6880a14ec0000846d20fffbfba8&c=%7B112%7Dcc400000e30000846c20fffbfb26&c=%7B112%7Dcc4007114b0000846c20fffbfbe5&c=%7B112%7Dcc400103e40000846d21fffbfb0e&c=%7B112%7Dcc440000100000843a1cfffbfb5a&c=%7B112%7Dcc440000100000842b1dfffbfb6e&c=%7B112%7Dcc4400000d0000831c4afffbfb8b&c=%7B112%7Dcc4400000d0000831b49fffbfb4c&c=%7B112%7Dcc4400000c0000831b49fffbfbf9&c=%7B112%7Dcc4400000d0000831948fffbfb40&c=%7B112%7Dcc4400000d000083164afffbfb22&c=%7B112%7Dcc4400000c000082fb50fffbfb9b&c=%7B112%7Dcc4400000c000083034ffffbfbb2&c=%7B112%7Dcc440103030000838d25fffbfb01&c=%7B112%7Dcc44181e01004e839e4cfffbfbdb&c=%7B112%7Dcc401c24e8004e83984ffffbfba1&c=%7B112%7Dcc401f24f20051839650fffbfbab&c=%7B112%7Dcc4410140c0051839452fffbfbad&c=%7B112%7Dcc400a0d820054839654fffbfb2c&c=%7B112%7Dcc400507900054839754fffbfbcf&c=%7B112%7Dcc40191ed6005783a456fffbfb6a&c=%7B112%7Dcc401d21c1005783a154fffbfbd1&c=%7B112%7Dcc441f242a00a883165dfffbfb00&c=%7B112%7Dcc440b113e00a883165dfffbfb47&c=%7B112%7Dcc44181e4600a883145dfffbfb69&c=%7B112%7Dcc44141e2c00a883135dfffbfb97&c=%7B112%7Dcc4421282700a883115dfffbfb3e&c=%7B112%7Dcc441e285a00a883115dfffbfbe3&c=%7B112%7Dcc441c282000a883115dfffbfb6d&c=%7B112%7Dcc40232b2900ab830a5cfffbfb97&c=%7B112%7Dcc401a2e4900ab830c5cfffbfb43&c=%7B112%7Dcc44282e2400ab830c5dfffbfb98&c=%7B112%7Dcc4028380400ae82fc5dfffbfb26&c=%7B112%7Dcc4027350600b182f85dfffbfbd9&c=%7B112%7Dcc4411143600b182f95efffbfbf1&c=%7B112%7Dcc44151a3a00b182f75efffbfbd8&c=%7B112%7Dcc44202e5c00b482f55efffbfb6a&c=%7B112%7Dcc40090d7c00de831c5afffbfb58&f=4b%208b%20BATT%3A1b%20DIR%2B%3A1b%20GUST%2B%3A1b%20WIND%2B%3A1b%20WIND%3Ad%20GUST%3Ad%20DIR%3Ad%204b%20RAIN%3A12d%204b%20TEMP-F%3A12d%20HUM%3Ad%2016b%208b%20CHK%3A8h) now. Even though I found and mapped the values displayed on the indoor unit there are still some bits I haven't figured out yet... id?, channel?, padding for crc-calculation? other stuff usually in these sequences? Bits in **bold** are mapped to values displayed on the indoor unit.

- 4 bits = Stays the same all the time, might be some kind of id??
- 8 bits = Changes when removing batteries and putting them back in, might be some channel info?
- **1 bit**  = 0 = Ok battery, 1 = Low battery
- 1 bit  = Wind direction is defined by a byte later on in the sequence. When degrees are over 255 (byte max value) this value is set to 1 and the wind direction starts from 1 again. So when this is = 1 it basically means "wind direction value + 255"
- 1 bit  = Wind and gust are also defined by a byte later on in the sequence. Specification states it can show wind strength up to 50m/s. Guessing this bit works same as previous bit for wind direction i.e. if this is s [... cut ...]
- **_12 bits_** = Rain value / 10 => mm. Don't know if this is actually 12 bits, if it cycles or how it works. So far I have only seen this value increasing (exept for when device was reset)
- 4 bits = Always the same, 1000, no clue
- **12 bits** = Temperature. This value is given in fahrenheit with the formula: (value - 400) / 10
- **8 bits** = Humidity, decimal value giving the percentage
- 24 bits = Always the same, no clue, crc-padding?
- **8 bits** = CRC8 Checksum (width=8  poly=0x31  init=0xc0  refin=false  refout=false  xorout=0x00  check=0x0d  residue=0x00)

To be continued...

zuckschwerdt, 2020-05-25:
Added with #1382

closed 2020-05-25 by zuckschwerdt