merbanan/rtl_433 #1375: Add device support for Eurochron EFTH-800 weather station

thorfae, 2020-05-07:
Hi

I have a request to add support for a new weather station, __Eurochron EFTH-800__. I have tried with the protocol already implemented in RTL_433 named Eruochron (129) but it does not give me any results. I have also tried `rtl_433 -G 4` but did not find any data that looked like the Eurochron sensors.

I have a few pictures of the sensors:
[image] [image] [image]

I already have RTL_433 running with succes collecting data from Ambientweather-F007TH (protocol number 20) so I know my setup is working.

Using `rtl_433 -S` I have recorded the below signals to cu8-files. I disconnected the antenna from the DVB dongle to avoid any noise from other sensors and it seems to work fine.

[cu8 files.zip](https://github.com/merbanan/rtl_433/files/4592195/cu8.files.zip)

Number  | Temperature<br />Celcius   | Temperature<br />Farenheit  | Relative<br />Humidity | Channel
:-- | :-: | :-: | :-: | :-:
001 |  22.8 | 73.0 | 43 | 1 
002 |  22.8 | 75.9 | 42 | 2
003 |  22.8 | 76.2 | 40 | 5
004 |  22.8 |  -   | 39 | 8
005 |  22.8 |  -   | 38 | 1
006 |  22.8 |  -   | 44 | 2
009 | -18.2 | -0.7 | 35 | 1
011 | -12.9 |  8.7 | 43 | 1
012 | -11.2 | 11.8 | 45 | 1
014 |  -9.7 | 14.5 | 48 | 1
015 |  -5.0 | 23.0 | 57 | 1
024 |   1.5 | 34.7 | 63 | 1
029 |   4.1 | 39.3 | 63 | 1

When I use the pulse analyser (-A) I get different results. The pulse look the same but for some I don't get a suggestion for a flex decoder and for some I get one. For g001 I get an output as well.

Running >>  `rtl_433 -r g001_433.92M_250k.cu8 -A`  << I get

```
Test mode active. Reading samples from file: g001_433.92M_250k.cu8
Detected OOK package    @0.127084s
Analyzing pulses...
Total count:   72,  width: 59.52 ms             (14880 S)
Pulse width distribution:
 [ 0] count:   50,  width:  236 us [160;280]    (  59 S)
 [ 1] count:   18,  width:  472 us [432;504]    ( 118 S)
 [ 2] count:    4,  width:  732 us [720;744]    ( 183 S)
Gap width distribution:
 [ 0] count:   48,  width:  492 us [444;564]    ( 123 S)
 [ 1] count:   18,  width:  252 us [232;284]    (  63 S)
 [ 2] count:    1,  width: 4876 us [4876;4876]  (1219 S)
 [ 3] count:    4,  width:  740 us [724;760]    ( 185 S)
Pulse period distribution:
 [ 0] count:   66,  width:  728 us [668;780]    ( 182 S)
 [ 1] count:    1,  width: 5124 us [5124;5124]  (1281 S)
 [ 2] count:    4,  width: 1472 us [1468;1484]  ( 368 S)
Level estimates [high, low]:   1572,    262
RSSI: -10.2 dB SNR: 7.8 dB Noise: -18.0 dB
Frequency offsets [F1, F2]:   -1744,      0     (-6.7 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 236, long_width: 472, reset_limit: 4880, sync_width: 732
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=236,l=472,r=4880,g=0,t=0,y=732'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 2
[00] {19} ef f2 e0             : 11101111 11110010 111
[01] {49} fd 77 b1 1f bf cb 80 : 11111101 01110111 10110001 00011111 10111111 11001011 1
```
I have tried running with the suggested flex decoder but it does not seem to be working correct. 

Also, I found out the FHEM project they already have an integration of this sensor (EFTH-800) and have a perl script to decode the signals. I have copied the relevant code in below if this helps. It looks like the short width from my recording is similar to the one from the perl script: 244 us and long 488 us.

I hope this is sufficient information, else I will try to collect any other needed information. My expertise is not collecting the data but analyzing and evaluating indoor climate date :-)

https://github.com/RFD-FHEM/RFFHEM/blob/master/FHEM/lib/SD_ProtocolData.pm  - this section starts at line 823

```
"27"	=>	## Temperatur-/Feuchtigkeitssensor EuroChron EFTH-800 (433 MHz)
        # SD_WS_27_TH_2 - T: 15.5 H: 48 - MU;P0=-224;P1=258;P2=-487;P3=505;P4=-4884;P5=743;P6=-718;D=0121212301212303030301212123012123012123030123030121212121230121230121212121212121230301214565656561212123012121230121230303030121212301212301212303012303012121212123012123012121212121212123030121;CP=1;R=53;
        # SD_WS_27_TH_3 - T:  3.8 H: 76 - MU;P0=-241;P1=251;P2=-470;P3=500;P4=-4868;P5=743;P6=-718;D=012121212303030123012301212123012121212301212303012121212121230303012303012123030303012123014565656561212301212121230303012301230121212301212121230121230301212121212123030301230301212303030301212301;CP=1;R=23;
        # SD_WS_27_TH_3 - T:  5.3 H: 75 - MU;P0=-240;P1=253;P2=-487;P3=489;P4=-4860;P5=746;P6=-725;D=012121212303030123012301212123012121212303012301230121212121230303012301230303012303030301214565656561212301212121230303012301230121212301212121230301230123012121212123030301230123030301230303030121;CP=1;R=19;
        # short pulse of 244 us followed by a 488 us gap is a 0 bit
        # long pulse of 488 us followed by a 244 us gap is a 1 bit
        # sync preamble of pulse, gap, 732 us each, repeated 4 times
        # sensor sends two messages at intervals of about 57-58 seconds
    {
        name            => 'EFTH-800',
        comment         => 'EuroChron weatherstation EFTH-800',
        id              => '27',
        knownFreqs      => '433.92',
        one             => [2,-1],
        zero            => [1,-2],
        start           => [3,-3,3,-3,3,-3,3,-3],
        clockabs        => '244',
        format          => 'twostate',
        preamble        => 'W27#',
        clientmodule    => 'SD_WS',
        modulematch     => '^W27#.{12}',
        length_min      => '48',    # 48 Bit + 1 Puls am Ende
        length_max      => '48',
    },
```			

https://github.com/RFD-FHEM/RFFHEM/blob/master/FHEM/14_SD_WS.pm  -  this section starts at line 188:

```	
27 =>
    {
        # Protokollbeschreibung: Temperatur-/Feuchtigkeitssensor EuroChron EFTH-800
        # -----------------------------------------------------------------------------------
        # 0    4    | 8    12   | 16   20   | 24   28   | 32   36   | 40   44
        # 0000 1001 | 0001 0110 | 0001 0000 | 0000 0000 | 0100 1001 | 0100 0000
        # ?ccc iiii | iiii iiii | bstt tttt | tttt ???? | hhhh hhhh | xxxx xxxx
        # c:  3 bit channel valid channels are 0-7 (stands for channel 1-8)
        # i: 12 bit random id (changes on power-loss)
        # b:  1 bit battery indicator (0=>OK, 1=>LOW)
        # s:  1 bit sign temperature (0=>negative, 1=>positive)
        # t: 10 bit unsigned temperature, scaled by 10
        # h:  8 bit relative humidity percentage (BCD)
        # x:  8 bit CRC8
        # ?: unknown (Bit 0, 28-31 always 0 ???)
        # The sensor sends two messages at intervals of about 57-58 seconds
        sensortype => 'EFTH-800',
        model      => 'SD_WS_27_TH',
        prematch   => sub {my $rawData = shift; return 1 if ($rawData =~ /^[0-9A-F]{7}0[0-9]{2}[0-9A-F]{2}$/); },    # prematch 113C49A 0 47 AE
        channel    => sub {my (undef,$bitData) = @_; return (SD_WS_binaryToNumber($bitData,1,3) + 1 ); },
        id         =>    sub {my (undef,$bitData) = @_; return substr($rawData,1,3); },
        bat        => sub {my (undef,$bitData) = @_; return substr($bitData,16,1) eq "0" ? "ok" : "low";},
        temp       => sub {my (undef,$bitData) = @_; return substr($bitData,17,1) eq "0" ? ((SD_WS_binaryToNumber($bitData,18,27) - 1024) / 10.0) : (SD_WS_binaryToNumber($bitData,18,27) / 10.0);},
        hum        => sub {my (undef,$bitData) = @_; return (SD_WS_binaryToNumber($bitData,32,35) * 10) + (SD_WS_binaryToNumber($bitData,36,39));},
        crcok      => sub {my $rawData = shift;
            my $rc = eval
            {
                require Digest::CRC;
                Digest::CRC->import();
                1;
            };
            if ($rc) {
                my $datacheck1 = pack( 'H*', substr($rawData,0,10) );
                my $crcmein1 = Digest::CRC->new(width => 8, poly => 0x31);
                my $rr3 = $crcmein1->add($datacheck1)->hexdigest;
                Log3 $name, 4, "$name: SD_WS_27 Parse msg $rawData, CRC $rr3";
                if (hex($rr3) == hex(substr($rawData,-2))) {
                    return 1;
                } else {
                    return 0;
                }
            } else {
                Log3 $name, 1, "$name: SD_WS_27 Parse msg $rawData - ERROR CRC not load, please install modul Digest::CRC";
                return 0;
            }  
        }
    } ,
```

zuckschwerdt, 2020-05-09:
Thanks. That enough detailed information to get this decoder done.
The flex decoder (`-X 'n=EFTH-800,m=OOK_PWM,s=236,l=472,y=732,r=4880'`) should work:
- g001_433.92M_250k.cu8: `{19}eff2e, {49}fd77b11fbfcb8`
- g002_433.92M_250k.cu8: `{1}0, {49}e487b0bfbdd88, {49}e487b0bfbdd88`
- g003_433.92M_250k.cu8: `{1}0, {49}b8ebb09fbf788, {49}b8ebb09fbf788`
- g004_433.92M_250k.cu8: `{49}8883b09fc6008, {49}8883b09fc6008`
- g005_433.92M_250k.cu8: `{49}fca7b07fc6528, {49}fca7b07fc6528`
- g006_433.92M_250k.cu8: `{1}0, {49}ed9fb14fbbe98, {49}ed9fb14fbbe98`
- g009_433.92M_250k.cu8: `{1}0, {49}f90bcb5fca1b8, {49}f90bcb5fca1b8`
- g011_433.92M_250k.cu8: `{1}0, {49}f90bc80fbc688, {49}f90bc80fbc688`
- g012_433.92M_250k.cu8: `{1}0, {49}f90bc6ffba218, {49}f90bc6ffba218`
- g014_433.92M_250k.cu8: `{1}0, {49}f90bc53faff78, {49}f90bc53faff78`
- g015_433.92M_250k.cu8: `{1}0, {49}f90bc31fa8198, {49}f90bc31fa8198`
- g024_433.92M_250k.cu8: `{1}0, {49}f90bbf0f9c228, {49}f90bbf0f9c228`
- g029_433.92M_250k.cu8: `{1}0, {49}f90bbd6f9cfb8, {49}f90bbd6f9cfb8`

thorfae, 2020-05-11:
Thanks a lot - That got me started and a few step further. I think it was the sync key that was missing or maybe I did something else wrong :-)

I have tried to decipher the results using BitBench - but I could not get anything working. I think hex code is not something I understand - at all. I got the results changed to bits instead and think I cracked it. 49 bits in total are now a little more understandable :-)

`xccc iiii iiii iiyy bstt tttt tttt ???? hhhh hhhh rrrr rrrr r`

* x: The first bit is not used for anything meaningful (at least to me)
* c: three bits giving the channel number. Bits need to be inverted (I don't know if this is the right technical term) - 0 => 1 and 1 => 0. Gives 0 to 7 (eight channels)
* i: eight bit, device id? - changes with battery change or reset
* y: two bits, not used for anything meaningful
* b: battery indicator - 1 = fresh batteries, 0 = low battery
* s: sign of temperature - 1 = negative, 0 = positive
* t: ten bit giving the temperature. If two first bits are 11, then inverse bits, else, use as is
* h: eight bit giving the relative humidity. First four gives first digit (BCD) and last four the second. Needs to be inverted
* r: eight bit I guess could be CRC8? 

How do I proceed from here? Can someone point me in the right direction?

zuckschwerdt, 2020-05-11:
I can use that information to prototype a decoder soon, will take week or so. If anyone has time to do it right away go ahead. I'll ping here when I get around to start this.

closed 2020-05-13 by zuckschwerdt

merbanan, 2020-05-13:
@zuckschwerdt missing MIC in output.

merbanan, 2020-05-13:
@zuckschwerdt Other then that the code is a work of art :)

zuckschwerdt, 2020-05-13:
:) thanks. Just the same old pattern. I want all the decoders to look that dead simple if possible.
Tests went in https://github.com/merbanan/rtl_433_tests/tree/master/tests/Eurochron-EFTH800

thorfae, 2020-05-27:
Hi. Thank you for the very fast work! Me, on the other hand, have been a bit slower. Due to a poor quality dvb-t dongle (based on the FC0012 chip and not the r820t2) I have had some difficulties. Now with a more stable dongle I have found that the decoder has a bug. Maybe I made a mistake in BitBench when I decoded the signals.
The problem is that channel 5 - 8 gets assigned the numbers 1 - 4. I checked the codes from the first four signals [bitbench](https://triq.net/bitbench?c=%7B49%7Dfd77b11fbfcb8&c=%7B49%7De487b0bfbdd88&c=%7B49%7Db8ebb09fbf788&c=%7B49%7D8883b09fc6008&f=b) but it looks fine to me.

| Full string                                                                                  | code | inverted | number | real channel | sent channel |
| ---                                                                                             | --- | --- | --- | --- | --- |
| 1111110101110111101100010001111110111111110010111 | 111 | 000 | 0 | 1 | 1 |
| 1110010010000111101100001011111110111101110110001 | 110 | 001 | 1 | 2 | 2 |
| 1011100011101011101100001001111110111111011110001 | 011 | 100 | 4 | 5 | 1 |
| 1000100010000011101100001001111111000110000000001 | 000 | 111 | 7 | 8 | 2 |

zuckschwerdt, 2020-05-27:
Oops, sorry, I botched that. I'll fix it right away.

zuckschwerdt, 2020-05-27:
Channel numbers fixed with f82c025

thorfae, 2020-05-27:
Very much appreciated! I will test right away

commit 2e33d43b (2020-05-13): Add Eurochron EFTH-800 support (closes #1375)