merbanan/rtl_433 #1377: HCSXXX Keeloq wrong bit-order

neo7530, 2020-05-11:
The HCS200 Decoder shows the Serial and Hopcode in wrong Bitorder.

e.g.: (The same Message decoded with rtl433 and  Keeloq Dev-Kit)

RTL433:
		 
model     : Microchip-HCS200                       id        : 0CAB2100
encrypted : D06A1BBB     button1   : ON            button2   : OFF           button3   : OFF           button4   : OFF
misc      :              battery_ok: 1 
		 
Keeloq Dev Kit:

Repeat = 1
Battery Low = 0
Button-Stat: 0010 = 0x02 HEX = Button S0
HCS200 Serial No: 00084D53
Encr. Hop-Code: DDD8560B
		 
MSB 10 0010 0000 0000 1000 0100 1101 0101 0011 1101 1101 1101 1000 0101 0110 0000 1011 LSB
         2    0    0    8    4    d    5    3    d    d    d    8    5    6    0    b

zuckschwerdt, 2020-05-11:
Interesting find. Not sure if we can adopt the "official" style though. It's a breaking change and users may be relying on the current output.

merbanan, 2020-05-11:
We could add another key with the official order, how about that ?

merbanan, 2020-07-06:
Added bit reversed keys.

closed 2020-07-06 by merbanan