merbanan/rtl_433 #1380: SoapySDR & SDRplay issues

goscickiw, 2020-05-15:
I'm having problems with getting rtl_433 to work with SDRplay RSP1A.
When I run `SoapySDRUtil --probe=sdrplay` it seems to detect the device:

```
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device sdrplay

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=SDRplay
  hardware=RSP1A
  mir_sdr_api_version=2.130000
  mir_sdr_hw_version=255
  serial=1903154196

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO
  Other Settings:
     * RF Gain Select - RF Gain Select
       [key=rfgain_sel, default=4, type=string, options=(0, 1, 2, 3, 4, 5, 6, 7, 8, 9)]
     * IF Mode - IF frequency in kHz
       [key=if_mode, default=Zero-IF, type=string, options=(Zero-IF, 450kHz, 1620kHz, 2048kHz)]
     * IQ Correction - IQ Correction Control
       [key=iqcorr_ctrl, default=true, type=bool]
     * AGC Setpoint - AGC Setpoint (dBfs)
       [key=agc_setpoint, default=-30, type=int, range=[-60, 0]]
     * BiasT Enable - BiasT Control
       [key=biasT_ctrl, default=true, type=bool]
     * RfNotch Enable - RF Notch Filter Control
       [key=rfnotch_ctrl, default=true, type=bool]
     * DabNotch Enable - DAB Notch Filter Control
       [key=dabnotch_ctrl, default=true, type=bool]

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: NO
  Supports AGC: YES
  Stream formats: CS16, CF32
  Native format: CS16 [full-scale=32767]
  Antennas: RX
  Corrections: DC removal
  Full gain range: [0, 48] dB
    IFGR gain range: [20, 59] dB
    RFGR gain range: [0, 9] dB
  Full freq range: [0.01, 2000] MHz
    RF freq range: [0.01, 2000] MHz
    CORR freq range:  MHz
  Sample rates: 0.25, 0.5, 1, 2, 2.048, 6, 7, 8, 9, 10 MSps
  Filter bandwidths: 0.2, 0.3, 0.6, 1.536, 5, 6, 7, 8 MHz
```

However when I try to use it with rtl_433 with this command: `rtl_433 -d driver=sdrplay -v`

I get this error:
`Failed to open sdr device matching 'driver=sdrplay'.`

What am I doing wrong? And where can I find good instructions on how to do this properly?

zuckschwerdt, 2020-05-15:
Could this be a problem with case-sensitivity? I get `driver = sdrplay` with `SoapySDRUtil --find` but `driver=SDRplay` with `SoapySDRUtil --probe=SDRplay`.

goscickiw, 2020-05-15:
I tried `-d driver=SDRplay` and I get the same error as before.
Also I tried `-d sdrplay` and `-d SDRplay` but it just acts as if I did `-d ""`.

zuckschwerdt, 2020-05-15:
All those strings are just passed to SoapySDR's `SoapySDRDevice_makeStrArgs()`. IIRC only known keys act as filters. E.g.
```
SoapySDRUtil --make="driver=sdrplay,foo=bar"
```
will work. `SoapySDRUtil --make=...` is the same call as `rtl_433 -d ...`

zuckschwerdt, 2020-05-15:
One thing to check: is there really only a single SoapySDR version installed? Mismatched headers/libs will show similar problems.

goscickiw, 2020-05-15:
`SoapySDRUtil --make="driver=sdrplay"` gives me this:
```
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Make device driver=sdrplay
libusb: error [udev_hotplug_event] ignoring udev action bind
  driver=SDRplay
  hardware=RSP1A
  mir_sdr_api_version=2.130000
  mir_sdr_hw_version=255
  serial=1903154196
```

I decided to test on Windows as well, and it looks like this:
```
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Make device driver=sdrplay
  driver=SDRplay
  hardware=RSP1A
  mir_sdr_api_version=2.130000
  mir_sdr_hw_version=255
  serial=1903154196
```
Also I noticed that I get the `Failed to open sdr device matching 'driver=sdrplay'.` error only on Ubuntu, on Windows the program just stops after registering the decoding protocols:

```
(...)
Registering protocol [148] "DSC Security Contact (WS4945)"
Registering protocol [149] "ERT"
Registering protocol [151] "Visonic powercode"
Registered 123 out of 151 device decoding protocols

C:\Users\user>
```

zuckschwerdt, 2020-05-15:
Perhaps somethings else has a handle to the device open?

Did you compile a windows version with SoapySDR support? It's not included in our CI builds.

goscickiw, 2020-05-15:
I didn't compile, I just downloaded the binary build. If it doesn't have SoapySDR support that would explain why it doesn't work on Windows. But it still should work on Ubuntu...

zuckschwerdt, 2020-05-25:
I might get around to add SoapySDR support for the windows builds. But that's a lengthy process and might not happen too soon, sorry :/

goscickiw, 2020-05-26:
This isn't a Windows issue. Why is it not working on Ubuntu? What am I doing wrong?

zuckschwerdt, 2020-05-26:
Did you check that only a single SoapySDR version is installed? (esp. 0.7 vs 0.8)
I tested with `rtl_433 -d "driver=sdrplay"` on Debian Buster with SoapySDR 0.8 from source and it works fine.

goscickiw, 2020-05-26:
I did `sudo apt purge *soapy* &&  sudo apt autoremove`. It removed rtl_433 as well, so i reinstalled it with `sudo apt install rtl-433` and just let it install SoapySDR on its own.
Still getting the same error.

zuckschwerdt, 2020-05-26:
You are on Ubuntu 20.04 LTS "Focal"? So it's https://packages.ubuntu.com/focal/rtl-433 when you `sudo apt install rtl-433`? I don't see any "soapysdr0.7-module-sdrplay" in soapysdr0.7-module-all or soapysdr0.7-module. How did you add the support module?

goscickiw, 2020-05-26:
I use Ubuntu 19.10. I decided to reinstall the whole system, so I'm currently trying to create a bootable flash drive for 20.04 that will show up in UEFI boot menu, which seems to be extremely hard for some unknown reason (tried several ways).
On 19.10, when I install rtl-433, it automatically installs a bunch of soapysdr0.6 packages. I don't remember if it installed the sdrplay module (I'm writing from Windows because I'm not logged in to GitHub on Ubuntu), but SoapySDRUtil detects the device so it probably did.

zuckschwerdt, 2020-05-26:
If you still have Ubuntu 19.10 around, please post the output of `SoapySDRUtil --info`.

zuckschwerdt, 2020-05-26:
(on the UEFI issue, maybe using the [rEFInd usb image](https://www.rodsbooks.com/refind/getting.html) can help to boot some next stage. I use rEFInd to manage all my UEFI boots.)

goscickiw, 2020-05-26:
I solved the UEFI issue - for some reason the bootable drive showed up when i plugged it into a different USB port. No idea why this was a problem. I'm currently finishing the 20.04 installation.

zuckschwerdt, 2020-05-26:
Great to hear. As for the Ubuntu 19.10 install, I suspect that you installed SoapySDR 0.7 or 0.8 to get the sdrplay module and the rtl_433 linked with 0.6 but loading 0.7/8 can't work.

goscickiw, 2020-05-27:
I finally got the RSP1A working with rtl_433. On a freshly reinstalled Ubuntu, I did `sudo apt install rtl-433` and let it install SoapySDR automatically as before. Then I manually compiled and installed the SoapySDRPlay module. In the process I learned about checkinstall, which I think will make using Ubuntu less painful in the future. `rtl_433 -d driver=sdrplay` now starts normally and detects signals.

Now, how do I properly set gain? What do I put in the `-g` parameter? The help for `-g` only has a LimeSDR example.

zuckschwerdt, 2020-05-27:
Like listed above `Full gain range: [0, 48] dB` or component gains of `IFGR gain range: [20, 59] dB`, `RFGR gain range: [0, 9] dB`

closed 2020-05-27 by zuckschwerdt