merbanan/rtl_433 #1381: Date on output filename

Molder1233, 2020-05-15:
Hello.
I use the rtl_433 software and find it very interesting. 

I surch the way to add the date on the name of each output file. (Not on metadata)
Something  like 20200515_14h00m00s_g001_433.92M_250k.cu8
I may try to change the source code but don't know where it is...

Many thanks

zuckschwerdt, 2020-05-15:
The quickest route would be to use a script to just rename the files based on the filesystem timestamp on them. E.g.
```
for f in g*.cu8 ; do
    mv -n "$f" $(date -r "%Y%m%d_%Hh%Mm%Ss_$f")
done
```
If we were to add this option we'd likely use a format of `yyyymmdd"T"HHMMSS.uuu` and that still leaves the TZ open…

Molder1233, 2020-05-16:
Thank you for your answer.
I thought about it, but I didn't want another script to run just for that.
Do you know which C file determinate the name? 

> If we were to add this option we'd likely use a format of `yyyymmdd"T"HHMMSS.uuu` and that still leaves the **TZ open**…

Sorry for the next question but what means "TZ open"?

zuckschwerdt, 2020-05-16:
https://github.com/merbanan/rtl_433/blob/master/src/samp_grab.c#L104

TZ is timezone, if we write datetime information we might need to known wether it's GMT or Local.

Molder1233, 2020-05-16:
That's it.
Thank you so much.

closed 2020-05-16 by Molder1233