merbanan/rtl_433 #1385: Problem decoding Smoke detector

Chrominator, 2020-05-17:
Hi,

Got 2 smoke detectors with a raw value, don't know their status.

```
Raspi 2 on Ubuntu 
Distributor ID: Ubuntu
Description:    Ubuntu 20.04 LTS
Release:        20.04
Codename:       focal

rtl_433 version 20.02-55-gc1d1f9f branch master at 202005162227 inputs file rtl_tcp RTL-SDR
```

Command line : 
`rtl_433 -s 1024k -f 433.92e6`

[image]

[image]

Thank you for your support.

zuckschwerdt, 2020-05-17:
Likely a false positive. Best to use `-R -86`

closed 2020-05-17 by zuckschwerdt

gizmocuz, 2020-05-18:
@zuckschwerdt , I am afraid it is not a false positive in my location. I have various KD101 smoke sensors and they all report like above, but with a different code
Can I send you one or would it help if i log it and trigger the test button ?

zuckschwerdt, 2020-05-18:
You can run with `-A` and post the output. That way we can see the exact timing and data. It should have something like:
```
Detected OOK package	@0.082876s
Total count:  215,  width: 431.66 ms		(107916 S)
Pulse width distribution:
 [ 0] count:  127,  width:  432 us [432;452]	( 108 S)
 [ 1] count:   88,  width: 1196 us [1196;1200]	( 299 S)
Gap width distribution:
 [ 0] count:  104,  width: 1300 us [1288;1324]	( 325 S)
 [ 1] count:  103,  width:  516 us [476;532]	( 129 S)
 [ 2] count:    7,  width: 11760 us [11756;11764]	(2940 S)
Pulse period distribution:
 [ 0] count:  192,  width: 1728 us [1700;1756]	( 432 S)
 [ 1] count:   15,  width:  928 us [908;956]	( 232 S)
 [ 2] count:    7,  width: 12192 us [12188;12200]	(3048 S)
Level estimates [high, low]:  11925,      5
RSSI: -1.4 dB SNR: 33.8 dB Noise: -35.2 dB
Frequency offsets [F1, F2]:     691,      0	(+2.6 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
Attempting demodulation... short_width: 432, long_width: 1196, reset_limit: 11768, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=432,l=1196,r=11768,g=1328,t=304,y=0'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 8
[00] {27} c9 6a 7c e0 : 11001001 01101010 01111100 111
[01] {27} c9 6a 7c e0 : 11001001 01101010 01111100 111
...
```

Also, are those messages (simulated )alarms? Or is that some kind of periodic status?