merbanan/rtl_433 #1391: InfluxDB format stops working after a while

iandees, 2020-05-23:
I run rtl_433, influxdb, and grafana in Docker on my Synology. I recently updated my rtl433 image and set it up to write directly to my InfluxDB setup. I'm executing it like this:

```
docker run -d \
    --restart always \
    --name="rtl433_to_influxdb" \
    --privileged -v /dev/bus/usb:/dev/bus/usb \
    --link influxdb:influxdb \
    iandees/rtl433_to_influxdb rtl_433 -d 0 -F influx://influxdb:8086/write?db=telegraf -M time:unix:usec:utc
```

This works for a little while, and then it seems to stop writing anything to Influx:

[image]

Other data in the same Influx database continues collecting and showing fine. Restarting rtl_433 will usually cause data to start showing up again.

Neither InfluxDB or rtl_433 show anything interesting in the logs when data stops.

It doesn't look like the Influx output in rtl_433 has any debug logging. Do you have any suggestions for how I might narrow down what's causing this?

iandees, 2020-05-23:
I went back to running rtl_433 with `-F json` and piping it through an old Python wrapper for InfluxDB. That worked for a similar amount of time and then stopped updating. This indicates to me that the problem is with my SDR or with rtl_433, not the InfluxDB output.

iandees, 2020-05-23:
Running without `-F json` seems to let this run continuously without hanging.

zuckschwerdt, 2020-05-23:
So
- stops with `-F influx` and also `-F json` and script
- works with (implied) `-F kv`

Try `-F json` without the wrapper script. Try `-F influx` and also `-F json` plus script towards a dummy like [socat](http://www.dest-unreach.org/socat/) (`socat tcp-listen:1883 -`). What do you get?
Best guess: Influx stalls the connection.

iandees, 2020-05-23:
* `-F json` works
* `-F influx` towards `while true ; do nc -lkv -p 8086 -e echo -e "HTTP/1.1 200 OK\n\n $(date)"; done` in a loop seems to work
* `-F json` with my wrapper doesn't seem to work, but probably because `nc` isn't a real web server

So yea, I agree that something is going on with influx. I guess I'll bug them. Thanks for the debug help!

closed 2020-05-23 by iandees

iandees, 2020-05-23:
When I add a timeout to the HTTP call in my Python wrapper it seems to eventually run into a timeout. Something on the Influx side must be causing the request to take forever.

Does the rtl_433 Influx HTTP output have a timeout?

zuckschwerdt, 2020-05-24:
Perhaps you could log the TCP traffic and try to see where it stalls? E.g. `tcpdump -np port 1883`

iandees, 2020-05-24:
I'm not sure what's different today, but I eventually stop receiving any readings, even while running with `-F kv`.

zuckschwerdt, 2020-05-24:
My first action in these instances is to look at the spectrum. Try CubicSDR, Qqrx, or SDRangel.
- Is there a signal? Is it strong?
- Is it in the assumed frequency band?
- Are there collisions in that band?

rct, 2020-05-24:
So rtl_433 is hanging, what state does `ps` say it is in?
- Any output on STDERR?  Are you capturing STDERR? 
- Anything in dmesg?
- When it seems to be hung, can you try `strace -p ... ` to see if it's blocked in a system call?
- Try enabling the periodic stats dump. Whether it still dumps stats periodically might give some clues about whether rtl_433 is hung or if it's something related to the rtl-sdr stick or the rf environment.

iandees, 2020-05-24:
The periodic stats dump shows that it receives messages normally and then for whatever reason stops:

https://gist.github.com/iandees/755ffa40cfb90c331b0d8bfba4e63807

This is running on a Synology machine and I haven't found a great way of installing strace yet.

iandees, 2020-05-24:
Ok, I got rtl_433 into a state where it's not receiving any data, got `strace` installed on the container, and strace gives me output like this:

```
ioctl(9, USBDEVFS_SUBMITURB, 0x55cbbd02f400) = 0
ioctl(9, USBDEVFS_REAPURBNDELAY, 0x7ffe07595e40) = -1 EAGAIN (Resource temporarily unavailable)
poll([{fd=6, events=POLLIN}, {fd=8, events=POLLIN}, {fd=9, events=POLLOUT}], 3, 1000) = 1 ([{fd=9, revents=POLLOUT}])
ioctl(9, USBDEVFS_REAPURBNDELAY, 0x7ffe07595e40) = 0
setitimer(ITIMER_REAL, {it_interval={tv_sec=0, tv_usec=0}, it_value={tv_sec=3, tv_usec=0}}, {it_interval={tv_sec=0, tv_usec=0}, it_value={tv_sec=2, tv_usec=477872}}) = 0
ioctl(9, USBDEVFS_SUBMITURB, 0x55cbbd02f460) = 0
ioctl(9, USBDEVFS_REAPURBNDELAY, 0x7ffe07595e40) = -1 EAGAIN (Resource temporarily unavailable)
poll([{fd=6, events=POLLIN}, {fd=8, events=POLLIN}, {fd=9, events=POLLOUT}], 3, 1000) = 1 ([{fd=9, revents=POLLOUT}])
ioctl(9, USBDEVFS_REAPURBNDELAY, 0x7ffe07595e40) = 0
setitimer(ITIMER_REAL, {it_interval={tv_sec=0, tv_usec=0}, it_value={tv_sec=3, tv_usec=0}}, {it_interval={tv_sec=0, tv_usec=0}, it_value={tv_sec=2, tv_usec=474475}}) = 0
ioctl(9, USBDEVFS_SUBMITURB, 0x55cbbd02f4c0) = 0
ioctl(9, USBDEVFS_REAPURBNDELAY, 0x7ffe07595e40) = -1 EAGAIN (Resource temporarily unavailable)
poll([{fd=6, events=POLLIN}, {fd=8, events=POLLIN}, {fd=9, events=POLLOUT}], 3, 1000) = 1 ([{fd=9, revents=POLLOUT}])
ioctl(9, USBDEVFS_REAPURBNDELAY, 0x7ffe07595e40) = 0
setitimer(ITIMER_REAL, {it_interval={tv_sec=0, tv_usec=0}, it_value={tv_sec=3, tv_usec=0}}, {it_interval={tv_sec=0, tv_usec=0}, it_value={tv_sec=2, tv_usec=476413}}) = 0
ioctl(9, USBDEVFS_SUBMITURB, 0x55cbbd02f520) = 0
ioctl(9, USBDEVFS_REAPURBNDELAY, 0x7ffe07595e40) = -1 EAGAIN (Resource temporarily unavailable)
poll([{fd=6, events=POLLIN}, {fd=8, events=POLLIN}, {fd=9, events=POLLOUT}], 3, 1000) = 1 ([{fd=9, revents=POLLOUT}])
ioctl(9, USBDEVFS_REAPURBNDELAY, 0x7ffe07595e40) = 0
setitimer(ITIMER_REAL, {it_interval={tv_sec=0, tv_usec=0}, it_value={tv_sec=3, tv_usec=0}}, {it_interval={tv_sec=0, tv_usec=0}, it_value={tv_sec=2, tv_usec=475460}}) = 0
ioctl(9, USBDEVFS_SUBMITURB, 0x55cbbd02f580) = 0
ioctl(9, USBDEVFS_REAPURBNDELAY, 0x7ffe07595e40) = -1 EAGAIN (Resource temporarily unavailable)
poll([{fd=6, events=POLLIN}, {fd=8, events=POLLIN}, {fd=9, events=POLLOUT}], 3, 1000) = 1 ([{fd=9, revents=POLLOUT}])
ioctl(9, USBDEVFS_REAPURBNDELAY, 0x7ffe07595e40) = 0
setitimer(ITIMER_REAL, {it_interval={tv_sec=0, tv_usec=0}, it_value={tv_sec=3, tv_usec=0}}, {it_interval={tv_sec=0, tv_usec=0}, it_value={tv_sec=2, tv_usec=475873}}) = 0
ioctl(9, USBDEVFS_SUBMITURB, 0x55cbbd02f5e0) = 0
ioctl(9, USBDEVFS_REAPURBNDELAY, 0x7ffe07595e40) = -1 EAGAIN (Resource temporarily unavailable)
poll([{fd=6, events=POLLIN}, {fd=8, events=POLLIN}, {fd=9, events=POLLOUT}], 3, 1000^Cstrace: Process 72 detached
 <detached ...>
```

iandees, 2020-05-24:
I don't see anything interesting in `dmesg`, by the way.

iandees, 2020-05-25:
This might be a hardware problem. I have a mini USB hub with two of the "nano" SDRs plugged in. I just switched over to using the other one (it was previously pulling in water usage data from my 900MHz water meter) and things seem to be running much more smoothly.

zuckschwerdt, 2020-05-25:
That's strange. If the strace shows stalled USB packets (I'm not familiar with failure modes here) then rtl_433 should abort with the watchdog (no rf data received for 3 secods -> abort).
Perhaps you can stick a `printf` in rtl_433.c's sdr_callback() and watch if rf data still arrives?
https://github.com/merbanan/rtl_433/blob/master/src/rtl_433.c#L293
`fprintf(stderr, ".\n");` or if you want to see something change maybe the current timestamp:
`fprintf(stderr, "%lld\n", (long long)demod->now.tv_sec);`