merbanan/rtl_433 #1392: ct solution

Alf-deb, 2020-05-23:
I have 2 sensors here sold under the same name "TFA 30.3197". They are however distinguished by protocol:
old one: Rubicson (02), frequency 434.0346 MHz
new one: Solight (85), frequency 433.9916 MHz.
So they are 43 kHz apart.
The interference problem arises because they both have the same repetition rate of 57 sec. and probably also because the protocols are very similar. Currently the both overlap when sending their temperature message. This can be seen in the attached (Sensors-interferce.jpg) waterfall diagram as recorded by "gqrx". In this situation only one sensor is decoded - generally the first detected (here the Rubicson), while I am unable to decode the second one (here Solight). To double check I just wrapped the Rubicson sensor in aluminum foil - and immediately rtl_433 picks up and decodes the Solight sensor! Signal/Noise ratio for both sensors is good, generally 25 - 30db.

A perfect solution would be to apply a software filter of 4 to 10kHz bandwidth, i.e. restrict the scan range in frequency to a few kHz around a given center frequency. This would greatly reduce RF-interference. A notch filter for the undesired signal could probably help too.

To work around for a while I can reset one sensor (remove battery for short time) to bring the two out of sync, but sooner or later (maybe weeks) they will drift and reach sync again.

Or is there any such possibility already available in rtl_433 and just not documented?

Best regards,
Ingo

[image]

merbanan, 2020-05-23:
They will drift out of sync eventually but the problem you describe here has no real solution. Adding a filter is not really possible either as there is no way to guarantee that the frequency isn't perfectly matching. The only way to have something semi reliable is if the timings don't sync up. That way you will get a signal 99% of the time and that is good enough.

But you are onto something, it should be possible to filter the signal so only the needed frequency is received. Then you can tune to another frequency and then repeat the cycle. Anyway this is nothing I have the time to try and implement. There are to many unknowns for it to work out of the box.

Alf-deb, 2020-05-23:
@membanan:
Thanks for your comment - so I'll have to arrange with the situation somehow.

The idea only came up to me because in the documentation says that a scan rate (parameter "-s", default 250 000 Hz) causes rtl_433 to cover a range of 250kHz around the center frequency (parameter "-f", default 433.92MHz). I tried to set center frequency for my device and reduce scan rate to 10kHz - but that does not work.

merbanan, 2020-05-23:
You could tune so that only one signal is not part of the spectrum and then cycle it for 5 minutes each.

merbanan, 2020-05-23:
250ksps is the lowest you can go. My point was that you can tune to the "left" so that the Rubicson signal is not part of the reception range. And then tune to the "right" so that the Solight sensor is not part of the range. That should work but will require that there are no other stationary signal in that range.

Alf-deb, 2020-05-24:
I understand what you mean, however I doubt that the edges of the tuning range can be set with the required precision and stability over time. The frequency difference of the 2 signals is just 48kHz, I'll need at least 4kHz of bandwidth for reception and an attenuation of say 20db which in my opinion only can be achieved with a well designed filter without major signal distortion.

The good news:
I have checked the "adjustment" of the 2 signals right now (1 day later then the graph above) and the Rubicson sensor has fallen roughly 0.3 seconds behind (signal/message duration is 1s). Repetition rate of messages is 57 seconds - so it will take some 200 days! until the coincidence repeats. I just have reset the Rubicson sensor so it is some seconds behind now, giving me freedom for rest of the year :-)

Hopefully I remember that issue when it happens again or I have to change batteries. For now I'll consider it "solved".

Many thanks for your input.

closed 2020-05-25 by zuckschwerdt