merbanan/rtl_433 #1393: Some help decoding RadioHead ASK payloads

ant-thomas, 2020-05-24:
I was wondering if someone could give me some help decoding some RadioHead ASK payloads?

I already have a working Arduino to Arduino/ESP8266 setup but I'd like to use rtl_433 instead as I think it will work better. These are homemade temperature sensors.

I know the transmitted data, and have the rtl_433 received payloads, but I want to be able to decode them

Section of Arduino code defining the payloads
`struct SensorData { 
  float ds18b20_t;
  long battery;
  char* address;
};

SensorData sensor;`

Transmission of payload
`radio.send((uint8_t*)&sensor, sizeof(sensor));`

Some data examples

Payload   : 0, 0, 108, 65, 60, 16, 0, 0, 97, 0
temp=14.75
bat=4156
adr=a 

Payload   : 0, 0, 63, 65, 30, 16, 0, 0, 111, 0
temp=11.94
bat=4126
adr=o 

Payload   : 0, 0, 157, 65, 98, 19, 0, 0, 98, 0
temp=19.62
bat=4962
adr=b 

Payload   : 0, 0, 144, 65, 208, 17, 0, 0, 108, 0 
temp=18.00
bat=4560
adr=l

I've figured out that the 9th number is the ASCii code for the address - which is a unique letter that each sensor transmits to define itself.

If anyone wants to help and need more info I can provide it.

zuckschwerdt, 2020-05-24:
Looks like the struct is right, it's a `float`, a `long` (for 32 bit or LLP64 platform, in other words `int`) with Little-Endian encoding, the a `char` (single byte).
What are you programming in? Unpack as those types. (btw. using a float really is bad design, it should have been a scaled integer. also the battery field could reasonably have been 16 bit i.e. `short`).

ant-thomas, 2020-05-24:
Thanks for the info. I guess my plan was to try and modify rtl_433 in some way to either output the data in a readable format to a file or directly to a mysql server. 

Thanks also for the other tips, something I will definitely look at once I've got the decoding figured out.

zuckschwerdt, 2020-05-24:
Quick and dirty this should do for C:
- use the struct but change `long` to `int32_t` and `char*` to `char`
- get a typed ref: `struct SensorData *data = (struct SensorData *)&rh_payload[5];`
- then e.g. `data->ds18b20_t`

Generally the struct should have known fixed size fields, esp. not `long`. For the temperatur it's best to use a fixed-point integer, e.g. `int16_t` with a scale of 100 (depending on your needs) i.e. `sensor.ds18b20fp = realtemp * 100` (scaled range -327.68 to 327.67). And a ASCII character is `char` not `char *`.

ant-thomas, 2020-05-25:
Thanks for the pointers. I've decided to have rtl_433 output to a file and manipulate the data using a relatively simple bash script (that's my current ability!)

So far I've decoded the char address and battery value, just need to figure out the conversion for the temperature

Payload : 0, 0, 108, 65, 60, 16, 0, 0, 97, 0
temp=14.75
bat=4156
adr=a

60, 16 being the 8bit to 16bit conversion for battery 4156

Would you be able to give some pointers to convert the `0, 0, 108, 65` to `14.75` ?

zuckschwerdt, 2020-05-25:
The float is [IEEE 754](https://en.wikipedia.org/wiki/IEEE_754) -- there is no easy way to handle this. If the float is the first thing in the payload, i.e. byte 5 then:
`float temp = *((float *)&rh_payload[5]);` should work.

zuckschwerdt, 2020-05-25:
Feel free to discuss further, closing because there is nothing to change rtl_433.

closed 2020-05-25 by zuckschwerdt

ant-thomas, 2020-05-26:
Got it figured out now.
Converted the first 4 parts of the payload to binary, ordered them back to front then run them through a simple binary to IEEE 754 decimal converter and I now have all the correct temperature

Thanks for the pointers, very helpful.

ant-thomas, 2021-01-01:
I'm in the process of changing my sketch to reduce payload size as advised. Down to 5 from 10.

Just trying to figure out decoding of the short/int/int16_t for temperature.

I've rounded temp to 2dp and then * 100 to get it within the signed 16bit range (-32768 to 32767)
Tested with positive temperature and the output/payload is expected and converts correctly.

I've tried to fake some negative results and the payload doesn't look as expected.
Eg - I've sent ``-356`` (which would be a conversion of -3.56C)
The two parts of the payload comes out as ``54; 53`` - whereas I'd expect a negative signed payload, unless I don't understand the encoding/decoding (entirely possible!)
I believe I should have a payload of ``-2; -100``

What should the payload look like?

The sensor is currently in the freezer so I can attempt to get a true negative reading out.

ant-thomas, 2021-01-01:
Figured it out. Payloads are 8bit unsigned, conversion works correct now I've realised that bit, freezer temps are converting as they should. Good stuff.

zuckschwerdt, 2021-01-01:
Sounds good. If you collect the bytes (uint8_t) then you cann perform the conversion implictly:
`int temp_raw = (int16_t)((b[0] << 8) | (b[1]));` -- depending on byte order (endianess) of course.