merbanan/rtl_433 #1395: OS-BTHGN129 is missing comfort & forecast information

ROV28, 2020-05-25:
LS,

the Oregon Scientific BTHGN129 has also comfort and forecast information, but that is not shown at the output.

I think it is like this:

`unsigned int comfort = msg[7] >>4;`
`char *comfort_str="Normal";`
`if      (comfort == 4)   comfort_str = "Comfortable";`
`else if (comfort == 8)   comfort_str = "Dry";`
`else if (comfort == 0xc) comfort_str = "Humid";`
`unsigned int forecast = msg[9]>>4;`
`char *forecast_str="Cloudy";`
`if      (forecast == 3)   forecast_str = "Rainy";`
`else if (forecast == 6)   forecast_str = "Partly Cloudy";`
`else if (forecast == 0xc) forecast_str = "Sunny";`

Can this be added?

Best regards,
Robert

zuckschwerdt, 2020-05-25:
The code is from https://github.com/merbanan/rtl_433/blob/master/src/devices/oregon_scientific.c#L276 ?
Not sure why it's disabled, I guess it's a very basic lookup on humidity and temperature and that way of no real value?

gdt, 2023-09-25:
Is there anything to do?   It's not clear to me that we should output things that are not really in the packet, when a console is doing some interpretation like this.

ROV28, 2023-09-26:
The info is in the packet but not displayed any more. So it is not an interpretation of the console.

gdt, 2023-09-26:
Then it makes sense.   Please code, test and open a PR. 

As a fine point, don't default to Normal/Cloudy but match on the defined codepoint, and set "invalid" or some such for the string when there is an unexpected value.

gdt, 2024-08-04:
@ROV28 We are awaiting a PR.

ROV28, 2024-08-06:
I know the changes for "oregon_scientific.c", but how it is processes and ends up in the output, I have no clue... (It is beyond my skills).

zuckschwerdt, 2024-08-06:
The code was never removed but is part of another device, BHTR968:
https://github.com/merbanan/rtl_433/blob/6ebad62cdc874dac5ce57e19892d4dd8c8a011c5/src/devices/oregon_scientific.c#L276-L285

The code was added 10 years back in 8f81020, but always known to be a very basic lookup on humidity and pressure/temperature and that way of no real value.

We could output the same guess on comfort and forecast for any device with humidity or pressure/temperature reading.
I see it like putting a green/orange/red backdrop on a value gauge -- it's not data but a choice of display.

We could add something like this as a general case to an MQTT script, maybe the HASS bridge, but not as a special case just for one sensor that is a bit odd.

closed 2024-08-06 by zuckschwerdt