merbanan/rtl_433 #1396: Sample rate per frequency

fmuntean, 2020-05-26:
I am not able to set the sample rate for each of the frequencies provided 
I am using the rtl_433 with multiple frequencies   -f 315M -f 433.92M -f 915M
The problem is that I can't set different sample rate for each of the passed frequencies.
Can you add support for this please?
for example would like to use something like this -f 315M:250k -f433.92M:250k -f 915M:1M
aka -f <freq>:<sample rate>
this is due to the fact that it seems that the tool is not able to decode at 1M sample rate correctly on the lower frequencies.
Thank you

evilpete, 2020-06-22:
@fmuntean it's a little more complicated then what you are requesting because the you may need to explicitly set demodulator..

Normally the  minmax demodulator is automatically used for 800Mhz and above and "classic" is used below.

Q: Is the sample FSK based ? if so you may want to try minmax using  the option`-Y minmax`

merbanan, 2020-06-27:
A higher sample rate will give bigger frequency receive window. Most likely a stationary signal is disturbing the reception at lower bands. But yeah I think we need to change the sample rate per band for frequency hopping to work optimally. Right now the solution is to get another device.

sheilbronn, 2021-11-04:
I have been trying to use frequency hopping, too: For 433 Mhz and 868 Mhz.
However, signals on the 433 Mhz band are only decoded if I use a sample rate of 256 S/s, and on 868 Mhz band only, if I set to 1024 S/s. So it seems, that for me the necessity to for different sample rates is quite high, if frequency hopping is to become meaningful. (BTW: I'm running rtl_433 on a Rasperry Pi 4)
Thanks!

zuckschwerdt, 2021-11-05:
Yes, something like a whole "profile" per hop would be nice. Not much to code but the argument syntax needs some thought. S.a. #1838

zuckschwerdt, 2021-11-05:
Since we do have a HTTP API, it may be simpler to have a "controller" script where you can code the rules.
"On X (time, reception, ...) do Y (set freq, rate, protocols, ...)".

rogovst, 2021-12-29:
Want it too