merbanan/rtl_433 #1399: Is there a way to send LF signal to activate the TPMS sensor out of tire?

soyoeml, 2020-05-30:
Hi,

This is not a issue. I just try to find a way to send LF signal to activate the TPMS sensor out of tire. I have four pacific PMV-107J sensors out of tire. But they don't send signal if they're not in the tire. Does this software tool have a way to send LF signal to them and let them transmit signal? I know some TPMS scanners have a way to scan TPMS sensor. Could we send the signal as TPMS scanners do?

zuckschwerdt, 2020-05-30:
You need hardware for that. Not all SDR have TX capability. Those that do can't send LF. But you can just build or buy an LF transmitter -- no SDR functionality needed.

closed 2020-05-30 by zuckschwerdt

soyoeml, 2020-10-16:
I found I don't need send LF signal to let it send data. When I move my PMV-107J sensor, it keeps sending data out. 

Another way to let it send data is using a paper clap to clap the sensor. It will let it send out about 40Kpa press on it. thanks.