merbanan/rtl_433 #1400: Yarn docs:build

gvanem, 2020-06-03:
I tried to build some documentation based on `docs/build-docs.sh`, but on Windows.
So I created this `docs/bild-docs.bat`:
```bat
@echo off
sed 's/docs\///' ..\README.md > README.md
copy ..\CHANGELOG.md
copy ..\rtl_433_tests\README.md TESTS.md
copy ..\vs15\README.md          VS15.md
cmd /c yarn.cmd install
cmd /c yarn.cmd docs:build
```
which seems to work:
```
wait Rendering static HTML...
success Generated static files in .vuepress\dist.
``` 

But the generated `.vuepress\dist\index.html` looks like shit in Chrome. 
It seems no links to `.vuepress\dist\assets\*` are correct.
E.g. the `index.html` has a:
```
 <link rel="stylesheet" href="/rtl_433/assets/css/0.styles.1b02b6fa.css">
```
Who/what is setting an absolute path here?

Editing `index.html` and removing all `/rtl_433/`, kinda works.

*PS*. How can I avoid getting 802!! packages installed under `docs\node_modules`?
I'd like Yarn or Node to put these in some global package-dir. But then again, I'm a Node *n00b*.

zuckschwerdt, 2020-06-03:
That path is from `base: "/rtl_433/",` in https://github.com/merbanan/rtl_433/blob/master/docs/.vuepress/config.js
The docs are meant to go to https://triq.org/rtl_433/
All the node_modules are a dependency of https://vuepress.vuejs.org/ -- it's an easy way for me to get the various READMEs pretty and functional on the web.

gvanem, 2020-06-03:
> The docs are meant to go to https://triq.org/rtl_433/

Okay, I see. This issue was more an exercise in Node, Yarn and Vuepress. They (yarn?) creates a lot of dependencies under `docs\node_modules` which seems odd.

zuckschwerdt, 2020-06-03:
Yarn is mostly just a "better" NPM. And the ridiculous amounts of dependencies is just the state-of-art with nodejs. Writing your own "helpers" (even if it is single-line functions) is not considered good practice with JS folks -- I guess it's because you need to dig rather deep to get even simple appearing stuff right when JS and the state of browsers is full of quirks. There are serious WTF modules like this: [Check if something is a positive number](https://github.com/kevva/is-positive)

zuckschwerdt, 2020-06-03:
A good reality check for JS is [Wat](https://www.destroyallsoftware.com/talks/wat). Then give up and just import All-the-Things from NPM ;)

gvanem, 2020-06-03:
Okay, I utterly confused and tired of with the whole Node monster:
```
npm install -g is-positive
<created a test-is-positive.js from the example>
node test-is-positive.js
```
that spits out these error with ANSI-crap included:
```
Require stack:
- F:\gv\dx-radio\RTL_433\docs\test-is-positive.js
←[90m    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:892:15)←[39m
←[90m    at Function.Module._load (internal/modules/cjs/loader.js:785:27)←[39m
←[90m    at Module.require (internal/modules/cjs/loader.js:956:19)←[39m
←[90m    at require (internal/modules/cjs/helpers.js:74:18)←[39m
    at Object.<anonymous> (F:\gv\dx-radio\RTL_433\docs\test-is-positive.js:1:20)
←[90m    at Module._compile (internal/modules/cjs/loader.js:1063:30)←[39m
←[90m    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1103:10)←[39m
←[90m    at Module.load (internal/modules/cjs/loader.js:914:32)←[39m
←[90m    at Function.Module._load (internal/modules/cjs/loader.js:822:14)←[39m
←[90m    at Function.Module.runMain (internal/modules/cjs/loader.js:1143:12)←[39m {
  code: ←[32m'MODULE_NOT_FOUND'←[39m,
  requireStack: [ ←[32m'F:\\gv\\dx-radio\\RTL_433\\docs\\test-is-positive.js'←[39m ]
}
```
Nice!

I guess there is no point for me pursuing Node any more.

gvanem, 2020-07-06:
> I guess there is no point for me pursuing Node any more.

So let me close it.

closed 2020-07-06 by gvanem