merbanan/rtl_433 #1401: Acurite 06045M Decoded Temperature Reading Low

Stormtracker017, 2020-06-04:
I have a node set up in Node Red that displays several 06045M. I noticed the decoded value for the temperature is about 1.8 F to low compared to reading on my console. The humidity matches the console. As for distance of lightning strikes, from what  I saw is the distance data is being decoder closer then what the console reports.  I don't have enough data to compare.  The console I'm using is a 06047M.

Mindavi, 2020-08-25:
I don't think this can be resolved without packet traces. If you'd like this resolved add packet traces and the expected values (those visible on the console) with them. Then someone can take a stab at it. Without a way to check if the values are correct/incorrect, this issue is hard to resolve.

There is [some data](https://github.com/merbanan/rtl_433_tests/tree/bf2aa0aebbe75e331b1f1b512001a53920c86c4a/tests/acurite/Acurite_6045M) available in the `rtl_433_tests` repo, but if I may believe the comments only the humidity and strike count is verified, the others are uncertain. So if you could provide samples with a description of all expected values, that would help figuring this out a lot. They could then also be added to the tests repository for reference.

rct, 2020-08-27:
Comparing the values from rtl_433 to the values captured from the Acurite Access, it looks like the the decoded temperature_f is -2.0 F. 

The Access values are only sent every 5 minutes, so work needs to be done to try to do a more rigorous comparison.  However, just checking the high and low temperatures for the last two days shows a 2.0 F difference.

rct, 2020-08-27:
Here's my data in case anyone wants to do into it.

[mya-6045-temp-20200826.zip](https://github.com/merbanan/rtl_433/files/5139044/mya-6045-temp-20200826.zip)

rct, 2020-08-27:
temperature from Acurite Access
[image]

temperature from rtl_433
[image]

rct, 2020-08-27:
@Stormtracker017 - what sort of temperature range did you notice a difference of 1.8 F over?

I'm seeing a -2.0F difference over a range of 67.6 - 91.9 F (24.3)

It would be easy to say the difference between the -2.0 F I'm seeing versus the -1.8 F you are seeing is a rounding error, but I'd like to be sure there isn't something else going on.

Fixing this could be as simple as changing the offset by -200.
https://github.com/merbanan/rtl_433/blob/cf4f4d8e37400edd2f5c26e58c2833cd77adc732/src/devices/acurite.c#L357

PR #1482 (2020-08-30): Acurite 6045 fix temperature 2.0F too low
Fixes #1401 

* 6045 documentation clean up
* exception check clean up
* deleted old debugging output that should no longer be necessary
commit 1c0ac486 (2020-08-30): Fix Acurite 6045 temperature 2.0F too low (#1482)

Fixes #1401
closed 2020-08-30 by Mindavi