merbanan/rtl_433 #1402: Inverted data from hackrf

evilpete, 2020-06-07:
when I used my hackrf with command :

```
src/rtl_433 -f 915M \
        -s 1000K \
        -d driver=hackrf \
        -R 154 -R 76 \
        -t bandwidth=2500000  \
        -g "LNA=40,AMP=14,VGA=30" \
        -Y classic \
        -vvv
```

I noted I was not receiving the LaCrosse nor Insteon data I normally receive, on further investigation ( and a few extra debugging printf's ) it is clear the demodulated data is inverted .

SoapySDRUtil reports:
```
Found device 1
  device = HackRF One
  driver = hackrf
  label = HackRF One #0 457863c82f74521f
  part_id = a000cb3c006a4337
  serial = 0000000000000000457863c82f74521f
  version = 2018.01.1
```

Note: the bandwidth and gain settings used were _total_ guess work but needed to received any kind of data from the hackrf

[hackrf-rtl_433.zip](https://github.com/merbanan/rtl_433/files/4741284/hackrf-rtl_433.zip)

zuckschwerdt, 2020-06-07:
Are perhaps I/Q swapped? Maybe you can try to swap them in rtl_433 and see if that fixed it?

evilpete, 2020-06-07:
I assumed I/Q swapping also, i guess the question is where..

I'll play around and try to locate the cause and determine if it is a Soapy layer or hackrf issue.
  an option to swap I/Q may be a needed future option

gdt, 2023-09-25:
@evilpete status?

evilpete, 2023-10-02:
havent  looked into  it  / used my hackrf in a while,  i'll try to retest / verify within a week or two

gdt, 2025-07-04:
@evilpete Could you retest with all up-to-date code?  I am not sure this issue being open is useful.