merbanan/rtl_433 #1403: Filtering a signal?

brianmichalk1, 2020-06-08:
I am trying to decode data from the Ecowitt WH51.  There is a decoder, but it's not working for me, so I decided to try my luck at decoding the signal.  I have Universal Radio Hacker, and can analyze it there after acquiring with this:
rtl_433 -S unknown -f915M -g 1
I have to filter the signal in URH, and it gives me good looking data:

[image]

When I try to analyze the signal in rtl_433, I get:

```
root@michalk-desktop:~/bkm1# rtl_433 -R 0 -X 'n=new,m=FSK_PCM,s=100,l=100,r=2000' g001_915M_1000k.cu8 
rtl_433 version 20.02-61-gf82c025 branch master at 202005272108 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Registered 1 out of 153 device decoding protocols [ ]
Test mode active. Reading samples from file: g001_915M_1000k.cu8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.259660s
model     : new          count     : 1             num_rows  : 1             rows      : 
len       : 435          data      : 3398c639ce7318c6318c6318c63018f9fcc603180c00007c07ff9f0003000ffff00001ffe0001fcfffffffffffffff18c0631f0300430
codes     : {435}3398c639ce7318c6318c6318c63018f9fcc603180c00007c07ff9f0003000ffff00001ffe0001fcfffffffffffffff18c0631f0300430
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.416072s
model     : new          count     : 1             num_rows  : 1             rows      : 
len       : 434          data      : 3318c6318c6318c6318c6398c63018f9fcc603180c00007c07ff9f0003000ffff80001ffe0001fcfffffffffffffff1cc0733e0e00864
codes     : {434}3318c6318c6318c6318c6398c63018f9fcc603180c00007c07ff9f0003000ffff80001ffe0001fcfffffffffffffff1cc0733e0e00864

```
It seems like I need to enable a filter in rtl_433.  I am also looking for more examples on how to use the flex decoder.

zuckschwerdt, 2020-06-08:
The sample is at 1000 rate, you need to add that parameter when reading the sample back in:
`rtl_433 -s 1000k -R 0 -X 'n=new,m=FSK_PCM,s=100,l=100,r=2000' g001_915M_1000k.cu8`

brianmichalk1, 2020-06-08:
Thank you!!  I'm getting close with this string:
codes     : {184}5555555555516ea288063ec083f807c077fffffea2b10a
It begins off by one bit.  Where can I read more about the flex decoder to trim out the preamble and then the sync (2dd4) characters?

brianmichalk1, 2020-06-08:
Okay.  It looks like I have it.
```
rtl_433 -s 1M -R 0 -X 'n=new,m=FSK_PCM,s=100,l=100,r=2000,preamble={32}aaaa2dd4' g001_915M_1000k.cu8
```
gives
```
time      : @0.064915s
model     : new          count     : 1             num_rows  : 1             rows      : 
len       : 123          data      : 5100c7d8107f00f80effffffd456214
codes     : {123}5100c7d8107f00f80effffffd456214
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.104018s
model     : new          count     : 1             num_rows  : 1             rows      : 
len       : 123          data      : 5100c7d8107f00f80effffffd456204
codes     : {123}5100c7d8107f00f80effffffd456204

```

brianmichalk1, 2020-06-08:
I've narrowed the problem down to the decoder.
this does not decode:
rtl_433 -y "{167}aaaaaaa2dd45100c7d8107f00f80effffffd456204"
but this does decode:
rtl_433 -y "{163}aaaaaaa2dd45100c7d8107f00f80effffffd45620"

merbanan, 2020-06-08:
I bet this line eats you messages:

if (bitbuffer->bits_per_row[0] < 136) {  // Minimum length 14 bytes data + 3 bytes preamble

https://github.com/merbanan/rtl_433/blob/e5bdd68f129c3bc69b75d7797ceef376fb801ea5/src/devices/fineoffset.c#L557

brianmichalk1, 2020-06-08:
You are correct.  I am testing now.

merbanan, 2020-07-03:
Increased to 184 bits. Should be fixed with aa7436424394bc5eb0a5b85f827ac4e9d26ad69e

closed 2020-07-03 by merbanan