merbanan/rtl_433 #1405: Add suport for Chacon On/Of remote plug (54663)

rosorio, 2020-06-13:
Hi,
I'm trying to analyze the radio signal sent to turn on:off  new Chacon remote plugs,
but at this point I didn't succeed. As indicated in the CONTRIBUTING page,
I attach 4 text files for the corresponding signals:

- Button A ON: [btAon.txt](https://github.com/merbanan/rtl_433/files/4775172/btAon.txt)
- Button A OFF: [btAoff.txt](https://github.com/merbanan/rtl_433/files/4775171/btAoff.txt)
- Button B ON : [btBon.txt](https://github.com/merbanan/rtl_433/files/4775174/btBon.txt)
- Button B OFF : [btBoff.txt](https://github.com/merbanan/rtl_433/files/4775173/btBoff.txt)

I also have a wave capture of the signal Button A ON done with GQRX: [wav](http://files.osorio.me/misc/chacon/BT1_sample.wav)

I don't have too much experience in radio analyze, so feel free to ask any additional information, test, whatever I can do.

Best regards;
-- rodrigo

zuckschwerdt, 2020-06-13:
The file `BT1_ON.cu8` is more interesting. The text output only contains the decoded messages -- and rtl_433 doesn't know how to handle your remote, yet.
You can run `rtl_433 -A BT1_ON.cu8` to get a rough analysis. Drop the cu8 on https://triq.org/iqs/ to visualize the sample (like the Gqrx waterfall).
Demodulate the pulses with `rtl_433 -w BT1_ON.ook BT1_ON.cu8` and drop the resulting file on https://triq.org/pdv/ to inspect the timing and pulses.

rosorio, 2020-06-13:
@zuckschwerdt Thanks for the hints, even If don't really know how to interpret or use the results :/

Just in case I put all the cu9 files in [my share](http://files.osorio.me/misc/chacon/) , including the ook obtained, just in case they can help.

I'm trying to understand or just mimic the protocol since a week without success, and using different methods and tools. At some point I'm able to interact with the plug and force a 'ON' but this happens too randomly (30/40 seconds to turn on the plug) to say I succeed.

zuckschwerdt, 2020-06-13:
It's OOK. Visible in the I/Q Spectrogram as kind of "morse code" line. The Pulseview can properly decode this.

Look like there are actually two different codes or protocols here: The first 4 packets are:
0 and 1 databits as 1200/400µs / 400/1200µs pulse/gap. And a 1000µs gap at the end.

The next 4 packet are: start with a sync of 3000µs pulse, 7000µs gap. Then follow 0 and 1 databits as 500/1000µs / 1000/500µs. There is no pause or gap after a packet, the next packet sync start immediatly. A transmission seems to be those 8 repeats.

This is likely for compatibility to target an older and newer receiver.
You can either analyze all different samples like this and find some pattern, or just replay those ook files using the timing to switch your transmitter on/off (or the cu8 if you got the hardware -- HackRF, Pluto, LimeSDR).

rosorio, 2020-06-14:
Thanks you so much @zuckschwerdt for your help. The fact we can have here 2 different protocols for compatibility reasons inspires me a lot, and helps me to find the correct sequence.  I focus on the first one and now I'm able to turn on the switch at every single message.

I don't have HackRF like card, so I have to do it with my arduino, a 433 emitter and the RCSwitch library. The protocol 3 is the one who looks the most close to my signal except for the sync bit, so I create a new protocol <ith the following properties:

pulse 350
sync bite : 4,22
0 waveform: 4,11
1 waveform 9,6
inverted signal= false

and sending the following bites does the job: 101101111011011010101100

Now I have to do the same job for the other buttons looking for a logic, and probably come back with an MR to support my devices.

Thanks again for your help.

zuckschwerdt, 2020-06-14:
Glad I could help. You can use `rtl_433 -X 'n=Chacon,m=OOK_PWM,s=450,l=1100,y=3000,g=1600,r=8000' BT1_ON.cu8` to get all codes decoded.
Then put the codes in [BitBench](https://triq.net/bitbench) to [view the bits](https://triq.net/bitbench?c=49a3e3&f=4b%20).

gdt, 2023-09-29:
Is this something that should lead to a PR?  Or is it just done?

gdt, 2024-08-04:
@rosorio What is the status of this issue?