merbanan/rtl_433 #1406: Add support for Kerui WD51 Water leak sensor

Nauttor, 2020-06-15:
Hi, 

I have several Kerui sensors working with rtl_433, various are already supported, like the door reed and PIR, but I have 2 new devices that are not supported. 

One of them is a Kerui Water leak sensor, model WD51 and the other a Kerui mini Pir P831.
Both of them are detected by rtl_433 as a Smoke detector GS 558. I'm using both with that protocol, as they report an ID and a raw code, but is far from clean an optimal. I want to try to add support for both of them with your help. 

This is what I get from the WD51 water leak sensor when I trigger the alarm:

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.233632s
model     : Smoke detector GS 558                  id        : 1796
unit      : 1            learn     : 0             Raw Code  : a0e081
Analyzing pulses...
Total count:  668,  width: 985.99 ms		(246497 S)
Pulse width distribution:
 [ 0] count:  304,  width:  292 us [244;304]	(  73 S)
 [ 1] count:  135,  width:  852 us [680;1044]	( 213 S)
 [ 2] count:    1,  width:   72 us [72;72]	(  18 S)
 [ 3] count:   62,  width:  552 us [440;676]	( 138 S)
 [ 4] count:   39,  width: 1988 us [1600;2208]	( 497 S)
 [ 5] count:  124,  width: 1384 us [1192;1612]	( 346 S)
 [ 6] count:    1,  width:  220 us [220;220]	(  55 S)
 [ 7] count:    1,  width: 2560 us [2560;2560]	( 640 S)
 [ 8] count:    1,  width: 1072 us [1072;1072]	( 268 S)
Gap width distribution:
 [ 0] count:    6,  width:   48 us [48;56]	(  12 S)
 [ 1] count:   18,  width:  112 us [100;144]	(  28 S)
 [ 2] count:  336,  width:  868 us [700;1040]	( 217 S)
 [ 3] count:  159,  width:  276 us [232;340]	(  69 S)
 [ 4] count:   11,  width: 8668 us [8660;8680]	(2167 S)
 [ 5] count:   48,  width:  416 us [352;512]	( 104 S)
 [ 6] count:   32,  width:  184 us [152;220]	(  46 S)
 [ 7] count:   14,  width:   80 us [68;92]	(  20 S)
 [ 8] count:   40,  width:  592 us [516;692]	( 148 S)
 [ 9] count:    1,  width: 5628 us [5628;5628]	(1407 S)
 [10] count:    1,  width:   64 us [64;64]	(  16 S)
 [11] count:    1,  width: 2856 us [2856;2856]	( 714 S)
Pulse period distribution:
 [ 0] count:    9,  width:  380 us [332;416]	(  95 S)
 [ 1] count:  369,  width: 1136 us [912;1416]	( 284 S)
 [ 2] count:    1,  width:  120 us [120;120]	(  30 S)
 [ 3] count:   11,  width: 8964 us [8952;8980]	(2241 S)
 [ 4] count:  114,  width: 2296 us [1868;2600]	( 574 S)
 [ 5] count:   25,  width:  528 us [420;628]	( 132 S)
 [ 6] count:   80,  width: 1576 us [1424;1828]	( 394 S)
 [ 7] count:   52,  width:  748 us [616;900]	( 187 S)
 [ 8] count:    5,  width: 3152 us [3076;3272]	( 788 S)
 [ 9] count:    1,  width: 5908 us [5908;5908]	(1477 S)
Level estimates [high, low]:  15932,    885
RSSI: -0.1 dB SNR: 12.5 dB Noise: -12.7 dB
Frequency offsets [F1, F2]:   21573,      0	(+82.3 kHz, +0.0 kHz)
Guessing modulation: No clue...

*** signal_start = -8607, signal_end = 314906, signal_len = 323513, pulses_found = 858
Iteration 1. t: 247    min: 113 (694)    max: 382 (164)    delta 41562
Iteration 2. t: 245    min: 112 (690)    max: 379 (168)    delta 10
Iteration 3. t: 245    min: 112 (689)    max: 379 (169)    delta 0
Pulse coding: Short pulse length 112 - Long pulse length 379

Short distance: 147, long distance: 1059, packet distance: 2168

p_limit: 245
bitbuffer:: Number of rows: 18
[00] {25} 00 00 00 00                                                          : 00000000 00000000 00000000 0
[01] {25} 00 00 00 00                                                          : 00000000 00000000 00000000 0
[02] {25} 00 00 00 00                                                          : 00000000 00000000 00000000 0
[03] {25} 00 00 00 00                                                          : 00000000 00000000 00000000 0
[04] {25} 00 00 00 00                                                          : 00000000 00000000 00000000 0
[05] {25} 00 00 00 00                                                          : 00000000 00000000 00000000 0
[06] {43} 00 04 00 00 00 00                                                    : 00000000 00000100 00000000 00000000 00000000 000
[07] {180} 00 00 01 74 b6 84 e1 3d 23 58 00 00 0c 90 85 a9 15 46 7b 12 60 00 00
[08] {73} 05 56 57 64 77 13 60 93 00 00
[09] {25} 00 00 00 00                                                          : 00000000 00000000 00000000 0
[10] {25} 00 00 00 00                                                          : 00000000 00000000 00000000 0
[11] {25} 00 00 00 00                                                          : 00000000 00000000 00000000 0
[12] {25} 00 00 00 00                                                          : 00000000 00000000 00000000 0
[13] {25} 00 00 00 00                                                          : 00000000 00000000 00000000 0
[14] {25} 00 00 00 00                                                          : 00000000 00000000 00000000 0
[15] {25} 00 00 00 00                                                          : 00000000 00000000 00000000 0
[16] {175} 00 00 0f 50 4a 8b 3c 3e 56 48 00 00 0f 2c 22 12 fc 35 5f 93 00 00
[17] {62} 15 4a 05 24 95 85 ef 24
Iteration 1. t: 0    min: 0 (0)    max: 0 (0)    delta -727379968
Iteration 2. t: 0    min: 0 (0)    max: 0 (0)    delta 0
Distance coding: Pulse length 0

Short distance: 1000000, long distance: 0, packet distance: 0

p_limit: 0
bitbuffer:: Number of rows: 0

zuckschwerdt, 2020-06-15:
Can you record a sample, convert to OOK and post here? Also try yourself on https://triq.org/pdv/
`rtl_433 -S all` to record, `rtl_433 -w FILENAME.ook FILENAME.cu8` to convert.

Nauttor, 2020-06-15:
I try to catch it, but I'm not sure if is the one that appears before or after the "parsed" signal 

*** Saving signal to file g009_433.92M_250k.cu8
*** Writing data from 1126614, len 262144
signal_bsize = 131072  -      sg_index = 2621440
start_pos    = 2098090  -   buffer_size = 3145728
*** Saving signal to file g010_433.92M_250k.cu8
*** Writing data from 2098090, len 131072
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-06-15 18:01:25
model     : Smoke detector GS 558                  id        : 1796
unit      : 1            learn     : 0             Raw Code  : a0e081
signal_bsize = 786432  -      sg_index = 2883584
start_pos    = 1845164  -   buffer_size = 3145728
*** Saving signal to file g011_433.92M_250k.cu8
*** Writing data from 1845164, len 786432

Nauttor, 2020-06-15:
I upload the g011 to  https://triq.org/pdv/ but It appears to not detect a signal. 

[g011_433.92M_250k.ook.zip](https://github.com/merbanan/rtl_433/files/4781715/g011_433.92M_250k.ook.zip)

zuckschwerdt, 2020-06-15:
That's not the .ook but the .cu8. (An .ook is a readable text file). This file is just a short unreadable signal. Capture samples and drop them on https://triq.org/iqs/ too see if it looks like something usable. Also you can read the samples back in to let rtl_433 analyze them: `rtl_433 -A FILENAME.cu8` -- you want a file that shows lots of pulses like in your first post.

Nauttor, 2020-06-15:
OK, I'll try again, and how can I filter all the other "devices"/noise from the signal I'm trying to get?

zuckschwerdt, 2020-06-15:
Ideally you unplug the antenna and put the sensor at about 10cm to the receiver. That should then only capture that sensor.

Nauttor, 2020-06-16:
OK, I think now I got some of them right, at least I can see them in the triq.org and get some info with the suggested Flex decoder output. 

Test mode active. Reading samples from file: g001_433.92M_250k.ook
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.000000s
model     : Smoke-GS558  id        : 1796
unit      : 1            learn     : 0             Raw Code  : a0e081
Analyzing pulses...
Total count:  899,  width: 1294.60 ms		(323650 S)
Pulse width distribution:
 [ 0] count:  252,  width:  920 us [860;1016]	( 230 S)
 [ 1] count:  646,  width:  436 us [328;560]	( 109 S)
 [ 2] count:    1,  width:  336 us [336;336]	(  84 S)
Gap width distribution:
 [ 0] count:   89,  width:  168 us [120;208]	(  42 S)
 [ 1] count:  612,  width:  692 us [572;804]	( 173 S)
 [ 2] count:  162,  width:  240 us [200;280]	(  60 S)
 [ 3] count:   35,  width: 8540 us [8440;8620]	(2135 S)
Pulse period distribution:
 [ 0] count:  863,  width: 1132 us [1116;1156]	( 283 S)
 [ 1] count:   35,  width: 8964 us [8944;8980]	(2241 S)
Level estimates [high, low]:      0,      0
RSSI: 0.0 dB SNR: 0.0 dB Noise: 0.0 dB
Frequency offsets [F1, F2]:       0,      0	(+0.0 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with sync/delimiter
Attempting demodulation... short_width: 436, long_width: 920, reset_limit: 8624, sync_width: 336
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=436,l=920,r=8624,g=0,t=0,y=336'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 25
[00] { 2} 40        : 01
[01] { 4} e0        : 1110
[02] { 1} 80        : 1
[03] { 4} 80        : 1000
[04] { 2} c0        : 11
[05] { 7} 56        : 0101011
[06] { 4} d0        : 1101
[07] { 9} e3 80     : 11100011 1
[08] { 1} 00        : 0
[09] { 6} 5c        : 010111
[10] { 4} b0        : 1011
[11] { 1} 80        : 1
[12] {22} 1f 57 ec  : 00011111 01010111 111011
[13] { 5} c0        : 11000
[14] { 2} c0        : 11
[15] { 7} aa        : 1010101
[16] { 3} e0        : 111
[17] { 9} 7c 00     : 01111100 0
[18] { 2} c0        : 11
[19] { 2} 80        : 10
[20] { 7} 5e        : 0101111
[21] { 5} b8        : 10111
[22] { 4} 80        : 1000
[23] { 1} 80        : 1
[24] { 5} ff        : 11111
... Maximum number of rows reached. Message is likely truncated.

Test mode active. Reading samples from file: g002_433.92M_250k.ook
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.000000s
model     : Smoke-GS558  id        : 1796
unit      : 1            learn     : 0             Raw Code  : a0e081
Analyzing pulses...
Total count:  899,  width: 1294.39 ms		(323598 S)
Pulse width distribution:
 [ 0] count:  252,  width:  900 us [852;980]	( 225 S)
 [ 1] count:  434,  width:  420 us [340;508]	( 105 S)
 [ 2] count:  213,  width:  324 us [304;356]	(  81 S)
Gap width distribution:
 [ 0] count:  179,  width:  256 us [208;292]	(  64 S)
 [ 1] count:  612,  width:  740 us [620;828]	( 185 S)
[... 88 lines cut ...]
len       : 1            data      : 8,
len       : 4            data      : 1,
len       : 5            data      : d0,
len       : 9            data      : 7e8,
len       : 6            data      : e0,
len       : 7            data      : f4,
len       : 1            data      : 0,
len       : 20           data      : fbe3e,
len       : 35           data      : 5fbe3eafc,
len       : 7            data      : f0,
len       : 3            data      : e,
len       : 1            data      : 0,
len       : 18           data      : 5fbe0,
len       : 8            data      : ea,
len       : 5            data      : e8,
len       : 16           data      : e3ea,
len       : 1            data      : f
codes     : {17}7ef88, {12}eaf, {9}7c0, {5}f0, {1}0, {8}7e, {2}c, {23}8fabf6, {1}8, {4}1, {5}d0, {9}7e8, {6}e0, {7}f4, {1}0, {20}fbe3e, {35}5fbe3eafc, {7}f0, {3}e, {1}0, {18}5fbe0, {8}ea, {5}e8, {16}e3ea, {1}f
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-06-16 11:07:49
model     : Smoke-GS558  id        : 1796
unit      : 1            learn     : 0             Raw Code  : a0e081

[g001_433.92M_250k.zip](https://github.com/merbanan/rtl_433/files/4785607/g001_433.92M_250k.zip)
[g002_433.92M_250k.zip](https://github.com/merbanan/rtl_433/files/4785608/g002_433.92M_250k.zip)

zuckschwerdt, 2020-06-16:
The data files are both perfect. One problem: the sync width is too close to the short width and gets confused. Use `rtl_433 -R 0 -X 'n=WD51,m=OOK_PWM,s=440,l=940,g=900,r=9000' g001_433.92M_250k.cs16` i.e. without the sync. Then you get consisten readings of:
`{25}7ef8fa8` which is to be read as  `{24}7ef8fa` -- the last bit is actually the sync of the next packet. You can also run that flex line live to experiment with the switches and you should get only the codes without the noise (don't use `-A`).

Nauttor, 2020-06-16:
I try it on live mode, That's what I get,

 rtl_433 -d driver=hackrf -R 0 -X 'n=WD51,m=OOK_PWM,s=440,l=940,g=900,r=9000'
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/Users/nautor/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Registered 1 out of 153 device decoding protocols [ ]
[INFO] Opening HackRF One #0 87c867dc2919425f...
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-06-16 11:51:20
model     : WD51         count     : 25            num_rows  : 25            rows      :
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 25           data      : 7ef8fa8,
len       : 24           data      : 7ef8fa
codes     : {25}7ef8fa8, {25}7ef8fa8, {25}7ef8fa8, {25}7ef8fa8, {25}7ef8fa8, {25}7ef8fa8, {25}7ef8fa8, {25}7ef8fa8, {25}7ef8fa8, {25}7ef8fa8, {25}7ef8fa8, {25}7ef8fa8, {25}7ef8fa8, {25}7ef8fa8, {25}7ef8fa8, {25}7ef8fa8, {25}7ef8fa8, {25}7ef8fa8, {25}7ef8fa8, {25}7ef8fa8, {25}7ef8fa8, {25}7ef8fa8, {25}7ef8fa8, {25}7ef8fa8, {24}7ef8fa

Nauttor, 2020-06-16:
With that Flex line you indicate I always get a clean read of the same data.

zuckschwerdt, 2020-06-16:
Great! Are there configuration switches on/in the sensor? Usually part of that code should be user-settable to distinguish devices. Otherwise it could also be a "globally unique" code.
For a stable setup also try: `rtl_433 -R 0 -X 'n=WD51,m=OOK_PWM,s=440,l=940,g=900,r=9000,unique'` for a shorter output, or `rtl_433 -R 0 -X 'n=WD51,m=OOK_PWM,s=440,l=940,g=900,r=9000,match=7ef8fa,unique'` to just show your single sensor (this should be a good filter to trigger actual alarms on.
Create a documented .conf file (see the conf dir for examples) with all this and PR or post here, if you like.

Nauttor, 2020-06-16:
The sensor is quite simple, no jumpers, no whatsoever to be configured. It have an external dipole that if in contact with water it switch ON the internal jumper and trigger the alarm. It have too a button to turn on/off the device and to trigger the alarm manually as a "test" trigger. 

 I change the batteries and even put old batteries in it to see if it changes the behaviour, I see nothing in terms of the received signal always the same 7ef8fa8. 

From the data received can we obtain an ID of the device? If for example I have 2 devices how can I differentiate both?

zuckschwerdt, 2020-06-16:
Yeah, `7ef8fa` is the ID.

Nauttor, 2020-06-16:
I'm guessing the conf file but, can we use this to guess the ID to a more readable value like the other KERUI devices?

decoder {
    name=KERUI-WD51,
    modulation=OOK_PWM,
    short=440,
    long=940,
    reset=9000,
    gap=900,
    bits=25,
    get=@0:{20}:id,
    match=7ef8fa,
    unique
}

zuckschwerdt, 2020-06-16:
Yes, that will get the ID. You should use `{24}` for the full ID. But with the `match` it will only ever show ID=7ef8fa.

Nauttor, 2020-06-16:
Ok, then this could work right?

decoder {
name=KERUI-WD51,
modulation=OOK_PWM,
short=440,
long=940,
reset=9000,
gap=900,
bits=25,
get=@0:{24}:id,
match=7ef8fa,
unique
}

zuckschwerdt, 2020-06-16:
Should work perfectly. (note that 0x7ef8fa == 8321274)

Nauttor, 2020-06-16:
OK now for the KERUI Mini PIR, I got a similar situation. 

The device is quite simple, it have only a CR2450 and no jumper, switch, button or whatsoever. So all in all is plain basic a ON/OFF device. 

rtl_433 -A g023_433.92M_250k.cu8
Test mode active. Reading samples from file: g023_433.92M_250k.cu8
Detected OOK package	@0.286612s
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.286612s
model     : Smoke detector GS 558                  id        : 1689
unit      : 31           learn     : 0             Raw Code  : 50d33f
Analyzing pulses...
Total count:  299,  width: 424.27 ms		(106067 S)
Pulse width distribution:
 [ 0] count:  156,  width:  840 us [832;848]	( 210 S)
 [ 1] count:  143,  width:  276 us [268;288]	(  69 S)
Gap width distribution:
 [ 0] count:  156,  width:  292 us [284;304]	(  73 S)
 [ 1] count:  131,  width:  856 us [848;864]	( 214 S)
 [ 2] count:   11,  width: 8680 us [8672;8692]	(2170 S)
Pulse period distribution:
 [ 0] count:  287,  width: 1132 us [1116;1148]	( 283 S)
 [ 1] count:   11,  width: 8952 us [8944;8964]	(2238 S)
Level estimates [high, low]:  16310,  10741
RSSI: -0.0 dB SNR: 1.8 dB Noise: -1.8 dB
Frequency offsets [F1, F2]:   -5036,      0	(-19.2 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
Attempting demodulation... short_width: 276, long_width: 840, reset_limit: 8696, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=276,l=840,r=8696,g=868,t=224,y=0'
pulse_demod_pwm(): Analyzer Device
bitbuffer:: Number of rows: 12
[00] {25} 03 34 f5 80 : 00000011 00110100 11110101 1
[01] {25} 03 34 f5 80 : 00000011 00110100 11110101 1
[02] {25} 03 34 f5 80 : 00000011 00110100 11110101 1
[03] {25} 03 34 f5 80 : 00000011 00110100 11110101 1
[04] {25} 03 34 f5 80 : 00000011 00110100 11110101 1
[05] {25} 03 34 f5 80 : 00000011 00110100 11110101 1
[06] {25} 03 34 f5 80 : 00000011 00110100 11110101 1
[07] {25} 03 34 f5 80 : 00000011 00110100 11110101 1
[08] {25} 03 34 f5 80 : 00000011 00110100 11110101 1
[09] {25} 03 34 f5 80 : 00000011 00110100 11110101 1
[10] {25} 03 34 f5 80 : 00000011 00110100 11110101 1
[11] {24} 03 34 f5    : 00000011 00110100 11110101

rtl_433 -A g024_433.92M_250k.cu8
Test mode active. Reading samples from file: g024_433.92M_250k.cu8
Detected OOK package	@0.204768s
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.204768s
model     : Smoke detector GS 558                  id        : 1689
unit      : 31           learn     : 0             Raw Code  : 50d33f
Analyzing pulses...
Total count:  899,  width: 1292.56 ms		(323139 S)
Pulse width distribution:
 [ 0] count:  468,  width:  864 us [860;880]	( 216 S)
 [ 1] count:  431,  width:  304 us [296;316]	(  76 S)
Gap width distribution:
 [ 0] count:  468,  width:  264 us [256;276]	(  66 S)
 [ 1] count:  395,  width:  828 us [816;840]	( 207 S)
 [ 2] count:   35,  width: 8652 us [8648;8660]	(2163 S)
Pulse period distribution:
 [ 0] count:  863,  width: 1132 us [1128;1144]	( 283 S)
 [ 1] count:   35,  width: 8956 us [8952;8964]	(2239 S)
Level estimates [high, low]:  15959,   1666
RSSI: -0.1 dB SNR: 9.8 dB Noise: -9.9 dB
Frequency offsets [F1, F2]:   -5517,      0	(-21.0 kHz, +0.0 kHz)
Guessing modulation: Pulse Width Modulation with multiple packets
Attempting demodulation... short_width: 304, long_width: 864, reset_limit: 8664, sync_width: 0
Use a flex decoder with -X 'n=name,m=OOK_PWM,s=304,l=864,r=8664,g=844,t=224,y=0'
pulse_demod_pwm(): Analyzer Device
[... 54 lines cut ...]
len       : 25           data      : 0334f58       id        : 13135,
len       : 25           data      : 0334f58       id        : 13135,
len       : 25           data      : 0334f58       id        : 13135,
len       : 25           data      : 0334f58       id        : 13135,
len       : 24           data      : 0334f5        id        : 13135
codes     : {25}0334f58, {25}0334f58, {25}0334f58, {25}0334f58, {25}0334f58, {25}0334f58, {25}0334f58, {25}0334f58, {25}0334f58, {25}0334f58, {25}0334f58, {25}0334f58, {25}0334f58, {25}0334f58, {25}0334f58, {25}0334f58, {25}0334f58, {25}0334f58, {25}0334f58, {25}0334f58, {25}0334f58, {25}0334f58, {25}0334f58, {25}0334f58, {24}0334f5

I change the battery, and the same "0334f5" / ID:13135 persist.

decoder {
name=Kerui-MiniPir-P831,
modulation=OOK_PWM,
short=304,
long=864,
reset=8664,
gap=844,
bits=25,
get=@0:{24}:id,
match=0334f5,
unique
}

Maybe the s,l,r,g need to be adjusted. 

It appears correct to you?

Nauttor, 2020-06-18:
HI Christian @zuckschwerdt,

Doing some test and looking at the Manual of the kerui alarm that works against the WD51, I read that the sensor make a 3 beeps and send a signal to the central when it's on low battery. 

I redo my tests with low power batteries and only once I get to see this in the console. However I'm not able to reproduce it again. 

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-06-18 13:53:55
model     : Kerui-Security                         ID (20bit): 0x81070
Command (4bit): 0xf      State     : battery
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-06-18 13:53:55
model     : Kerui-Security                         ID (20bit): 0x81070
Command (4bit): 0xf      State     : battery
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-06-18 13:53:55
model     : Kerui-Security                         ID (20bit): 0x81070
Command (4bit): 0xf      State     : battery
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-06-18 13:53:55
model     : Smoke-GS558  id        : 1796
unit      : 1            learn     : 0             Raw Code  : f0e081
*** Saving signal to file g001
_433.92M_250k.cs16 (233794 samples, 1048576 bytes)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

[g001_433.92M_250k.cs16.zip](https://github.com/merbanan/rtl_433/files/4798763/g001_433.92M_250k.cs16.zip)

So my question is, If it is using the same "schema" to communicate the battery state, maybe there is a new cmd state not registered in the kerui-security protocol 68 that is used for the water leak sensor. 

I don't know if you can see it from the above file.

zuckschwerdt, 2020-06-18:
Good find. You are right, if this is like the other Kerui then use:
```
 invert,
 get=@0:{20}:id,
 get=@20:{4}:cmd,
```
Compare here to guess what the `cmd` might mean: https://github.com/merbanan/rtl_433/blob/master/src/devices/kerui.c#L33

zuckschwerdt, 2020-06-18:
Thanks for the detailed analysis! I guess using `sync` will be a problem because the older protocol has longer syncs where the newer protocol has short sync. Best to just parse them as (25th) data bit and ignore.
So, comment the sync_width, set short_width to 400, and tolerance to 160 should work.
Could be worth it to also enable gap_limit=1100 and reset_limit 9900. That would capture multiple packets as a single transmission. Then we could add a check to require multiple matching packets to improve reliability:
```
int r = bitbuffer_find_repeated_row(bitbuffer, 9, 25); // expected are 25 packets, require 9
if (r < 0)
    return DECODE_ABORT_LENGTH;
```

Nauttor, 2020-06-18:
OK, I do all the modifications and I get a bit strange result. 

If I do:
rtl_433 -d 1 -R 0 -X 'n=WD51,m=OOK_PWM,s=400,l=940,g=1100,t=160,r=9900,bits=25,invert,get=@0:{20}:id,get=@20:{4}:cmd'

I always get a response from the devices, olds and news, both get registered, but when I use the R 68 with the modifications, the new devices always works, the old one is a hit and miss. 

You can see above the log when doing it manually. 139688 is the old device.

_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-06-18 22:03:44
model     : WD51         count     : 24            num_rows  : 25            rows      :
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 25           data      : 8107050       id        : 528496
cmd       : 5,
len       : 24           data      : 810705        id        : 528496
cmd       : 5
codes     : {25}8107050, {25}8107050, {25}8107050, {25}8107050, {25}8107050, {25}8107050, {25}8107050, {25}8107050, {25}8107050, {25}8107050, {25}8107050, {25}8107050, {25}8107050, {25}8107050, {25}8107050, {25}8107050, {25}8107050, {25}8107050, {25}8107050, {25}8107050, {25}8107050, {25}8107050, {25}8107050, {25}8107050, {24}810705
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-06-18 22:03:48
model     : WD51         count     : 24            num_rows  : 25            rows      :
len       : 25           data      : fccb0a0       id        : 1035440
cmd       : 10,
len       : 25           data      : fccb0a0       id        : 1035440
cmd       : 10,
[... 91 lines cut ...]
cmd       : 10,
len       : 25           data      : 221a8a0       id        : 139688
cmd       : 10,
len       : 25           data      : 221a8a0       id        : 139688
cmd       : 10,
len       : 25           data      : 221a8a0       id        : 139688
cmd       : 10,
len       : 6            data      : 21            id        : 138839
cmd       : 5
codes     : {21}21a880, {25}221a8a0, {25}221a8a0, {25}221a8a0, {25}221a8a0, {25}221a8a0, {25}221a8a0, {25}221a8a0, {25}221a8a0, {25}221a8a0, {25}221a8a0, {25}221a8a0, {25}221a8a0, {25}221a8a0, {25}221a8a0, {25}221a8a0, {25}221a8a0, {25}221a8a0, {25}221a8a0, {25}221a8a0, {25}221a8a0, {25}221a8a0, {25}221a8a0, {25}221a8a0, {6}21

Here is the modification I do to the code:
r_device kerui = {
    .name          = "Kerui PIR / Contact Sensor / Water leak",
    .modulation    = OOK_PULSE_PWM,
    .short_width   = 400,
    .long_width    = 940,
    .reset_limit   = 9900,
    .gap_limit     = 1100,
    //.sync_width    = 480,
    .tolerance     = 160,
    .decode_fn     = &kerui_callback,
    .disabled      = 0,
    .fields         = output_fields,
};

zuckschwerdt, 2020-06-18:
Ah, sorry. The first row might be short we need to use the actual found row. I've commited this patch now. Please test, should support the WD51 now!

closed 2020-06-18 by zuckschwerdt

Nauttor, 2020-06-18:
I`m gonna try your code, but I already added the 

int r = bitbuffer_find_repeated_row(bitbuffer, 9, 25); // expected are 25 packets, require 9

In my test as you suggest before, that why I Said is strange to obtain a different result.

zuckschwerdt, 2020-06-18:
You need to change the next thwo `[0]` to `[r]`, see https://github.com/merbanan/rtl_433/commit/1e49e5058fe462140fc9ec5bf52c41f37f50fdf1#diff-d6ef22fcef07f9eda059f309fe3258f2L22 :)

Nauttor, 2020-06-18:
My bad, genius, now it works. Good job man, thanks for your help.  

Just one thing, you may consider to add a comment in the readme about the R68 protocol, or at least in the code on kerui.c, that it now supports Kerui old protocol and new one, with at least Water leak sensor WD51 and Mini Pir P831.

commit 1e49e505 (2020-06-18): Add support for Kerui WD51 Water leak sensor (closes #1406)