merbanan/rtl_433 #1414: Quitting rtl_433 with a RTL-SDR device on Windows causes async read failed (-5) when it should return 0

pclov3r, 2020-06-23:
So, when quitting rtl_433 on Windows i get this error before it quits.

```
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
Signal caught, exiting!

LIBUSB_ERROR_NOT_FOUND: Entity not found!
Check your RTL-SDR dongle, USB cables, and power supply.

WARNING: async read failed (-5).
```

For reference of the issue from a another project see https://github.com/theori-io/nrsc5/issues/202#issuecomment-472305370

zuckschwerdt, 2020-06-23:
Thanks. Ok, we should ignore the error code if we already issued a `cancel`, because of: "a bug in librtlsdr or libusb. The problem is that when we cancel the async transfer (using rtlsdr_cancel_async) the return value of rtlsdr_read_async will be -5 on Windows, indicating an error, instead of 0"

gvanem, 2021-03-10:
I get this -5 error a lot when and hopping in frequency every 20 sec. And especially when using the web-interface; `-F http` option.
And sometimes compiled with a crash inside libusb since some `CRITICAL_SECTION` is `NULL`!

gvanem, 2022-01-16:
I switched to using [this](https://github.com/old-dab/rtlsdr) version of RTLSDR some time ago. After *old-DAB* switched to
using WinUsb in his Windows version, it's been running rock solid.

None of these async-errors that was typical when libusb was used. @pclov3r it's worth a try.

schmiernippel, 2023-05-18:
> I switched to using [this](https://github.com/old-dab/rtlsdr) version of RTLSDR some time ago. After _old-DAB_ switched to using WinUsb in his Windows version, it's been running rock solid.
> 
> None of these async-errors that was typical when libusb was used. @pclov3r it's worth a try.

How can i use this? Nothing happens when i open the program in CMD ??

zuckschwerdt, 2023-05-18:
All RTL-SDR drivers on Windows need to the usual Zadig install, right?

gdt, 2023-09-29:
Is this issue still valid?  What's the path forward?

@pclov3r Can you test with up-to-date git master?    If that is problematic, can you identify a minimal change from the other repo and prepare a PR?

gdt, 2024-08-04:
This has been in feedback for over 10 months, and nobody has cared enough to retest.  I'm going to just close it, and if there is still a problem someone can open a new, high-quality bug report.

closed 2024-08-04 by gdt