merbanan/rtl_433 #1416: Support for Bresser 5-in-1 7002581

crapulax-sudo, 2020-06-26:
Hi,
I purchased this weather station : Bresser 7002581
Wifi colour Weather Station with 5 in 1 Multisensor

[https://www.bresser.de/en/Weather-Time/Weather-Center/BRESSER-WIFI-color-weather-center-with-5in1-profi-sensor-white.html]

I am running on a Raspi.

The following commands does not produce any output :
rtl_433 -f 868428000  
rtl_433 -f 868428000  -R 119

So I tried this one
`rtl_433 -f 868428000 -A
rtl_433 version 20.02-70-g3018ae9 branch master at 202006242214 inputs file rtl_tcp RTL-SDR
<..>
New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 126 out of 154 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-154 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.428MHz.
Detected FSK package    2020-06-26 07:54:41
Analyzing pulses...
Total count:   81,  width: 24.73 ms             (24735 S)
Pulse width distribution:
 [ 0] count:    1,  width:    0 us [0;0]        (   0 S)
 [ 1] count:    6,  width:    4 us [4;4]        (   4 S)
 [ 2] count:    8,  width:    2 us [2;2]        (   2 S)
 [ 3] count:    8,  width:    3 us [3;3]        (   3 S)
 [ 4] count:    7,  width:    5 us [5;6]        (   5 S)
 [ 5] count:    1,  width:    7 us [7;7]        (   7 S)
 [ 6] count:    4,  width:    1 us [1;1]        (   1 S)
 [ 7] count:   28,  width:  121 us [119;123]    ( 121 S)
 [ 8] count:    7,  width:  243 us [241;244]    ( 243 S)
 [ 9] count:    3,  width:  365 us [363;367]    ( 365 S)
 [10] count:    2,  width:  671 us [610;732]    ( 671 S)
 [11] count:    2,  width:  912 us [851;973]    ( 912 S)
 [12] count:    3,  width:  488 us [488;489]    ( 488 S)
 [13] count:    1,  width: 1461 us [1461;1461]  (1461 S)
Gap width distribution:
 [ 0] count:   13,  width:    3 us [3;3]        (   3 S)
 [ 1] count:    6,  width:    2 us [2;2]        (   2 S)
 [ 2] count:    6,  width:    5 us [5;6]        (   5 S)
 [ 3] count:    5,  width:    1 us [1;1]        (   1 S)
 [ 4] count:    4,  width:    4 us [4;4]        (   4 S)
 [ 5] count:    1,  width:  161 us [161;161]    ( 161 S)
 [ 6] count:   27,  width:  121 us [120;124]    ( 121 S)
 [ 7] count:    8,  width:  365 us [365;367]    ( 365 S)
 [ 8] count:    3,  width:  486 us [485;488]    ( 486 S)
 [ 9] count:    2,  width: 1219 us [1097;1341]  (1219 S)
 [10] count:    3,  width:  243 us [243;244]    ( 243 S)
 [11] count:    2,  width:  609 us [608;610]    ( 609 S)
Pulse period distribution:
 [ 0] count:    1,  width:    3 us [3;3]        (   3 S)
 [ 1] count:   13,  width:    5 us [5;7]        (   5 S)
 [ 2] count:    5,  width:    4 us [4;4]        (   4 S)
 [ 3] count:   12,  width:    7 us [7;9]        (   7 S)
 [ 4] count:    2,  width:   10 us [10;11]      (  10 S)
 [ 5] count:    1,  width:    2 us [2;2]        (   2 S)
 [ 6] count:    1,  width:  166 us [166;166]    ( 166 S)
 [ 7] count:   22,  width:  243 us [241;246]    ( 243 S)
 [ 8] count:    6,  width:  486 us [484;489]    ( 486 S)
 [ 9] count:    3,  width:  365 us [365;366]    ( 365 S)
 [10] count:    8,  width:  746 us [610;853]    ( 746 S)
 [11] count:    4,  width: 1248 us [1095;1340]  (1248 S)
 [12] count:    2,  width: 1765 us [1582;1949]  (1765 S)
Level estimates [high, low]:  15960,     10
RSSI: -0.1 dB SNR: 32.0 dB Noise: -32.1 dB
Frequency offsets [F1, F2]:  -15312,  -5550     (-233.6 kHz, -84.7 kHz)
Guessing modulation: Pulse Code Modulation (Not Return to Zero)
Attempting demodulation... short_width: 1, long_width: 1, reset_limit: 1024, sync_width: 0
Use a flex decoder with -X 'n=name,m=FSK_PCM,s=1,l=1,r=1024'`

and this one
`rtl_433 -f 868428000 -X n=bresser-test,m=FSK_PCM,s=124,l=124,r=25000
rtl_433 version 20.02-70-g3018ae9 branch master at 202006242214 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 127 out of 154 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-154 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.428MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-06-26 08:05:18
model     : bresser-test count     : 1             num_rows  : 1             rows      :
len       : 365          data      : 555555555516ea383d0c6003600c7ee67e92447fff6dff80ff800000000000000000000000000000000000000000
codes     : {365}555555555516ea383d0c6003600c7ee67e92447fff6dff80ff800000000000000000000000000000000000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-06-26 08:05:18
model     : Efergy-e2CT  Transmitter ID: 0
Battery   : 0            Current   : 0.00 A        Interval  : 6s            Learning  : NO
Integrity : CHECKSUM`

I can provide other traces.

zuckschwerdt, 2020-06-26:
The "codes" data looks roughly ok. Perhaps there is too much noise. Try to grab a sample `rtl_433 -S all`, playback with e.g. `rtl_433 -A YOURFILE.cu8`. Also Drop it on https://triq.org/iqs/ to see if the band is clear and the signal well in-band. Try `rtl_433 -w YOURFILE.ook YOURFILE.cu8` and drop onto https://triq.org/pdv/ to see if the pulse detection works. (there are some stray <10µs pulse in your analyze above). Have fun, and if in doubt upload zipped samples here.

peteakalad, 2020-06-26:
Also try running

`rtl_433 -f 868300000 -R 119 -Y classic -s 1M`

This works for my 5in1 7002510 model, which looks like an identical station but different receiver.

crapulax-sudo, 2020-06-27:
> 
> 
> Also try running
> 
> `rtl_433 -f 868300000 -R 119 -Y classic -s 1M`
> 
> This works for my 5in1 7002510 model, which looks like an identical station but different receiver.

I tried without success.

crapulax-sudo, 2020-06-27:
Here are the capture files created with the following commands :
rtl_433 -f 868300000 -S all
rtl_433 -w g001_868.3M_1000k.ook g001_868.3M_1000k.cu8

[bresser-7002581-capture-868300000.zip](https://github.com/merbanan/rtl_433/files/4840328/bresser-7002581-capture-868300000.zip)

crapulax-sudo, 2020-07-01:
Making more tests this morning

Using the following command

`./rtl_433 -f 868300000 -s 1M -X 'n=Bresser-5in1-7002581,m=FSK_PCM,s=120,l=120,r=2500'`

I get
```
{221}555555555516ea38478c6003600c7fffff92440ba84afff85a800000
{221}555555555516ea38478c6003600c7fffff92440ba84afff85a800000
{221}555555555516ea38478c6003600c7fffff92440ba84afff85a800000
```
Using bitbench and applying a 3 shift I get 

```
02dd4708f18c006c018ffffff2488175095fff0b500000
02dd4708f18c006c018ffffff2488175095fff0b500000
02dd4708f18c006c018ffffff2488175095fff0b500000
```
where we can see the sync word 2dd4

The current external T° was 17.5 with a 95 % humidity that we can spot 

`
02dd4708f18c006c018ffffff2488_**1750_95**_fff0b500000`

 
Something curious is that I get different packets 
```
{218}555555555516ea7ea30c6003600c7feeff912c7fff3fff80bc00000
{221}555555555516ea4f360c6003600c7fffff912c0b9849fff804800000
{220}555555555516ea4e6e8c6003600c7fffff912c7fff3fff80ab00000
{221}555555555516ea2c658c6003600c7fddff92440b9849fff80d800000
{221}555555555516ea419e0c6003600c7fc47f92447fff3fff80cd800000
{221}555555555516ea4dfe8c6003600c7fffff92440b9849fff86b800000
```

then bitbench 
```
02dd4fd4618c006c018ffddff2258fffe7fff017800000
02dd49e6c18c006c018ffffff2258173093fff00900000
02dd49cdd18c006c018ffffff2258fffe7fff015600000
02dd458cb18c006c018ffbbff2488173093fff01b00000
02dd4833c18c006c018ff88ff2488fffe7fff019b00000
02dd49bfd18c006c018ffffff2488173093fff0d700000
```
Each time I receive packet, the weather base station is updated but the syntax seems different 1 over 2 packets. (we can see clearly the 17.3 T° each 2 packets), any guess ?

crapulax-sudo, 2020-07-01:
Capture with base station displayed data

02dd49e6c18c006c018ffffff2258173093fff00900000    : wind average 0 km/h wind direction SW
02dd458cb18c006c018ffbbff2488173093fff01b00000 : wind average 1.4 km/h wind direction WSW
02dd49bfd18c006c018ffffff2488173093fff0d700000   : wind average 0 km/h wind direction WSW

peteakalad, 2020-07-18:
Had a quick look not sure but might have got the wind direction in degrees?   My previous model has it encoded as a single byte 0..F = N..NNE..E..S..W..NNW, but looking at these examples there is no 0xA (SW) or 0xB (WSW) between the examples so thinking it might be degrees instead now?
 
```
9e 6c 18 c0 06 c0 18 ff ff ff 22 58 17 30 93 ff f0 09 00 00 0 : wind average 0 km/h wind direction SW 
9b fd 18 c0 06 c0 18 ff ff ff 24 88 17 30 93 ff f0 d7 00 00 0 : wind average 0 km/h wind direction WSW 
58 cb 18 c0 06 c0 18 ff bb ff 24 88 17 30 93 ff f0 1b 00 00 0 : wind average 1.4 km/h wind direction WSW   ~0.4 m/s
                              dd d  TT T  HH 

ddd = Direction in degrees (SW = ~225°, WSW = ~248°)
TTT = Temperature in celcuis in 0.1 units
HH = Humidity in %
```
Did not mention but I'm assuming these were for temp of 17.3 and humidity of 93%..

There should be both wind average and gust which is probably encoded in m/s rather than km/h?  If you could get some more data examples with the direction and wind speeds in m/s the receiver shows will help to identify how they are encoded.

peteakalad, 2020-07-18:
Actually looks like similar/same data structure that is discussed here:

https://github.com/merbanan/rtl_433/issues/1172

Maybe try with a 5in1 device using the code mentioned by s118 towards the end..

crapulax-sudo, 2020-07-29:
Thanks for the look ;-)

As suggested, I made more tests this morning using the 6in1 branch, modified the "bresser_5in1.c" file with the one #s1172 , that was not working, no decoding, so I modified it by lowering the value of this test "bitbuffer->bits_per_row[0] < 200" , then the decoding was performed.

`./rtl_433 -f 868300000 -s 1M -X 'n=Bresser Weather Center 5-in-1,m=FSK_PCM,s=120,l=120,r=2500'
rtl_433 version 20.02-107-gabe2b10 branch master at 202007281722 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 134 out of 161 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.300MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-29 16:31:31
model     : Bresser Weather Center 5-in-1          count     : 1             num_rows  : 1
rows      :
len       : 221          data      : 555555555516ea2c0a8c6003600c7ff77f912c1488197ff844800000
codes     : {221}555555555516ea2c0a8c6003600c7ff77f912c1488197ff844800000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-29 16:31:31
model     : Bresser      id        : 192
Temperature: 29.1 C      Humidity  : 32            Wind Gust : 0.1 m/s       Wind Speed: 0.1 m/s
Direction : 225.0 °     Integrity : CHECKSUM
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-29 16:31:43
model     : Bresser Weather Center 5-in-1          count     : 1             num_rows  : 1
rows      :
len       : 221          data      : 555555555516ea099e0c6003600c7fffff912c7ffeebff80ff800000
codes     : {221}555555555516ea099e0c6003600c7fffff912c7ffeebff80ff800000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-29 16:31:43
model     : Bresser      id        : 192
Wind Gust : 0.0 m/s      Wind Speed: 0.0 m/s       Direction : 225.0 °      Rain      : 22.8 mm
Integrity : CHECKSUM`

T°, Humidity, Wind speed, Wind gust, Wind direction are identical to the values displayed on the base station.
For the rain, the total value on the base station is 10.4 mm.

The rain

peteakalad, 2020-07-29:
Nice one!  Maybe that bits per row difference could enable the decoder to differentiate the different station versions/packet types. 

Note the rain value is cumulative since it was reset, the station doesn’t keep track of time to reset so it’s done on the receiver by working out the delta/difference of that value over time.

zuckschwerdt, 2020-09-28:
Actually the newer "5-in-1" is a "6-in-1", likely with less options (there is no UV sensor, right?)
#1225 should support this now. We still need to figure out how the different fields are flagged as valid/invalid.

closed 2020-09-28 by zuckschwerdt