merbanan/rtl_433 #1417: rtl_433 stucked after long time running

fengjunCN, 2020-06-27:
first, i express my high respect to this repository.

my isuue is:
when i set up rtl_433 in my raspberry pi4b on Rasbian Lite and ran rtl_433 with arguments:rtl_433 -F json | mosquitto_pub -h localhost -p 1883  -t home/rtl_433 -l     at first, mqtt msg can be sent to mqtt broker normally as signal received by SDR USB(820T2 & SDR), but as it running for about 1day later, the mqtt broker can not receive the msg, i have to restart it, then it'll be back to normal.
now i'v no idea for this issue, anyone can help me to solve this?

Hardwares:
raspberry pi 4b
SDR:820T2

OS:
Rasbian Lite

power for raspberry:
5v 3A

thanks in advance!

peteakalad, 2020-06-27:
Tried running the built in mqtt client with no pipes to mosquitto_pub needed?

rtl_433 -F mqtt://localhost:1883

fengjunCN, 2020-06-29:
> Tried running the built in mqtt client with no pipes to mosquitto_pub needed?
> 
> `rtl_433 -F mqtt://localhost:1883`

@peteakalad thank u very much, i'll try it...

fengjunCN, 2020-07-03:
It works well now.

closed 2020-07-03 by fengjunCN