merbanan/rtl_433 #1419: Radiohead ASK Error

hisnameishb, 2020-06-29:
rlt_433 missinterprets the packet length:
The real packet length ist 8 bytes with this content (dezimal code):
50, 48, 46,54,56, 56, 0, 0

rtl_433 shows this:
model: RadioHead-ASK         Data len: 4    To: 50   From: 48   Id: 46
Flags: 54    Payload: 56, 56, 0, 0        Integrity : CRC

It missinterpretes the first 4 bytes as the fields "To", "From", "ID", "Flags"
Radiohead ASK has not such fields. It has iirc a meaningless preamble, a magicbyte, the payloadlength, the payload and two crc16 bytes

So it should show:
model: RadioHead-ASK     Data len: 8    To: -   From: -   Id: -
Payload: 50, 48, 46, 54, 56, 56, 0, 0    Integrity : CRC

zuckschwerdt, 2020-06-30:
The decoder is about "RadioHead Datagrams". Maybe we can add a raw RadioHead option, but I'd need links to the documention about that mode.

gdt, 2023-09-29:
@hisnameishb Are you still working on this?  Would you want to propose a specific change?

closed 2023-11-30 by zuckschwerdt