merbanan/rtl_433 #1423: pocsag decoding

evilpete, 2020-07-04:
@merbanan 

Any interest in POCSAG decoding?    If so I'll take a crack at it, there is lots of example code out there and the FSK signal bit rate is low enough that it will be easy to receive/demod

merbanan, 2020-07-04:
[pocsac_32.zip](https://github.com/merbanan/rtl_433/files/4872791/pocsac_32.zip)

merbanan, 2020-07-04:
I have been planning to add that but I did not have a strong enough signal for the current signal detector to trigger. So I came to the conclusion that I needed to test with a fft based segmenter but I never had the time to fix that. So start with and see if you are able to record the signal. My main focus currently  is to clean out all the PRs and issues.

evilpete, 2020-07-04:
I just did a quick automated scan, and you are right, there is not much transmission of pocsag out there anymore.

I'll make do a manual search sometime in the next few days..

psa-jforestier, 2020-08-11:
If you want to test pocsag signal localy, be aware that the Rpitx program (https://github.com/F5OEO/rpitx) can generate pocsag signal with a raspberry pi.
I used to make it works, by running rpitx pocsag and the RTL_SDR dongle on the very same Rpi, without any antenna connected to the PWM pin of the Rpi (the dongle is at 2cm from the board).
rtl_433 detect only garbage when transmitting.
[image]

evilpete, 2020-08-11:
I was gonna do this as an exercise (yeah I'm bored) but without a local "live" reference I'm not gonna try.

evilpete, 2020-08-12:
Closing due to no local "live" reference / lack of. use

closed 2020-08-12 by evilpete