merbanan/rtl_433 #1424: Support for Ambient Weather TX-3107 Pool Temperature Sensor

snicker, 2020-07-06:
I'm a bit inexperienced when it comes to reverse engineering RF, so I apologize if this request is malformed.

I recently acquired this:
https://www.ambientweather.com/amtx3107.html

I know it transmits at 915mhz, but I have no idea where to begin with decoding the signals. I have saved several cu8 files, but using the `-A` flag to replay them attempts to guess the modulation with mixed results (no clue, PCM). I've put the device right next to my RTLSDR radio antenna and filtered signals using the `-Y level=-1.0` option, and the pulses come at regular intervals, so I'm fairly certain that everything is coming from the TX-3107.

However, I don't know what to do next to attempt to decode the information into a useful set of data. Is there anything more I can do to help other than just uploading the cu8 files?

zuckschwerdt, 2020-07-07:
You can drop the cu8 files on https://triq.org/iqs/ to verify a good signal.
Then demodulate the I/Q data to OOK: `rtl_433 -w FILENAME.ook FILENAME.cu8` and drop onto https://triq.org/pdv/ to inspect the pulses.

snicker, 2020-07-19:
Alright, I'm quite out of my element here. I was able to verify consistent looking signal but I have no idea how to use the I/Q data once converted to OOK. Is there any kind of ... how to decode spectrograms for dummies resource anywhere?

snicker, 2020-07-19:
[image]
I've managed to get this far in terms of making this easy to look at, but I don't know what to do from here to determine exact pulse length, or what that even means (I'm assuming this converts to binary and then somehow I'd need to turn that into data)

snicker, 2020-07-19:
found this resource too https://www.osengr.org/WxShield/Downloads/Weather-Sensor-RF-Protocols.pdf

merbanan, 2020-07-19:
Can you share the ook file?

snicker, 2020-07-19:
[amtx3107_cu8+ook.zip](https://github.com/merbanan/rtl_433/files/4944683/amtx3107_cu8%2Book.zip)
Here are 3 samples, cu8 and ook

merbanan, 2020-07-19:
Well pulse coded modulation with a length of 480us looks suitable.

merbanan, 2020-07-19:
rtl_433 -r signals/amtx3107/g022_915M_1000k.cu8 -X "n=AMTX3107,m=FSK_PCM,s=484,l=484,r=10000,g=10000"

gives this output:

```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.221924s
model     : AMTX3107     count     : 1             num_rows  : 1             rows      : 
len       : 141          data      : 00555555555516ea04ea0c380e4d9f800000
codes     : {141}00555555555516ea04ea0c380e4d9f800000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.221924s
model     : AMTX3107     count     : 1             num_rows  : 1             rows      : 
len       : 29           data      : c5800000
codes     : {29}c5800000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.221924s
model     : AMTX3107     count     : 1             num_rows  : 1             rows      : 
len       : 24           data      : f00000
codes     : {24}f00000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.221924s
model     : AMTX3107     count     : 1             num_rows  : 1             rows      : 
len       : 23           data      : e00000
codes     : {23}e00000

```
Now we just need to sync on the data after the preamble.

snicker, 2020-07-19:
Awesome, I'll pull more samples and correlate them to channel and temperature and battery level.. I'm not sure what other data it reports but it looks like other Ambient Weather devices have 2 byte rolling code as well. 

Thanks for the expertise in looking at that. Just curious, what sort of mental gymnastics did you do to determine it was PCM with a 480us pulse length?

merbanan, 2020-07-19:
FSK is often PCM, then I zoomed in on the pulses at the start of the signal. The preamble starts with alternating levels.

merbanan, 2020-07-19:
rtl_433 -r g022_915M_1000k.cu8 -X "n=AMTX3107,m=FSK_PCM,s=484,l=484,r=10000,g=10000,preamble={16}0x2dd4"

```
Test mode active. Reading samples from file: signals/amtx3107/g022_915M_1000k.cu8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : @0.221924s
model     : AMTX3107     count     : 1             num_rows  : 1             rows      : 
len       : 76           data      : 09d418701c9b3f00000
codes     : {76}09d418701c9b3f00000

```

merbanan, 2020-07-19:
This should be enough to start looking into the meaning of the bits.

zuckschwerdt, 2020-07-20:
@merbanan the files are at 1000k, since we are still missing the auto-detection the examples above need `-s 1000k` or a divide by 4 in the results ;)

@snicker the ook files should be small text files (human readable). Someting went wrong with your conversion. Go to the folder with the .cu8 file (or don't name the folder itself .cu8 or .ook) then
`rtl_433 -s 1000k -w g022_915M_1000k.ook g022_915M_1000k.cu8` and check the .ook file, it should 55 lines of text.
Drop in on https://triq.org/pdv/ to see what Benjamin is talking about:
[image]

snicker, 2020-07-20:
So I've been collecting some data points and this is what I have found so far:

* the first 4 nibbles are always consistent for a channel/temperature pair, but I can't figure out how they correlate
* the next 9 nibbles always are always the following: 18701c9b3
* I thought the next nibble was the channel as there was a correlation, but there were two sample sets where this wasn't true, so I'm not super sure
* the last 4 are always 0000

I'm sharing the sheet below and I'll continue updating data

https://drive.google.com/file/d/1Jag_8SV3tW3lGHZBwKWvLXOo0RHIs6mQ/view?usp=sharing

snicker, 2020-07-20:
another bit of useful information that I hadn't confirmed previously: I was pulling the battery each time I changed the channel. I just pulled the battery twice and kept the channel and temperature constant and was getting the same results for the first four nibbles. I also noticed that the nibble at position 15 went back to reflecting the channel properly, see run21 and run22 in the sheet above

zuckschwerdt, 2020-07-20:
Interesting and good table, here is [a BitBench with that data](https://triq.net/bitbench?c=%7B75%7D851718701c9b3200000%09%5B2%20807%5D&c=%7B75%7D851718701c9b3200000%09%5B2%20807%5D&c=%7B75%7D851718701c9b3200000%09%5B2%20807%5D&c=%7B75%7D851718701c9b3200000%09%5B2%20807%5D&c=%7B75%7D851718701c9b3200000%09%5B2%20807%5D&c=%7B75%7D851718701c9b3200000%09%5B2%20807%5D&c=%7B75%7D966f18701c9b3a00000%09%5B2%20793%5D&c=%7B75%7D966f18701c9b3a00000%09%5B2%20793%5D&c=%7B75%7D966f18701c9b3a00000%09%5B2%20793%5D&c=%7B75%7D966f18701c9b3a00000%09%5B2%20793%5D&c=%7B75%7D966f18701c9b3a00000%09%5B2%20793%5D&c=%7B75%7D966f18701c9b3a00000%09%5B2%20793%5D&c=%7B75%7D8dc818701c9b3a00000%09%5B2%20791%5D&c=%7B75%7D8dc818701c9b3a00000%09%5B2%20791%5D&c=%7B75%7D8dc818701c9b3a00000%09%5B2%20791%5D&c=%7B75%7D8dc818701c9b3a00000%09%5B2%20791%5D&c=%7B75%7D8dc818701c9b3a00000%09%5B2%20791%5D&c=%7B75%7Dcfbd18701c9b3a00000%09%5B2%20797%5D&c=%7B75%7De35418701c9b3a00000%09%5B2%20798%5D&c=%7B75%7D67be18701c9b3a00000%09%5B2%20802%5D&c=%7B75%7D7c1918701c9b3a00000%09%5B2%20804%5D&c=%7B76%7D06de18701c9b3300000%09%5B3%20804%5D&c=%7B76%7D6a1018701c9b3300000%09%5B3%20806%5D&c=%7B74%7D43ad18701c9b3400000%09%5B4%20806%5D&c=%7B74%7D43ad18701c9b3400000%09%5B4%20806%5D&c=%7B74%7D43ad18701c9b3400000%09%5B4%20806%5D&c=%7B74%7D43ad18701c9b3400000%09%5B4%20806%5D&c=%7B75%7De31e18701c9b3600000%09%5B6%20806%5D&c=%7B75%7De31e18701c9b3600000%09%5B6%20806%5D&c=%7B75%7De31e18701c9b3600000%09%5B6%20806%5D&c=%7B76%7Dcaa318701c9b3100000%09%5B1%20806%5D&c=%7B76%7Dcaa318701c9b3100000%09%5B1%20806%5D&c=%7B76%7Dcaa318701c9b3100000%09%5B1%20806%5D&c=%7B76%7Dcaa318701c9b3100000%09%5B1%20806%5D&c=%7B76%7Dcaa318701c9b3100000%09%5B1%20806%5D&c=%7B76%7Dd10418701c9b3100000%09%5B1%20806-813%5D&c=%7B76%7Dd10418701c9b3100000%09%5B1%20806-813%5D&c=%7B76%7Dfded18701c9b3100000%09%5B1%20806-813%5D&c=%7B76%7Dfded18701c9b3100000%09%5B1%20806-813%5D&c=%7B76%7Dfded18701c9b3100000%09%5B1%20806-813%5D&c=%7B76%7De64a18701c9b3100000%09%5B1%20806-813%5D&c=%7B76%7De64a18701c9b3100000%09%5B1%20806-813%5D&c=%7B76%7Da43f18701c9b3100000%09%5B1%20813%5D&c=%7B76%7D804318701c9b3500000%09%5B5%20807%5D&c=%7B76%7D804318701c9b3500000%09%5B5%20807%5D&c=%7B76%7D804318701c9b3500000%09%5B5%20807%5D&c=%7B76%7D3b5718701c9b3700000%09%5B7%20806%5D&c=%7B76%7D3b5718701c9b3700000%09%5B7%20806%5D&c=%7B76%7D3b5718701c9b3700000%09%5B7%20806%5D&c=%7B76%7Dfe5718701c9b3900000%09%5B1%20852%5D&c=%7B76%7Dfe5718701c9b3900000%09%5B1%20852%5D&c=%7B76%7Dfe5718701c9b3900000%09%5B1%20852%5D&c=%7B76%7Dfe5718701c9b3900000%09%5B1%20852%5D&c=%7B76%7Dfe5718701c9b3900000%09%5B1%20852%5D&c=%7B76%7Dfe5718701c9b3900000%09%5B1%20852%5D&c=%7B76%7Dfe5718701c9b3900000%09%5B1%20852%5D&c=%7B76%7Dfe5718701c9b3900000%09%5B1%20852%5D&c=%7B76%7Dfe5718701c9b3900000%09%5B1%20852%5D&c=%7B76%7Dfe5718701c9b3900000%09%5B1%20852%5D&c=%7B76%7Dfe5718701c9b3900000%09%5B1%20852%5D&c=%7B76%7Dfe5718701c9b3900000%09%5B1%20852%5D&c=%7B76%7Dfe5718701c9b3900000%09%5B1%20852%5D&c=%7B76%7De5f018701c9b3900000%09%5B1%20854%5D&c=%7B76%7De5f018701c9b3900000%09%5B1%20854%5D&c=%7B76%7De5f018701c9b3900000%09%5B1%20854%5D&c=%7B76%7De5f018701c9b3900000%09%5B1%20854%5D&c=%7B76%7Db68a18701c9b3700000%09%5B7%20854%5D&c=%7B76%7Db68a18701c9b3700000%09%5B7%20854%5D&c=%7B76%7Db68a18701c9b3700000%09%5B7%20854%5D&c=%7B76%7Db68a18701c9b3700000%09%5B7%20854%5D&c=%7B76%7Db68a18701c9b3700000%09%5B7%20854%5D&c=%7B76%7Db68a18701c9b3700000%09%5B7%20854%5D&c=%7B76%7Db68a18701c9b3700000%09%5B7%20854%5D&c=%7B76%7Db68a18701c9b3700000%09%5B7%20854%5D&c=%7B76%7Db68a18701c9b3700000%09%5B7%20854%5D&f=16b%20ID%3F36h%20CH%3F4d%2020x).
What unit is the temperature in, is it °F x10?

snicker, 2020-07-20:
yes, exactly

snicker, 2020-07-20:
I will also try changing the mode to Celsius on the device and see if the data is reported differently

snicker, 2020-07-20:
[bitbench link with deduplicated data and F and C](https://triq.net/bitbench?c=%7B75%7D8dc818701c9b3a00000%20%5B2%20791%20261.666666666667%5D&c=%7B75%7D966f18701c9b3a00000%20%5B2%20793%20262.777777777778%5D&c=%7B75%7Dcfbd18701c9b3a00000%20%5B2%20797%20265%5D&c=%7B75%7De35418701c9b3a00000%20%5B2%20798%20265.555555555556%5D&c=%7B75%7D67be18701c9b3a00000%20%5B2%20802%20267.777777777778%5D&c=%7B75%7D7c1918701c9b3a00000%20%5B2%20804%20268.888888888889%5D&c=%7B76%7D06de18701c9b3300000%20%5B3%20804%20268.888888888889%5D&c=%7B76%7D6a1018701c9b3300000%20%5B3%20806%20270%5D&c=%7B74%7D43ad18701c9b3400000%20%5B4%20806%20270%5D&c=%7B75%7De31e18701c9b3600000%20%5B6%20806%20270%5D&c=%7B76%7Dcaa318701c9b3100000%20%5B1%20806%20270%5D&c=%7B76%7D3b5718701c9b3700000%20%5B7%20806%20270%5D&c=%7B75%7D851718701c9b3200000%20%5B2%20807%20270.555555555556%5D&c=%7B76%7D804318701c9b3500000%20%5B5%20807%20270.555555555556%5D&c=%7B76%7Da43f18701c9b3100000%20%5B1%20813%20273.888888888889%5D&c=%7B76%7Dfe5718701c9b3900000%20%5B1%20852%20295.555555555556%5D&c=%7B76%7De5f018701c9b3900000%20%5B1%20854%20296.666666666667%5D&c=%7B76%7Db68a18701c9b3700000%20%5B7%20854%20296.666666666667%5D&c=%7B76%7D29c718701c9b3700000%20%5B7%20856%20298%5D&c=%7B76%7D326018701c9b3700000%20%5B7%20858%20298.888888888889%5D&c=%7B76%7D326018701c9b3700000%20%5B7%20858%20299%5D&f=16b%20ID%3F36h%20CH%3F4d%2020b)

I can't seem to find a correlation to those first 16 bits and the temperature/channel.. there must be some kind of munging going on

zuckschwerdt, 2020-07-20:
Maybe there is something else going on, why are your samples not starting with 0xaaaaa (or 0x55555) like in the ook picture posted above (preamble of 20x `1` `0` pulses)?

snicker, 2020-07-20:
I was processing the cu8 files using @merbanan `-X "n=AMTX3107,m=FSK_PCM,s=484,l=484,r=10000,g=10000,preamble={16}0x2dd4"` which should be excerpting the preamble from the codes, right?

merbanan, 2020-07-20:
From the manual: 

2.2 Product Overview
```
CHANNEL slide switch
Assign each sensor to a different channel 1,
2, 3, 4, 5, 6 or 7. 
```

This means we need 3 bits to encode the channel (sensor id).

2.5.1 Low Battery Icon

This needs 1 bit, together with the channel this forms a 4-bit nibble which is nice.

6.1 Measurement Specifications

Resolution 0.1 °F/°C

Range -40 to 140F. This gives a dynamic range of 100F and with the resolution of 10 per degree F we get 100*10=1000. log2(1024)=10. So at least 10 of those bits of 16 should be used for the temperature. Most likely 12 bits because that is "nice".

So lets look at a few bitbuffers:

```
{76}caa3 18701c9b3 1	[1 806]
{74}43ad 18701c9b3 4	[4 806]
{76}3b57 18701c9b3 7	[7 806]
{76}6a10 18701c9b3 3	[3 806]
{75}e31e 18701c9b3 6    [6 806 270]
```
These should only differ in the channel setting. What must contain the temperature seems scrambled. But if we do a linearity test we get this:

caa3⊻43ad = 890E
caa3⊻3b57 = F1F4
caa3⊻6a10 = A0B3
caa3⊻e31e = 29BD

43ad⊻3b57 = 78FA
43ad⊻6a10 = 29BD
43ad⊻e31e = A0B3

3b57⊻6a10 = 5147
3b57⊻e31e = D849

6a10⊻e31e = 890E

As can be seen we have 3 matches:
caa3⊻6a10 = A0B3
43ad⊻e31e = A0B3

caa3⊻e31e = 29BD
43ad⊻6a10 = 29BD

caa3⊻43ad = 890E
6a10⊻e31e = 890E

Channel configuration
caa3 1, 1->001
43ad 4, 4->100
6a10 3, 3->011
e31e 6, 6->110

So it looks like the values are scrambled with a linear transform that depend on the channel.

merbanan, 2020-07-20:
@snicker please note that I forgot to adjust for the sample rate. For live capture you need to divide the s and l values by 4.

merbanan, 2020-07-20:
I'm note sure where to go from now so lets call in the experts. @evilpete care to look at this?

merbanan, 2020-07-20:
@snicker please note that channel changes needs the batteries removed for a while. And i think we need more readings of the same temperature but with different channels.

snicker, 2020-07-20:
I can confirm that I removed the batteries long enough for each channel change. the best data is the set of constant temperature at 80.6F: 5/7 channels are represented:

```
{76}6a1018701c9b3300000 [3 806 270]
{74}43ad18701c9b3400000 [4 806 270]
{75}e31e18701c9b3600000 [6 806 270]
{76}caa318701c9b3100000 [1 806 270]
{76}3b5718701c9b3700000 [7 806 270]
```

snicker, 2020-07-20:
I will try to gather some more data for all 7 channels.. keeping temperature constant was a bit of a challenge, ha

snicker, 2020-07-20:
all 7 channels @ 86.1F:
```
{76}b28a18701c9b3100000 [1 861 300.555555555556]
{75}ca7018701c9b3200000 [2 861 300.555555555556]
{76}123918701c9b3300000 [3 861 300.555555555556]
{74}3b8418701c9b3400000 [4 861 300.555555555556]
{76}e3cd18701c9b3500000 [5 861 300.555555555556]
{75}9b3718701c9b3600000 [6 861 300.555555555556]
{76}437e18701c9b3700000 [7 861 300.555555555556]
```

merbanan, 2020-07-20:
If the native reading is in C it is possible that 2 readings map to the same value in F.

snicker, 2020-07-20:
finding many of the same values with the linearity test on that dataset

```
+------+------+------+------+------+------+------+------+------+------+
| chan | temp |      | b28a | ca70 | 1239 | 3b84 | e3cd | 9b37 | 437e |
+------+------+------+------+------+------+------+------+------+------+
|    1 |  861 | b28a | 0    | CA70 | D849 | E3CD | 0    | 9B37 | D849 |
|    2 |  861 | ca70 | CA70 | 0    | D849 | 3B84 | 3B84 | A0B3 | 78FA |
|    3 |  861 | 1239 | D849 | D849 | 0    | 3B84 | 0    | A0B3 | D849 |
|    4 |  861 | 3b84 | E3CD | 3B84 | 3B84 | 0    | 0    | A0B3 | 78FA |
|    5 |  861 | e3cd | 0    | 3B84 | 0    | 0    | 0    | A0B3 | D849 |
|    6 |  861 | 9b37 | 9B37 | A0B3 | A0B3 | A0B3 | A0B3 | 0    | D849 |
|    7 |  861 | 437e | D849 | 78FA | D849 | 78FA | D849 | D849 | 0    |
+------+------+------+------+------+------+------+------+------+------+
```

merbanan, 2020-07-20:
1239^3b84 = 29Bd, 3^4=7
b28a^9b37 = 29Bd, 1^6=7

snicker, 2020-07-20:
I don't know what this means but the data from the incomplete set also will xor to the same table
```
+------+------+------+------+-----+------+------+-----+------+------+
| chan | temp |      | caa3 | N/A | 6a10 | 43ad | N/A | e31e | 3b57 |
+------+------+------+------+-----+------+------+-----+------+------+
|    1 |  806 | caa3 | 0    |   - | A0B3 | 890E |   - | 29BD | F1F4 |
|    2 |  806 | N/A  | -    |   - | -    | -    |   - | -    | -    |
|    3 |  806 | 6a10 | A0B3 |   - | 0    | 29BD |   - | 890E | 5147 |
|    4 |  806 | 43ad | 890E |   - | 29BD | 0    |   - | A0B3 | 78FA |
|    5 |  806 | N/A  | -    |   - | -    | -    |   - | -    | -    |
|    6 |  806 | e31e | 29BD |   - | 890E | A0B3 |   - | 0    | D849 |
|    7 |  806 | 3b57 | F1F4 |     | 5147 | 78FA |   - | D849 | 0    |
+------+------+------+------+-----+------+------+-----+------+------+
```

merbanan, 2020-07-20:
Because the transform is linear we get the same output. The question is how do we use this to reverse the values we want.

merbanan, 2020-07-20:
Can you trigger the low battery condition? I postulate that it will affect the encoded value.

merbanan, 2020-07-20:
Fun fact if we take the pair list:

1239^3b84 = 29Bd, 3^4=7
b28a^9b37 = 29Bd, 1^6=7
ca70^e3cd = 29Bd, 2^5=7

and change the order we get this:

1239^29Bd = 3b84
b28a^29Bd = 9b37
ca70^29Bd = e3cd

that gives we can calculate the channel 0 value by doing the same:
437e^29Bd = 6AC3

merbanan, 2020-07-20:
So the encryption key depends on the selected channel bits.

But why isn't channel 0 used? Well it might be because A^0=A. Ie setting a 0 channel might leak the key.

snicker, 2020-07-20:
I can't seem to find any batteries that are low enough to trigger the warning and also power the device. All of my previous data collected, ironically, was showing the battery icon (half full?), and I did find a fresher set of batteries that had a full charge, but the same values were reported:

```

+------+------+--------+------+------+------+------+------+------+------+
| chan | temp | first4 | 3f42 | 47b8 | 9ff1 | b64c | 6e05 | 16ff | ceb6 |
+------+------+--------+------+------+------+------+------+------+------+
|    1 |  879 | 3f42   | 0    | 78FA | A0B3 | 890E | 5147 | 29BD | F1F4 |
|    2 |  879 | 47b8   | 78FA | 0    | D849 | F1F4 | 29BD | 5147 | 890E |
|    3 |  879 | 9ff1   | A0B3 | D849 | 0    | 29BD | F1F4 | 890E | 5147 |
|    4 |  879 | b64c   | 890E | F1F4 | 29BD | 0    | D849 | A0B3 | 78FA |
|    5 |  879 | 6e05   | 5147 | 29BD | F1F4 | D849 | 0    | 78FA | A0B3 |
|    6 |  879 | 16ff   | 29BD | 5147 | 890E | A0B3 | 78FA | 0    | D849 |
|    7 |  879 | ceb6   | F1F4 | 890E | 5147 | 78FA | A0B3 | D849 | 0    |
+------+------+--------+------+------+------+------+------+------+------+

```

merbanan, 2020-07-20:
With linearity we might get that F4 = key ^ temperature. The key depends on the channel somehow (maybe repeated 4 times to form a 16 bit value). But that might not matter. If the guess is correct then there should be a key in 16-bit space that the following is true for:

key ^ 3f42 > key ^ b28a > key ^ caa3

merbanan, 2020-07-21:
With the above values I got 16 thousand keys but with one more value I got it down to 8192 values. Can you grab some more values with channel 1?

I also found this in the bitbench list:

{76}fe5718701c9b3900000	[1 852]
{76}e5f018701c9b3900000	[1 854]

Ie value 9. But in bits that is 1001. So I think we can confirm that the channel is present in clear text. And the obvious reason for that is that it is needed to build the decryption key. And the operation to build the key is at least piece wise linear.

snicker, 2020-07-21:
here are a few more values on just channel 1:
```
{76}f0ff18701c9b3100000 [1 870 305.555555555556]
{76}dc1618701c9b3100000 [1 869 305]
{76}c7b118701c9b3100000 [1 867 303.888888888889]
{76}85c418701c9b3100000 [1 865 302.777777777778]
{76}9e6318701c9b3100000 [1 863 301.666666666667]
{76}b28a18701c9b3100000 [1 861 300.555555555556]
```

let me know if you need more.

merbanan, 2020-07-21:
In mathematical terms this seems to hold:

f(channel_nr, temperature, unk?) = bistream_value

and any
(1) channel_nr_1 ^ channel_nr_2 = constant_1 ->
(2) f(channel_nr_1, temperature) ^ f(channel_nr_2, temperature) = constant_2

What functions will satisfy (2)?

If we set constant_1 to F we then can use the pair 0xE,1 and 0xC,3 as channel_nr. We then assume the function is temp^key^channel_nr with key set to 0xC. The temperature is set to A, this gives the following values:

A^C^E = 8
A^C^E = 7
A^C^C = A
A^C^3 = 5

8^7 = F and A^5 = F

so this function seems to satisfy (2).

merbanan, 2020-07-21:
Can you get recordings at value 852 or 854?

merbanan, 2020-07-21:
We need alot more values with different temperatures.

merbanan, 2020-07-21:
Around 40 per channel and the same temps.

snicker, 2020-07-21:
```
{76}fe5718701c9b3900000 [1 852]
```
Thought i got a reading at 854 too but it was a bad signal.

For the 40 per channel, same temps, you mean we need about 40 distinct temperatures for each channel, but the same temperatures on each channel? I'll need to set up a testing rig somehow to keep the temperature constant for that. might be a weekend project.

merbanan, 2020-07-21:
Well set a channel and raise/low the temp and record the values. Well see how many we need.

merbanan, 2020-07-21:
Just get as many unique values as possible.

snicker, 2020-07-21:
got it. Is there any way to publish the raw values to mqtt? I can get a testing rig set up to record and snap a webcam pretty quickly and collect data passively over the next couple days

snicker, 2020-07-21:
tried this:
`rtl_433 -Y level=-1.0 -f 915000000 -R 0 -X "n=AMTX3107,m=FSK_PCM,s=484,l=484,r=10000,g=10000,preamble={16}0x2dd4" -F mqtt://x.x.x.x:1883,retain=1,devices=rtl_433/id`
but nothing is published. If I read straight from a recorded cu8 file it will publish the data.

merbanan, 2020-07-21:
Divide the s and l values by 4. That should work.

merbanan, 2020-07-21:
-f > 800MHz implies -s 1000k

merbanan, 2020-07-21:
Can you document your setup also. Might be a Good example for others.

snicker, 2020-07-21:
setting 's' and 'l' to 121 worked to publish, but now I'm getting much longer values (though the message is still there):
`{250}e5f018701c9b3900000000002970000000ee000007000000000000000000000`

My plan for the set up is to have HomeAssistant read the data in to a sensor, and any time it changes, HA can take a snapshot from a webcam and send the whole thing as a telegram message to me. I'll still have to do some manual data entry later but the  collection will be passive at least. I won't be able to set this up until tonight or tomorrow.

merbanan, 2020-07-21:
Change the r and g values also. That should return everything back to what it looked like before.

merbanan, 2020-07-21:
Maybe you can train opencv to identify the digits 😉

merbanan, 2020-07-21:
Anyway math says we need to form enough equations to solve the unknowns for.

snicker, 2020-07-21:
ha, the thought crossed my mind to do image processing. I might just

merbanan, 2020-07-21:
Another fun relation.

b28a^ca70^3b84=437e
1^2^4=7

This is the scrambled value for when the channel is 7. So by knowing only 1,2 and 4 we can derive the value for 7.

snicker, 2020-07-21:
[image]
we're cooking! I'll continue to collect data over the next day or so. I'm saving the images and the data too in case I get motivated to do the image processing part. Otherwise some manual data entry is in order

merbanan, 2020-07-21:
Do you know if the native data is F or C?

[31 more replies]