merbanan/rtl_433 #1425: Support for TFA.303221.02

Fakstate, 2020-07-08:
Hello !

I am new to GitHub but I have been using RTL_433 for a long time, I recently bought an cheap outdoor temperature/humidity sensor TFA 30.3221.02. I noticed that there was no decoder installed in rtl_433 for this sensor, so I made one from an implementation in [RFFHEM](https://github.com/RFD-FHEM/RFFHEM/blob/a1222ecd68e1febeca894ebdabfe71c22ae7fa06/FHEM/14_SD_WS.pm#L591). 

I have never coded in C so my code may be incorrect and needs correction. There are also a lot of unknown bits which could be a checksum but it goes far beyond my skills.

https://github.com/Fakstate/TFA-30.3221.02/blob/master/tfa_30_3221.c

zuckschwerdt, 2020-07-08:
Looks great. Can you add a copyright header? See e.g. https://github.com/merbanan/rtl_433/blob/master/src/devices/proove.c

Usually you need to fork the rtl_433 repo, `git clone` the fork (SSH, not HTTPS), add a branch, put your changes there, then push to github and pull request. If we take your source as-is you won't get any attribution in the commits.

merbanan, 2020-07-19:
@Fakstate we need sample signals also to add the decoder.

rtega, 2020-08-24:
Correction: the code does work but I have to tune in to frequency 434Mhz or rtl_433 doesn't catch the signals. I didn't catch any samples at all with -S all so I decided to have a look at the raw data with CubicSDR and noticed that my sensor outputs at 434Mhz...  After tuning for 434Mhz (-f 434M) the supplied code does work. I've included two samples here.

[samples.zip](https://github.com/merbanan/rtl_433/files/5116289/samples.zip)

rtega, 2020-08-24:
The problem is that other signals no longer get caught when I tune in to 434Mhz. Is there a way around this?

rtega, 2020-08-24:
Tuning into 433.98Mhz seems to do the trick in my case.

zuckschwerdt, 2020-08-24:
There can be some drift with changing temperature, so all your signals should be well inside the frequency window (some distance to the edges, and preferably not dead center).
You can widen the window with a higher sample rate (`-s 1024k`), worth a shot.

rtega, 2020-08-24:
Mmm, that doesn't seem to help except increasing CPU usage which is not so good on a Raspberry pi zero. I'm hopping between 434Mhz and 433.92Mhz with 60 second hop interval. That seems to solve it mostly.

zuckschwerdt, 2020-08-24:
Yes, the default is 250k so 1024k will increase to 4x the load. Your hopping setup looks good, you always capture the core frequencies and also enable frequencies a bit farther out.
(at 250k sample rate the bands here are 434M +/-125k and 433.92M +/-125k, i.e. 433.88M to 434.12M and 433.80M to 434.04M)
If you you're constrained on resources: use a conf or -R options to only enable the used protocols. This might disable FSK if it's not used, which is the biggest load.

rtega, 2020-08-24:
Mmm, still getting spurious values. For example, the attached sample generates this output:
```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-08-24 14:29:36
model     : TFA-30322102 Sensor ID : 4
Channel   : 1            Battery   : 1             Temperature: -27.20 C     Humidity  : 48 %          Test ?    : No
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-08-24 14:29:36
model     : TFA-30322102 Sensor ID : 4
Channel   : 1            Battery   : 1             Temperature: -50.00 C     Humidity  : 0 %           Test ?    : No

```

Or this:
```
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-08-24 14:31:16
model     : TFA-30322102 Sensor ID : 4
Channel   : 1            Battery   : 1             Temperature: -50.00 C     Humidity  : 0 %           Test ?    : No
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
time      : 2020-08-24 14:31:17
model     : TFA-30322102 Sensor ID : 4
Channel   : 1            Battery   : 1             Temperature: -50.00 C     Humidity  : 0 %           Test ?    : No

```

[samples2.zip](https://github.com/merbanan/rtl_433/files/5117799/samples2.zip)

zuckschwerdt, 2020-08-25:
We likely need to figure out the checksum in the last 8 bits to ensure valid reception. Can you dump a good number of received bitbuffers by adding `bitbuffer_print(bitbuffer);` after `bitbuffer_invert()`?

rtega, 2020-08-26:
Here you go. I have no idea what to go about this. Any pointers on this?
[bitbuffers.zip](https://github.com/merbanan/rtl_433/files/5132410/bitbuffers.zip)

zuckschwerdt, 2020-08-26:
Ah great, thanks. Looks like reverse Galois with byte reflect, using gen 31 key f4.
I'll put this code in a branch and add this check soon.

rtega, 2020-08-26:
Ah, nice. Probably the same as tfa_drop_30.3233.c?

rtega, 2020-08-26:
Looks like this code works nicely as you suggested
```
    /*
     *      * Validate checksum
     *           */
    observed_checksum = bytes[4];
    computed_checksum = lfsr_digest8_reflect(bytes, 4, 0x31, 0xf4);
    if (observed_checksum != computed_checksum) {
            return DECODE_FAIL_MIC; 
                                    }

```

zuckschwerdt, 2020-08-26:
Very likely, yes! Very well spotted, didn't notice that.

zuckschwerdt, 2020-08-26:
Thanks for testing already!

rtega, 2020-08-26:
A simple grep turned up that result ;)

rtega, 2020-08-26:
While we're on it: is there a way to get rid of duplicates? The sensor sends in bursts of 2 (others in bursts of 3).

zuckschwerdt, 2020-08-26:
Depends on how far the transmission are apart. Can you check that with `-M time:usec`?
If the repeated transmission is not close enough (10x the longest usual gap) then rtl_433 will always treat it as a new event. And rtl_433, by design, will not handle multiple events -- it just decodes and outputs "what's on the air".

rtega, 2020-08-27:
I get timings like:
2020-08-27 07:46:16.182671
2020-08-27 07:46:16.474012

2020-08-27 07:47:06.182481
2020-08-27 07:47:06.476697

2020-08-27 07:48:46.182612
2020-08-27 07:48:46.477002

zuckschwerdt, 2020-08-28:
Looks like the sender is rather precise. But the gap of 290ms is far too big. If you are not logging this to an aggregator or database capable of coalescing, say RRD, Graphite, InfluxDB, … then perhaps look at the examples dir for scripts that can filter the output.

rtega, 2020-08-29:
Well, I can filter it out so it's not such a big issue.

zuckschwerdt, 2020-08-29:
This is now in branch feat-tfa303221 (294efbe), please test.

rtega, 2020-08-29:
Tested and works nicely! Thanks a lot!

zuckschwerdt, 2020-09-03:
Thanks for confirming, this got merged from #1426 now and some cleanup and the digest checksum added with 67b0131.

closed 2020-09-03 by zuckschwerdt

Q2090, 2021-10-05:
Hi, just installed the RTL-433, because I have the TFA 30.3221.02. It does not show these (but shows other sensors so it works).
Is there anything special I should do to enable these sensors?

rtega, 2021-10-05:
Hi, you need to tune into a different frequency: https://github.com/merbanan/rtl_433/issues/1425#issuecomment-678960163

Add the line ''''frequency     433.92M'''' to your `rtl_433.conf` file

Q2090, 2021-10-05:
> Hi, you need to tune into a different frequency: [#1425 (comment)](https://github.com/merbanan/rtl_433/issues/1425#issuecomment-678950175)
> 
> Add the line `frequency 434M` to your `rtl_433.conf` file.

I did...

Do I need to add another flag?

rtl_433 -f 434000000
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/lambert/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 122 out of 149 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 434.000MHz.
Allocating 15 zero-copy buffers
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2021-10-05 18:27:21                    brand     : OS
model     : Oregon-THGR810                         House Code: 106
Channel   : 1            Battery   : 1             Celsius   : 16.70 C       Humidity  : 98 %
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

Q2090, 2021-10-06:
A rtl_433 -R help does not show protocol 164. Bummer, I must have installed an old version...
I did a
sudo apt-get install rtl-433
and
sudo apt-get install rtl-sdr
on a fresh install: Ubuntu 20.04.3 LTS

Version unknown:
$ rtl_433 -V
rtl_433 version unknown inputs file rtl_tcp RTL-SDR SoapySDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/<user>/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

rtega, 2021-10-06:
You  probably need the latest version. Download and compile should suffice ;)

Q2090, 2021-10-06:
I upgraded to Ubuntu 21.04, and that included a new version of rtl_433, super!
It's now working as expected. @rtega thanks for the suggestion about versions!

rtega, 2021-10-06:
You're welcome. Happy it worked for you!