merbanan/rtl_433 #1427: Support for the weather station Alecto WS3000 (AlectoV2 protocol)

borsdorff, 2020-07-10:
I wonder if it is possible to add support for the Weather Station WS3000 from Alecto (ver 2). I already tried the available Fine Offset protocols of rtl_433 however nothing seems to work. I  tried to get it running with rtl_433 but I am stuck some how. It would be great if someone could help or contribute ideas.

(a) with gqxt I can isolate the signal (please see attachment gqrx.jpg). It is at 868346.600kHz and comes about every 40 sec. Sometimes even two packages can be detected.

(b) I had a look at the signal using Audacity. It clearly consists of 80 bits and follows the protocol description of other projects e.g. by the Nodo software (please see the header of the attached file plugin_009.c)

(c) I tried to run rtl_433 with the -A parameter (please have a look at the file rlt_433_output.txt ). The software suggest to use -X 'n=name,m=FSK_PCM,s=1,l=1,r=1024'. Is that correct considering the Audacity output before? Because the with this command I get many packages and not only one with 80 bit. Do you have an idea what is going wrong here? 

Thanks a lot. I would be so great to have support for this Weather Stations in this project. 

Files:

[image]
[image]
[Plugin_009.c.pdf](https://github.com/merbanan/rtl_433/files/4905299/Plugin_009.c.pdf)
[rtl_433_output.txt.pdf](https://github.com/merbanan/rtl_433/files/4905302/rtl_433_output.txt.pdf)

References:

I found some projects (Arduino based) that implement support for this stations and give some interesting information about the protocol.

http://mientki.ruhosting.nl/data_www/raspberry/doc/ws3000_hack.html

https://github.com/SevenW/wirelessweather/blob/master/WeatherStationFSKv2/WeatherStationFSKv2.ino

http://www.nodo-domotica.nl/index.php/AlectoWeerstations_868MHz

peteakalad, 2020-07-10:
Might work with the existing support for WH1080? 

Try adding this argument:
-Y classic

borsdorff, 2020-07-11:
You are right, thanks a lot :)

With the classic option it is detected as Fine Offset WH1050.

closed 2020-07-11 by borsdorff