merbanan/rtl_433 #1429: Arduino 433 MHz transmitter

bodokaiser, 2020-07-11:
Hi,

I would like to use `rtl_433` to receive signals sent from a [433 MHz Arduino transmitter][1].

I did the [protocol analysis][2] and found that the information is decoded using an OOK PWM modulation scheme. In total, a message is repeated ten times. The message starts with a 1 bit for synchronization, thereafter you find the data bits. A 1 bit is decoded as 1050 us high followed by 350 us low while a 0 bit is decoded as 350 us high followed by 1050 us low. The exact protocol is also explained in the [source code][3] of the corresponding Arduino library.

I expected to se `rtl_433` with custom demodulation but did not find any useful data:

```shell
rtl_433 -s 2048k -X 'n=arduino,m=OOK_PWM,short=1050,long=350,sync=350,reset=14000'
```

```
rtl_433 version unknown inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/Users/bodokaiser/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

	Consider using "-M newmodel" to transition to new model keys. This will become the default someday.
	A table of changes and discussion is at https://github.com/merbanan/rtl_433/pull/986.

Registered 107 out of 136 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-103 108-116 119 121 124-128 131-136 ]
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
Sample rate set to 2048000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
```

```
time      : 2020-07-11 21:53:21
model     : arduino      count     : 4             num_rows  : 4             rows      :
len       : 98           data      : 0000000000000000000000000,
len       : 2            data      : 0,
len       : 1            data      : 0,
len       : 0            data      :
codes     : {98}0000000000000000000000000, {2}0, {1}0, {0}
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-11 21:53:21
model     : arduino      count     : 4             num_rows  : 4             rows      :
len       : 126          data      : 00000000000000000000000000000000,
len       : 2            data      : 0,
len       : 1            data      : 0,
len       : 0            data      :
codes     : {126}00000000000000000000000000000000, {2}0, {1}0, {0}
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-11 21:53:21
model     : arduino      count     : 4             num_rows  : 4             rows      :
len       : 126          data      : 00000000000000000000000000000000,
len       : 2            data      : 0,
len       : 1            data      : 0,
len       : 0            data      :
codes     : {126}00000000000000000000000000000000, {2}0, {1}0, {0}
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-11 21:53:22
model     : arduino      count     : 4             num_rows  : 4             rows      :
len       : 126          data      : 00000000000000000000000000000000,
len       : 2            data      : 0,
len       : 1            data      : 0,
len       : 0            data      :
codes     : {126}00000000000000000000000000000000, {2}0, {1}0, {0}
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-11 21:53:22
model     : arduino      count     : 4             num_rows  : 4             rows      :
len       : 126          data      : 00000000000000000000000000000000,
len       : 2            data      : 0,
len       : 1            data      : 0,
len       : 0            data      :
codes     : {126}00000000000000000000000000000000, {2}0, {1}0, {0}
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-11 21:53:22
model     : arduino      count     : 4             num_rows  : 4             rows      :
len       : 126          data      : 00000000000000000000000000000000,
len       : 2            data      : 0,
len       : 1            data      : 0,
len       : 0            data      :
codes     : {126}00000000000000000000000000000000, {2}0, {1}0, {0}
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-11 21:53:23
model     : arduino      count     : 1             num_rows  : 1             rows      :
len       : 0            data      :
codes     : {0}
```

**Any suggestions on how to change the parameters to receive the message data?**

[1]: https://create.arduino.cc/projecthub/MisterBotBreak/how-to-communicate-using-433mhz-modules-9c20ed
[2]: https://nbviewer.jupyter.org/github/bodokaiser/arduino-rtlsdr/blob/master/notebooks/00011_00100.ipynb
[3]: https://github.com/sui77/rc-switch/blob/master/RCSwitch.cpp#L82

zuckschwerdt, 2020-07-11:
You generally can't swap `short` and `long`, also `sync` can't match `short` or `long` -- it needs to be a clearly different symbol.
We might need to implement some sanity checks to help get the parameters right.

bodokaiser, 2020-07-11:
Is there a workaround? I would do some post-processing on the json stream anyway.

zuckschwerdt, 2020-07-11:
You just need to give the shorter timing as `short` and respectivly for `long`. Just drop the `sync`.
`rtl_433 -s 2048k -X 'n=arduino,m=OOK_PWM,short=350,long=1050,gap=1200,reset=14000,invert' should do.

zuckschwerdt, 2020-07-11:
You can also convert an I/Q sample file (from `-S all`) to pulse data with `rtl_433 -w FILENAME.ook FILENAME.cu8` and drop that on https://triq.org/pdv/ to get a feel for the parameters.

bodokaiser, 2020-07-12:
Using `rtl_433 -s 2048k -X 'n=arduino,m=OOK_PWM,short=350,long=1050,gap=1200,reset=14000,invert,reflect'` gets some configurations right. The following table summarizes the configurations for transmission `transmitter.send(code, nbits)` and what `rtl_433` outputs.

| code or binary value | number of bits to transmit | transmitted | received | correct |
| -------------------- | -------------------------- | ------------ | -------- | ------- |
| 1 | 1 | 0b1 | 0x0 = 0b0 = 0 | ❌ |
| 1 | 2 | 0b01 | 0x0 = 0b0 = 0 | ❌ |
| 2 | 2 | 0b10 | 0x0 = 0b0 = 0 | ❌ |
| 2 | 3 | 0b010 | 0x0 = 0b0 = 0 | ❌ |
| 2 | 4 | 0b0010 | 0x04 = 0b100 = 4 | ❌ |
| 2 | 5 | 0b00010 | 0x08 = 0b1000 = 8 | ❌ |
| 3 | 2 | 0b11 | 0x00 = 0b00 = 0 | ❌ |
| 3 | 3 | 0b011 | 0x00 = 0b00 = 0 | ❌ |
| 4 | 3 | 0b100 | 0x0 = 0b0 = 0 |  ❌ |
| 4 | 4 | 0b0100 | 0x02 = 0b10 = 2 |  ❌ |
| 4 | 5 | 0b00100 | 0x04 = 0b100 = 4 | ✅ |
| 5 | 5 | 0b00101 | 0x14 = 0b10100 = 20 |  ❌ |
| 10 | 5 | 0b01010 | 0x0a = 0b1010 = 10 | ✅ |

[`transmitter.send(code, nbits)`](https://github.com/sui77/rc-switch/blob/master/RCSwitch.cpp#L511-L516) does

```c
    for (int i = length-1; i >= 0; i--) {
      if (code & (1L << i))
        this->transmit(protocol.one);
      else
        this->transmit(protocol.zero);
    }
```

In the attached file you find the result of `rtl_433 -s 2048k -S all` and `rtl_433 -w g003_433.92M_2048k.txt g003_433.92M_2048k.cu8` for [Pulseview](https://triq.org/pdv/).

[g001_433.92M_2048k.txt](https://github.com/merbanan/rtl_433/files/4908425/g001_433.92M_2048k.txt)

zuckschwerdt, 2020-07-12:
That looks like the demod isn't picking up the bits reliably. You should check the cu8 if the signal is strong enough and not close to the edges (but not exactly centered). You can use https://triq.org/iqs/ to get a spectrogram.

bodokaiser, 2020-07-12:
I got the antennas side-by-side, so there is not much I can do to increase the signal. I have a GNU radio flowgraph with some xlating filter and thresholding that outputs a good signal.

If `rtl_433` could read from `stdin` I can pipe data from GNU radio via a ZeroMQ sink to a process that pipes the data further to `rtl_433`.

[image]

ant-thomas, 2020-12-13:
Is it worth looking at the RadioHead library which is already supported with rtl_433?

I've been using it successfully and reliably for a while now.

gdt, 2023-09-29:
Is this issue closable?  If not what are the next steps?

gdt, 2024-06-04:
abandoned it seems

closed 2024-06-04 by gdt