merbanan/rtl_433 #1430: Add support for La Crosse TX141-B

bsharper, 2020-07-13:
I have a similar transmitter to what pictured in https://github.com/merbanan/rtl_433/issues/1268, but without humidity. The model number is TX141-B. 

I ran the command that @tvass ran in issue #1268 and just changed the n= value. I have no clue what these values mean, but I did get somewhat similar results.

```
rtl_433 -X 'n=lacrosse-tx141-b,m=OOK_PWM,s=256,l=500,r=1888,g=0,t=0,y=748'
[...]
 _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-12 20:44:28
model     : lacrosse-tx141-b                       count     : 15            num_rows  : 15            rows      :
len       : 32           data      : 2a6d2a6b,
len       : 32           data      : 2a6d2a6b,
len       : 32           data      : 2a6d2a6b,
len       : 32           data      : 2a6d2a6b,
len       : 32           data      : 2a6d2a6b,
len       : 32           data      : 2a6d2a6b,
len       : 32           data      : 2a6d2a6b,
len       : 32           data      : 2a6d2a6b,
len       : 32           data      : 2a6d2a6b,
len       : 32           data      : 2a6d2a6b,
len       : 32           data      : 2a6d2a6b,
len       : 32           data      : 2a6d2a6b,
len       : 32           data      : 2a6d2a6b,
len       : 32           data      : 2a6d2a6b,
len       : 34           data      : 2a6d2a6bc
codes     : {32}2a6d2a6b, {32}2a6d2a6b, {32}2a6d2a6b, {32}2a6d2a6b, {32}2a6d2a6b, {32}2a6d2a6b, {32}2a6d2a6b, {32}2a6d2a6b, {32}2a6d2a6b, {32}2a6d2a6b, {32}2a6d2a6b, {32}2a6d2a6b, {32}2a6d2a6b, {32}2a6d2a6b, {34}2a6d2a6bc

[...]

time      : 2020-07-12 20:44:59
model     : lacrosse-tx141-b                       count     : 16            num_rows  : 16            rows      :
len       : 32           data      : 2a6d2b6c,
len       : 32           data      : 2a6d2b6c,
len       : 32           data      : 2a6d2b6c,
len       : 32           data      : 2a6d2b6c,
len       : 32           data      : 2a6d2b6c,
len       : 32           data      : 2a6d2b6c,
len       : 32           data      : 2a6d2b6c,
len       : 32           data      : 2a6d2b6c,
len       : 32           data      : 2a6d2b6c,
len       : 32           data      : 2a6d2b6c,
len       : 32           data      : 2a6d2b6c,
len       : 32           data      : 2a6d2b6c,
len       : 32           data      : 2a6d2b6c,
len       : 32           data      : 2a6d2b6c,
len       : 29           data      : 2a6d2b68,
len       : 5            data      : 98
codes     : {32}2a6d2b6c, {32}2a6d2b6c, {32}2a6d2b6c, {32}2a6d2b6c, {32}2a6d2b6c, {32}2a6d2b6c, {32}2a6d2b6c, {32}2a6d2b6c, {32}2a6d2b6c, {32}2a6d2b6c, {32}2a6d2b6c, {32}2a6d2b6c, {32}2a6d2b6c, {32}2a6d2b6c, {29}2a6d2b68, {5}98
```
At a glance it looks like these values could be similar to the BV3, minus the humidity field. So maybe it's transmitting something like
`[id] [id] [flags] [temp] [temp] [temp] [chk] [chk]`
instead of 
`[id] [id] [flags] [temp] [temp] [temp] [humi] [humi] [chk] [chk]`
but this is just a guess. 

Let me know if there's anything else I can provide or data I can try and capture. Thank you!

[image]

evilpete, 2020-07-13:
@bsharper 

According to the FCCID it uses   433.92 MHz 

try the command 
```
rtl_433 -f 433.6M -S unknown
```  

to generate some sample files, they will be named something like `g002_433.6M_250k.cu8`,  please tar/zip a hand full of samples and post them here.

By examining the samples one can  quickly figure out the  the data rate and derive the values for the modulation (m=) and long/short 'pulse' widths and with this info generate the correct options to use with '-X' ,
A "driver" may need to be written, if you know C you can do it yourself, else someone like me will volunteer and crank one out some evening when they are too bored :-) 

To look at them yourself, make sure you have [pulseview](https://sigrok.org/wiki/Downloads) installed and run
the command :
```
rtl_433  -W output.sr 
```

this will process the sample file and drop you into a slightly annoying program called pulseview and allow you to view and measure  the signal.

bsharper, 2020-07-14:
Thanks for the information @evilpete.

I was able to grab some samples using 433.92M (not 433.6M, which I assume is a typo)

I think I'm seeing data similar to what is described here with similar timings: https://github.com/merbanan/rtl_433/blob/master/src/devices/lacrosse_tx141x.c. 

A preamble of 5 pulses (instead of 4) followed by 32 high/low signals, repeated 15 times.

The temperature showing on the screen for the remote sensor while it was being captured was 29.4&deg;

Looking at the data in pulseview, I was able to get the 1s and 0s after the preamble, and I think it's what I assumed:
`[id] [id] [flags] [temp] [temp] [temp] [chk] [chk]`
Or as the number of bits per field:
`8 4 12 8`

So basically the same as what's in https://github.com/merbanan/rtl_433/issues/1268 minus the humidity fields.

*Sanity check in python*
````
data = "11010101100100110001101011011001"
names = ["id", "flags", "temp (c)", "checksum"]
fns = [None, None, (lambda x: (float(x)/10.0)-50 ), None]
widths = [8,4,12,8]
h = 0
for i, w in enumerate(widths):
    b = data[h:h+w]
    val = int(b, 2)
    if fns[i] != None:
        val = str(fns[i](val))
    print ("%s: %s (%s)" % (names[i].ljust(10), str(val).ljust(20), b))
    h+=w
````
Output:
````
id        : 213                  (11010101)
flags     : 9                    (1001)
temp (c)  : 29.400000000000006   (001100011010)
checksum  : 217                  (11011001)
````

*The temp doesn't match the bit value because the temp formula from lacrosse_tx141x.c is applied: (value / 10) - 50.

I guess next I'll take a closer look at lacrosse_tx141x.c to see if I'm able to add this to the existing code.

Let me know if there's anything that smells funny with anything above, and thanks again for your help.

Captures:
[tx141b_capture.zip](https://github.com/merbanan/rtl_433/files/4916809/tx141b_capture.zip)

zuckschwerdt, 2020-07-14:
- insert `bitbuffer_print(bitbuffer);` anywhere in the decoder code to inspect the current bitbuffer (to get raw values as seen by the decoder or e.g. check if the alignment is as expected)
- to pick data apart easily you can [use BitBench](https://triq.net/bitbench?c=y11010101100100110001101011011001&f=ID%3A8d%20FLAGS%3A4b%20TEMP_C%3A12d%20CHK%3A8h)

bsharper, 2020-07-16:
Thanks @zuckschwerdt for that info, being able to grab the bits and verify the changes in C matched up against my test python code was really useful. 

I've added the necessary code for this sensor to work. The changes were pretty minimal. The test flag works properly, as does low battery detection.

I've submitted the changes under pull request https://github.com/merbanan/rtl_433/pull/1434

Thanks everyone for their help with this!

bsharper, 2020-07-20:
[PR](https://github.com/merbanan/rtl_433/pull/1434) merged.

closed 2020-07-20 by bsharper