merbanan/rtl_433 #1431: ADD: support for ecowitt wh32

ciasco92, 2020-07-13:
hi, sorry for my bad english, with R820T I read the values of the WS68, WH40 and WH31E ecowitt sensors at 868 mhz, but I have a WH32 sensor that is not decoded.
with the command `rtl_433 -f 868300000 -S unknown` I get this:

`Registered 128 out of 156 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23                                                                                                                                                              25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-1                                                                                                                                                             56 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.300MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-13 09:18:49
model     : EcoWitt-WH40 id        : 51515
Total Rain: 0.0 mm       Extra Data: 0002b30000    Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-13 09:18:54
model     : EcoWitt-WS68 id        : 919
Battery Raw: 84          Battery   : 1             lux       : 6386          Wind Speed: 51            Wind Gust : 61            Wind dir  : 69            Extra Data: 31 104        Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-13 09:19:10
model     : EcoWitt-WS68 id        : 919
Battery Raw: 84          Battery   : 1             lux       : 6386          Wind Speed: 58            Wind Gust : 71            Wind dir  : 97            Extra Data: 31 104        Integrity : CRC
***** Saving signal to file g001_868.3M_1000k.cu8 (86352 samples, 262144 bytes)**
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-13 09:19:22
model     : AmbientWeather-WH31E                   id        : 9
Channel   : 1            Battery   : 1             Temperature: 22.4 C       Humidity  : 52 %          Extra Data: bf00000000    Integrity : CRC`

I hope not to be wrong, given the [files cu8](https://github.com/merbanan/rtl_433/files/4911603/filecu8.zip) obtained which should be wh32. 

if I try to read them with the command `rtl_433 -R 0 -X 'n=new,m=FSK_PCM,s=58,l=58,r=290' FILENAME.cu8` I get this:

`rtl_433 version 20.02-86-gd362397 branch master at 202007062107 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Disabling all device decoders.
Registered 1 out of 156 device decoding protocols [ ]
Test mode active. Reading samples from file: g001_868.3M_1000k.cu8
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.244416s
model     : new          count     : 1             num_rows  : 1             rows      :
len       : 35           data      : fffffffc0
codes     : {35}fffffffc0
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.244416s
model     : new          count     : 1             num_rows  : 1             rows      :
len       : 193          data      : f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f00
codes     : {193}f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f0f00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.244416s
model     : new          count     : 1             num_rows  : 1             rows      :
len       : 53           data      : f0ff0fff0f0f00
codes     : {53}f0ff0fff0f0f00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.244416s
model     : new          count     : 1             num_rows  : 1             rows      :
len       : 41           data      : ff0ffffff00
codes     : {41}ff0ffffff00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.244416s
model     : new          count     : 1             num_rows  : 1             rows      :
len       : 9            data      : f00
codes     : {9}f00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.244416s
model     : new          count     : 1             num_rows  : 1             rows      :
len       : 21           data      : ff0f00
codes     : {21}ff0f00
[... 17 lines cut ...]
model     : new          count     : 1             num_rows  : 1             rows      :
len       : 53           data      : f0ff0fff0f0f00
codes     : {53}f0ff0fff0f0f00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.244416s
model     : new          count     : 1             num_rows  : 1             rows      :
len       : 41           data      : ff0ffffff00
codes     : {41}ff0ffffff00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.244416s
model     : new          count     : 1             num_rows  : 1             rows      :
len       : 9            data      : f00
codes     : {9}f00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.244416s
model     : new          count     : 1             num_rows  : 1             rows      :
len       : 21           data      : ff0f00
codes     : {21}ff0f00
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : @0.244416s
model     : new          count     : 1             num_rows  : 1             rows      :
len       : 125          data      : ff0ff0ffffffffffffffffffffff0f00
codes     : {125}ff0ff0ffffffffffffffffffffff0f00
`
I hope to do something nice, sorry if I made mistakes

zuckschwerdt, 2020-07-13:
We do have support for the WH32B. https://github.com/merbanan/rtl_433/blob/master/src/devices/fineoffset.c#L428
But there might be problems with demod and detection of bitlength.
Try adding `-Y classic`. If that does not zip and upload a sample (from -S unknown) here.

ciasco92, 2020-07-13:
unfortunately with -Y classic I'm not successful, I uploaded my cu8 files [here](https://github.com/merbanan/rtl_433/files/4911829/filecu8.zip)

zuckschwerdt, 2020-07-13:
It seems to work here?
```
rtl_433 -s 1000k g001_868.3M_1000k.cu8
...
[00] {922} ff 00 00 55 55 55 55 55 55 16 ea 6f c1 34 1b 7f ff fe 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 00 00 2a aa aa aa aa aa 8b 75 37 e0 9a 0d bf ff ff 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

time      : @0.061104s
model     : Fineoffset-WH32B                       ID        : 248
Battery   : 1            Temperature: 21.6 C       Humidity  : 54 %          Pressure  : 6553.5 hPa    Integrity : CRC
```

ciasco92, 2020-07-13:
I tried several files, after this string `Test mode active. Reading samples from file: g002_868.3M_1000k.cu8` I get nothing:

`pi@swpi ~ $ rtl_433 -s 1000k g001_868.3M_1000k.cu8
rtl_433 version 20.02-86-gd362397 branch master at 202007062107 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 128 out of 156 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-156 ]
Test mode active. Reading samples from file: g001_868.3M_1000k.cu8
pi@swpi ~ $ rtl_433 -s 1000k g002_868.3M_1000k.cu8
rtl_433 version 20.02-86-gd362397 branch master at 202007062107 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 128 out of 156 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-156 ]
Test mode active. Reading samples from file: g002_868.3M_1000k.cu8
pi@swpi ~ $ rtl_433 -s 1000k g003_868.3M_1000k.cu8
rtl_433 version 20.02-86-gd362397 branch master at 202007062107 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/pi/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 128 out of 156 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-156 ]
Test mode active. Reading samples from file: g003_868.3M_1000k.cu8
`

zuckschwerdt, 2020-07-13:
Ok, I can confirm this. It's my local changes that enable this. The `msg_type` expected is `0xe0` but we have `0xdf` here.
I'll add this.
Can you confirm that this is not a WH32B but a WH32 (no extra letters or anything?)

ciasco92, 2020-07-13:
yes, I confirm that the acronym wh32 is shown without any other letters.
thank you so much for all the work you do

zuckschwerdt, 2020-07-13:
Thanks :) If you pull the latest changes and rebuild it should be supported.

closed 2020-07-13 by zuckschwerdt

ciasco92, 2020-07-13:
Grazie, Thanks
```
time      : 2020-07-13 14:04:04
model     : Fineoffset-WH32                        ID        : 248
Battery   : 1            Temperature: 22.9 C       Humidity  : 51 %
Pressure  : 6553.5 hPa   Integrity : CRC
```

zuckschwerdt, 2020-07-13:
I just realize we might need to fix 0xffff pressure (6553.5 hPa) -> n/a

ciasco92, 2020-07-13:
if I'm not mistaken, the wh32 does not have a barometric sensor

zuckschwerdt, 2020-07-20:
With 9bbf20b pressure is now excluded on WH32.

commit f58e4355 (2020-07-13): Add support for Fineoffset WH32 (closes #1431)