merbanan/rtl_433 #1435: What have you done

Midimagic, 2020-07-18:


merbanan, 2020-07-18:
Please elaborate.

Midimagic, 2020-07-18:
Hi
I have downloaded your latest version and it hardly decodes anything, but when I use your older version it decodes a lot more. what have you done to it ?

merbanan, 2020-07-18:
Post complete command line. It might explain your issue.

Midimagic, 2020-07-18:
D:\Ham Programs\rtl_433-20.02-58-ga0b9b1b-win-x64>rtl_433.exe

Midimagic, 2020-07-18:
It used to display this when I started it:

D:\Ham Programs\rtl433_win64_daily>rtl_433.exe
Registering protocol [1] "Rubicson Temperature Sensor"
Registering protocol [2] "Prologue Temperature Sensor"
Registering protocol [3] "Waveman Switch Transmitter"
Registering protocol [4] "LaCrosse TX Temperature / Humidity Sensor"
Registering protocol [5] "Acurite 609TXC Temperature and Humidity Sensor"
Registering protocol [6] "Oregon Scientific Weather Sensor"
Registering protocol [7] "KlikAanKlikUit Wireless Switch"
Registering protocol [8] "AlectoV1 Weather Sensor (Alecto WS3500 WS4500 Ventus W155/W044 Oregon)"
Registering protocol [9] "Cardin S466-TX2"
Registering protocol [10] "Fine Offset Electronics, WH2 Temperature/Humidity Sensor"
Registering protocol [11] "Nexus Temperature & Humidity Sensor"
Registering protocol [12] "Ambient Weather Temperature Sensor"
Registering protocol [13] "Calibeur RF-104 Sensor"
Registering protocol [14] "DSC Security Contact"
Registering protocol [15] "GT-WT-02 Sensor"
Registering protocol [16] "Danfoss CFR Thermostat"
Registering protocol [17] "Chuango Security Technology"
Registering protocol [18] "Generic Remote SC226x EV1527"
Registering protocol [19] "TFA-Twin-Plus-30.3049 and Ea2 BL999"
Registering protocol [20] "Fine Offset Electronics WH1080/WH3080 Weather Station"
Registering protocol [21] "WT450"
Registering protocol [22] "LaCrosse WS-2310 Weather Station"
Registering protocol [23] "Esperanza EWS"
Registering protocol [24] "Efergy e2 classic"
Registering protocol [25] "Generic temperature sensor 1"
Registering protocol [26] "WG-PB12V1"
Registering protocol [27] "Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning"
Registering protocol [28] "Acurite 986 Refrigerator / Freezer Thermometer"
Registering protocol [29] "HIDEKI TS04 Temperature, Humidity, Wind and Rain Sensor"
Registering protocol [30] "Watchman Sonic / Apollo Ultrasonic / Beckett Rocket oil tank monitor"
Registering protocol [31] "CurrentCost Current Sensor"
Registering protocol [32] "emonTx OpenEnergyMonitor"
Registering protocol [33] "HT680 Remote control"
Registering protocol [34] "S3318P Temperature & Humidity Sensor"
Registering protocol [35] "Akhan 100F14 remote keyless entry"
Registering protocol [36] "Quhwa"
Registering protocol [37] "OSv1 Temperature Sensor"
Registering protocol [38] "Proove"
Registering protocol [39] "Bresser Thermo-/Hygro-Sensor 3CH"
Registering protocol [40] "Springfield Temperature and Soil Moisture"
Registering protocol [41] "Oregon Scientific SL109H Remote Thermal Hygro Sensor"
Registering protocol [42] "Acurite 606TX Temperature Sensor"
Registering protocol [43] "TFA pool temperature sensor"
Registering protocol [44] "Kedsum Temperature & Humidity Sensor"
Registering protocol [45] "blyss DC5-UK-WH (433.92 MHz)"
Registering protocol [46] "Steelmate TPMS"
Registering protocol [47] "Schrader TPMS"
Registering protocol [48] "Elro DB286A Doorbell"
Registering protocol [49] "Efergy Optical"
Registering protocol [50] "Honda Car Key"
Registering protocol [51] "Fine Offset Electronics, XC0400"
Registering protocol [52] "Radiohead ASK"
Registering protocol [53] "Kerui PIR Sensor"
Registering protocol [54] "Fine Offset WH1050 Weather Station"
Registering protocol [55] "Honeywell Door/Window Sensor"
Registering protocol [56] "Maverick ET-732/733 BBQ Sensor"
Registering protocol [57] "LaCrosse TX141-Bv2/TX141TH-Bv2 sensor"
Registering protocol [58] "Acurite 00275rm,00276rm Temp/Humidity with optional probe"
Registering protocol [59] "LaCrosse TX35DTH-IT Temperature sensor"
Registering protocol [60] "LaCrosse TX29IT Temperature sensor"
Registering protocol [61] "Vaillant calorMatic 340f Central Heating Control"
Registering protocol [62] "Fine Offset Electronics, WH25 Temperature/Humidity/Pressure Sensor"
Registering protocol [63] "Fine Offset Electronics, WH0530 Temperature/Rain Sensor"
Registering protocol [64] "IBIS beacon"
Registering protocol [65] "Oil Ultrasonic STANDARD FSK"
Registering protocol [66] "Citroen TPMS"
Registering protocol [67] "Oil Ultrasonic STANDARD ASK"
Registering protocol [68] "Thermopro TP11 Thermometer"
Registering protocol [69] "Solight TE44"
Registering protocol [70] "Wireless Smoke and Heat Detector GS 558"
Registering protocol [71] "Generic wireless motion sensor"
Registering protocol [72] "Toyota TPMS"
Registering protocol [73] "Ford TPMS"
Registering protocol [74] "Renault TPMS"
Registering protocol [75] "FT-004-B Temperature Sensor"
Registering protocol [76] "Ford Car Key"
Registering protocol [77] "Philips outdoor temperature sensor"
Registering protocol [78] "Schrader TPMS EG53MA4"
Registering protocol [79] "Nexa"
Registering protocol [80] "Thermopro TP12 Thermometer"
Registering protocol [81] "GE Color Effects"
Registering protocol [82] "X10 Security"
Registering protocol [83] "Interlogix GE UTC Security Devices"
Registered 83 out of 102 device decoding protocols
Found 1 device(s)

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Found Rafael Micro R820T tuner
Using device 0: Generic RTL2832U OEM
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.

merbanan, 2020-07-18:
Yes, the removal of that list is by design. The decoders are still there (plus around 70 more).

Midimagic, 2020-07-18:
All I get now is:

rtl_433 version 20.02-58-ga0b9b1b branch  at 202005231014 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "D:\Ham Programs\rtl_433-20.02-58-ga0b9b1b-win-x64\rtl_433.conf"...
Trying conf file at "C:\Users\magic\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Registered 125 out of 153 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-153 ]
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-18 10:49:58
model     : TFA-Pool     Id        : 110
Channel   : 1            Battery   : 1             Temperature: 24.4 C       Integrity : CHECKSUM

merbanan, 2020-07-18:
Registered 125 out of 153 device decoding protocols. <- That is the important line.

merbanan, 2020-07-18:
Is there any issues with reception of signals? Or only the output during startup?

Midimagic, 2020-07-18:
The old  version picks up a lot more:

Found 1 device(s)

trying device  0:  Realtek, RTL2838UHIDIR, SN: 00000001
Found Rafael Micro R820T tuner
Using device 0: Generic RTL2832U OEM
Exact sample rate is: 250000.000414 Hz
Sample rate set to 250000.
Bit detection level set to 0 (Auto).
Tuner gain set to Auto.
Reading samples in async mode...
Tuned to 433920000 Hz.
2020-07-18 10:55:04 :   TFA pool temperature sensor
        Id:              96
        Channel number:          1
        Temperature:     23.5 C
2020-07-18 10:55:07 :   Nexus Temperature/Humidity
        House Code:      131
        Battery:         OK
        Channel:         1
        Temperature:     22.60 C
        Humidity:        57 %
2020-07-18 10:55:09 :   Nexus Temperature/Humidity
        House Code:      188
        Battery:         OK
        Channel:         1
        Temperature:     25.10 C
        Humidity:        50 %
2020-07-18 10:55:17 :   TFA pool temperature sensor
        Id:              110
        Channel number:          1
        Temperature:     25.2 C
2020-07-18 10:55:28 :   Fine Offset Electronics WH1080/WH3080 Weather Station
        Msg type:        0
        Station ID:      121
        Temperature:     20.9 C
        Humidity:        63 %
        Wind string:     NW
        Wind degrees:    315
        Wind avg speed:  7.34
        Wind gust:       15.91
        Total rainfall:  67.8
        Battery:         OK
2020-07-18 10:55:28 :   Fine Offset Electronics WH1080/WH3080 Weather Station
        Msg type:        0
        Station ID:      121
        Temperature:     20.9 C
        Humidity:        63 %
        Wind string:     NW
        Wind degrees:    315
        Wind avg speed:  7.34
        Wind gust:       15.91
        Total rainfall:  67.8
        Battery:         OK
2020-07-18 10:55:53 :   TFA pool temperature sensor
        Id:              110
        Channel number:          1
        Temperature:     25.1 C
2020-07-18 10:56:04 :   Nexus Temperature/Humidity
        House Code:      131
        Battery:         OK
        Channel:         1
        Temperature:     22.60 C
        Humidity:        57 %
2020-07-18 10:56:06 :   Nexus Temperature/Humidity
        House Code:      188
        Battery:         OK
        Channel:         1
        Temperature:     25.20 C
        Humidity:        49 %
2020-07-18 10:56:06 :   Springfield Temperature & Moisture
        SID:     188
        Channel:         1
        Battery:         LOW
        Transmit:        AUTO
        Temperature:     25.2 C
        Moisture:        15

Midimagic, 2020-07-18:
It picks up a lot less, or does not decode them.

Midimagic, 2020-07-18:
The old version of the program is 300KB in lengh  It does not give a version number.

merbanan, 2020-07-18:
Can you identify what sensors you are not receiving?

Midimagic, 2020-07-18:
I now get an error:
D:\Ham Programs\rtl_433-20.02-58-ga0b9b1b-win-x64>rtl_433.exe
rtl_433 version 20.02-58-ga0b9b1b branch  at 202005231014 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "D:\Ham Programs\rtl_433-20.02-58-ga0b9b1b-win-x64\rtl_433.conf"...
Trying conf file at "C:\Users\magic\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...
Registered 125 out of 153 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-153 ]
usb_open error -3
Please fix the device permissions, e.g. by installing the udev rules file rtl-sdr.rules

Midimagic, 2020-07-18:
2018-08-26 21:13:43 :	Fine Offset Electronics WH1080/WH3080 Weather Station
	Msg type:	 0
	Station ID:	 125
	Temperature:	 17.5 C
	Humidity:	 97 %
	Wind string:	 SW
	Wind degrees:	 225
	Wind avg speed:	 3.67
	Wind gust:	 7.34
	Total rainfall:	 457.5
	Battery:	 OK
2018-08-26 21:13:43 :	Fine Offset Electronics WH1080/WH3080 Weather Station
	Msg type:	 0
	Station ID:	 125
	Temperature:	 17.5 C
	Humidity:	 97 %
	Wind string:	 SW
	Wind degrees:	 225
	Wind avg speed:	 3.67
	Wind gust:	 7.34
	Total rainfall:	 457.5
	Battery:	 OK

Midimagic, 2020-07-18:
I am not getting these:

2018-09-03 12:29:21 :	Fine Offset Electronics WH3080 Weather Station
	Msg type:	 2
	UV Sensor ID:	 89
	Sensor Status:	 OK
	UV Index:	 3
	Lux	:	 47748.7
	Watts/m	:	 69.91
	Foot-candles:	 4437.61

2018-08-26 21:12:07 :	Fine Offset Electronics WH1080/WH3080 Weather Station
	Msg type:	 0
	Station ID:	 125
	Temperature:	 17.5 C
	Humidity:	 97 %
	Wind string:	 SW
	Wind degrees:	 225
	Wind avg speed:	 4.90
	Wind gust:	 11.02
	Total rainfall:	 457.5
	Battery:	 OK

Midimagic, 2020-07-18:
That is a weather station next door and it is always there.  TFA-Pool is my AURIOL Temp sensor.

merbanan, 2020-07-18:
The code for these devices have changed alot during development. To help us fix this issue you can record the signals and attach them to this issue.

Use the new version and run with "rtl_433 -S unknown" for a while. Then use the old version with rtl_433 -r g00XX.cu8". When you find signals that decode with the old version but not the new version attach that file together with the output. We can then look into the issue.

Midimagic, 2020-07-18:
When you say signals do you just mean the text ?

merbanan, 2020-07-18:
No, actual raw baseband recordings of the signal. Just run "rtl_433 -S unknown" and it should output files whenever the WH3080 device transmits.

Midimagic, 2020-07-18:
Ok I will give it 10 mins of each.

Midimagic, 2020-07-18:
[rtl_433-20.02-58-ga0b9b1b-win-x64.zip](https://github.com/merbanan/rtl_433/files/4941490/rtl_433-20.02-58-ga0b9b1b-win-x64.zip)

merbanan, 2020-07-18:
Can you post the output from one sample file that works with the old rtl_433 but doesn't with the new version?

Midimagic, 2020-07-18:
[unknown.zip](https://github.com/merbanan/rtl_433/files/4941503/unknown.zip)
There seems to be parameters missing like wind String.
This one seems to be missing:
2018-08-23 17:51:47 :	Fine Offset Electronics WH3080 Weather Station
	Msg type:	 2
	UV Sensor ID:	 89
	Sensor Status:	 OK
	UV Index:	 1
	Lux	:	 12594.7
	Watts/m	:	 18.44
	Foot-candles:	 1170.51

Midimagic, 2020-07-18:
What is the difference between the Static and the normal program ?

merbanan, 2020-07-18:
Can you upload the samples like you did the first time? The second archive looks corrupt.

merbanan, 2020-07-18:
The static program should be more self contained.

Midimagic, 2020-07-18:
I will try again

Midimagic, 2020-07-18:
[unknown.zip](https://github.com/merbanan/rtl_433/files/4941537/unknown.zip)

merbanan, 2020-07-18:
Same issue. Can you do exactly like you did the first time? And make sure you only include .cu8 files.

Midimagic, 2020-07-18:
Hi

Sorry but the earlier version does not do that, there is no output to the screen and only one file gets saved when I do a ctrl C at the end.

Midi

merbanan, 2020-07-18:
The old version used another syntax to record signals, I only meant you to use the newer version to record signals. To me it seems that the signals are decoded but not output. With -S unknown you only get signals rtl_433 can't decode. Try "rtl_433 -S all" with the new version of rtl_433 and upload a new set of files.

Midimagic, 2020-07-18:
Ok will do tomorrow. I think it decodes them but some of the parameters are missing Wind string and the UV settings.

Thanks
Midi

Midimagic, 2020-07-18:
Hi Benjamin

I am doing a session now. You don't happen to know how to decode the Thames Water meters do you ?

Midi

Midimagic, 2020-07-18:
Midi Magic has shared a OneDrive file with you. To view it, click the link below.
<https://1drv.ms/u/s!Ai5b7UawqlwzhBghKQlYPx-jm3x->
[https://r1.res.office365.com/owa/prem/images/dc-zip_20.png]<https://1drv.ms/u/s!Ai5b7UawqlwzhBghKQlYPx-jm3x->
rtl_433-20.02-58-ga0b9b1b-win-x64.zip<https://1drv.ms/u/s!Ai5b7UawqlwzhBghKQlYPx-jm3x->

Latest files. It is a bit big so I hope you can get it from the cloud.

Midi

merbanan, 2020-07-18:
https://github.com/weetmuts/wmbusmeters can read the Sensus iPERL (iperl) water meter. If you tune rtl_433 to 868.8M you might be able to pick up the wmbus telegrams.

merbanan, 2020-07-18:
The WH3080 Weather Station is found. But Msg type: 2 data is missing.

```
ime      : @0.218044s
model     : Fineoffset-WHx080                      Msg type  : 0             Station ID: 121
Battery   : 1            Temperature: 20.9 C       Humidity  : 72 %          Wind Direction: 270       Wind avg speed: 1.22
Wind gust : 2.45         Total rainfall: 67.8      Integrity : CRC

```

merbanan, 2020-07-18:
I took a look but I could not find any signal that should represent the uv data (etc) from the weather sensor. Use the old version and run with -a -t to record signals with that. Then when you get a decode of the uv data attach that signal to this ticket.

Midimagic, 2020-07-19:
How do I know when I have a decode as it does not display the decode on the screen ?

Midi

merbanan, 2020-07-19:
You can run "rtl_433 -r filename" to replay the signal. First make sure that the old rtl_433 actually picks up the signal the new version fails to pick up. Lets say the signal is transmitted once every minute. When that is confirmed you record signals for 2 minutes. Now you can be sure you recorded at least one transmission. After the 2 minutes have passed you maybe have 10 recordings. Now replay all recordings until you find the you are missing. When you found it upload it here.

Midimagic, 2020-07-19:
Hi

Here is the file that has some missing data on the new version.
Old version
Input format: uint8
@0.000000s :    Fine Offset Electronics WH1080/WH3080 Weather Station
        Msg type:        0
        Station ID:      121
        Temperature:     15.8 C
        Humidity:        99 %
        Wind string:     S
        Wind degrees:    180
        Wind avg speed:  1.22
        Wind gust:       2.45
        Total rainfall:  67.8
        Battery:         OK
@0.000000s :    Fine Offset Electronics WH1080/WH3080 Weather Station
        Msg type:        0
        Station ID:      121
        Temperature:     15.8 C
        Humidity:        99 %
        Wind string:     S
        Wind degrees:    180
        Wind avg speed:  1.22
        Wind gust:       2.45
        Total rainfall:  67.8
        Battery:         OK
Test mode file issued 1 packets

New version

time      : 2020-07-19 12:40:16
model     : Fineoffset-WHx080                      Msg type  : 0             Station ID: 121
Battery   : 1            Temperature: 16.3 C       Humidity  : 97 %          Wind Direction: 225
Wind avg speed: 2.45     Wind gust : 4.90          Total rainfall: 67.8      Integrity : CRC
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-19 12:40:16
model     : Fineoffset-WHx080                      Msg type  : 0             Station ID: 121
Battery   : 1            Temperature: 16.3 C       Humidity  : 97 %          Wind Direction: 225
Wind avg speed: 2.45     Wind gust : 4.90          Total rainfall: 67.8      Integrity : CRC

The one with the UV seems to have stopped transmitting now.

Midi

merbanan, 2020-07-19:
Ok, then there likely is no issue. Everything works as designed and the UV sensor has just stopped transmitting for some reason.

Midimagic, 2020-07-19:
Hi

So you have taken the compass points out then ?

Midi

merbanan, 2020-07-19:
Seems like that, I think we normalised with degrees:

https://github.com/merbanan/rtl_433/blob/master/src/devices/fineoffset_wh1080.c#L106

Midimagic, 2020-07-19:
Hi Benjamin

Do you know of any PC software to decode wmbusmeters ?
I had a look but I am far too old to follow all of that.

Midi

merbanan, 2020-07-19:
rtl_433 can receive the wmbus telegrams. rtl_433 -f 868.8M should do the trick. To actually decoding the data the only free option I know of is wmbusmeters. But first see if you get any readings at all.

Midimagic, 2020-07-19:
Hi

Ok thanks for all your help.
Not sure I like degrees though, as there only seems to be 8 options to choose from as it does not show the true wind direction. I.m too old to have to work out what 270 degrees is.

Midi

merbanan, 2020-07-19:
Anyway AFAIK what has changes is the uv sensor has stopped transmitting and the output has changed a bit. No real issue could be found and thus I close this issue.

closed 2020-07-19 by merbanan

Midimagic, 2020-07-19:
Thank you for all your help.

Midimagic, 2020-07-20:
Hi

I have set the frequency to 868.500Mhz and I am receiving signals every couple of seconds. But alas, no decodes.
I am using the -A command.

Many thanks
Midi

merbanan, 2020-07-20:
You can try using rtl-wmbus it is better at decoding wmbus signals. But there are other signals transmitting on the 868 band also.

merbanan, 2020-07-20:
My point is that it might not be wmbus signals you are receiving.

Midimagic, 2020-07-20:
Hi

I left it running for a number of hours today and got a lot of the same decode on 868.300Mhz
Maybe it is not a car remote ?

regards

Midi

D:\Ham Programs\rtl_433-20.02-58-ga0b9b1b-win-x64>rtl_433.exe -f 868300000 -G 4
rtl_433 version 20.02-58-ga0b9b1b branch  at 202005231014 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "D:\Ham Programs\rtl_433-20.02-58-ga0b9b1b-win-x64\rtl_433.conf"...
Trying conf file at "C:\Users\magic\AppData\Local\rtl_433\rtl_433.conf"...
Trying conf file at "C:\ProgramData\rtl_433\rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

        Use -G for testing only. Enable protocols with -R if you really need them.

Registered 147 out of 153 device decoding protocols [ 1-4 6-8 10-17 19-26 29-64 67-153 ]
rtl_433: warning: 106 "Wireless M-Bus, Mode R, 4.8kbps (-f 868330000)" does not support CSV output
rtl_433: warning: 107 "Wireless M-Bus, Mode F, 2.4kbps" does not support CSV output
Found Rafael Micro R820T tuner
Exact sample rate is: 1000000.026491 Hz
[R82XX] PLL not locked!
Sample rate set to 1000000 S/s.
Tuner gain set to Auto.
Tuned to 868.300MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-20 15:02:07
model     : Honda-CarRemote                        id        : 65535
code      : unknown
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-20 15:06:53
model     : Honda-CarRemote                        id        : 65502
code      : unknown
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-20 15:12:59
model     : Honda-CarRemote                        id        : 65535
code      : unknown
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-20 15:14:43
model     : Honda-CarRemote                        id        : 65535
code      : unknown
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-20 15:18:15
model     : Honda-CarRemote                        id        : 65535
code      : unknown
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-20 15:28:26
model     : Honda-CarRemote                        id        : 57341
code      : unknown
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-20 15:30:09
model     : Honda-CarRemote                        id        : 65535
code      : unknown
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-20 15:36:35
model     : Honda-CarRemote                        id        : 65535
code      : unknown
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-20 16:08:05
model     : Honda-CarRemote                        id        : 65535
code      : unknown
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-20 16:19:07
model     : Honda-CarRemote                        id        : 65535
code      : unknown
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
[... 58 lines cut ...]
code      : unknown
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-20 19:03:53
model     : Honda-CarRemote                        id        : 65535
code      : unknown
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-20 19:13:29
model     : Honda-CarRemote                        id        : 65535
code      : unknown
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-20 19:18:01
model     : Honda-CarRemote                        id        : 65535
code      : unknown
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-20 19:19:05
model     : Honda-CarRemote                        id        : 65535
code      : unknown
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-20 19:21:31
model     : Honda-CarRemote                        id        : 65535
code      : unknown
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2020-07-20 19:29:15
model     : Honda-CarRemote                        id        : 65535
code      : unknown

merbanan, 2020-07-20:
Yeah, it is not a Honda CarRemote. Newer releases has the decoder disabled. It is most likely the water sensors you talked about.

merbanan, 2020-07-20:
You can record a few signals and analyse them if you are curious.