merbanan/rtl_433 #1437: Running rtl_433 on a raspberry pi then exiting will cause rtlsdr_read_reg failed with -7 on the third run.

Id405, 2020-07-20:
If you run rtl_433 on a raspberry pi then exit with ctrl-c then the third run will give 

```
rtlsdr_read_reg failed with -7
rtlsdr_write_reg failed with -7
rtlsdr_read_reg failed with -7
rtlsdr_write_reg failed with -7
rtlsdr_read_reg failed with -7
rtlsdr_write_reg failed with -7
rtlsdr_read_reg failed with -7
rtlsdr_write_reg failed with -7
```

After that it will start normally but you will never recieve any data from weather stations, the only way to make it work again is to restart the raspberry pi.

zuckschwerdt, 2020-07-20:
There is a usb-reset` or `usbreset`tool around that should help reset the bus without restart of the raspi. https://askubuntu.com/questions/645/how-do-you-reset-a-usb-device-from-the-command-line

This isn't a fault in rtl_433 but in the state between the raspi and the dongle.

Shouldn't rtl_433 abort if there is no data? Can you show the output from a "third run"? Best use `rtl_433 -A` or even do `rtl_433 -w out.cu8 -T 60` and inspect the out.cu8 file (e.g. with https://triq.org/iqs/ )

Id405, 2020-07-20:
It does abort eventually. Thanks, Ill integrate the usb reset tool into my script.

closed 2020-07-20 by Id405

zuckschwerdt, 2020-07-20:
We have been pondering to include a usb reset, but we can't know if it's needed and which device to apply to. Maybe we should include the tool and leave using it to the user?
See also https://github.com/merbanan/rtl_433/issues/1394#issuecomment-660202961

merbanan, 2020-07-20:
Well, I'd like to provide a feature complete solution and people do use rpi for this so providing a simple switch to enable it would be "nice to have".

zuckschwerdt, 2020-07-20:
The reset code is simple but we would need to know which usb device librtlsdr did open.

Id405, 2020-07-21:
So I tried to use the reset code and it seems random if it actually makes rtl_433 work again or not. Sometimes it will take like a minute to run the reset and other times it will reset almost instantly. I was never able to find a consistent way to make the rtl_433 command work every time I ran it.

zuckschwerdt, 2020-07-21:
This isn't rtl_433's doing though. You could run rtl_sdr or some other SDR tool like rx_sdr and get the same stuck USB. Since this never happend for me I guess it has something to do with the actual dongle and perhaps power supply?